llvm/circt #5698: [FIRRTL][ExportVerilog] Mult/Add Width Cast

SingularityKChen, 2023-07-27:
Except the signedness mentioned in https://github.com/llvm/circt/issues/4152, seems the bit cast also doesn't work. I tried both Version 1.48.0 and 1.44.0.

Here we use the example in https://github.com/llvm/circt/pull/2825:

```firrtl
circuit Foo:
  module Foo:
    input a: UInt<4>
    input b: UInt<4>
    output d: UInt<8>

    node c = mul(a, b)
    d <= c
```

Runing `firtool Foo.fir` got:
```verilog
// Generated by CIRCT firtool-1.48.0
module Foo(
  input  [3:0] a,
               b,
  output [7:0] d
);

  assign d = {4'h0, a} * {4'h0, b};
endmodule

```

Runing `firtool --lowering-options=explicitBitcast Foo.fir` got:
```verilog
// Generated by CIRCT firtool-1.48.0
module Foo(
  input  [3:0] a,
               b,
  output [7:0] d
);

  assign d = 8'({4'h0, a} * {4'h0, b});
endmodule
```

From [the document](https://circt.llvm.org/docs/VerilogGeneration/) and https://github.com/llvm/circt/pull/2825, what we want here might be:
```verilog
// default
assign d = a * b;
// with --lowering-options=explicitBitcast
assign d = 8'(a * b);
```

Please kindly suggest whether I missed any option that may help with this.

seldridge, 2023-07-27:
There's no option to get the output that you're looking for. I agree that what you want is the desired output.

This is doable to fix, but a bit tricky. The problem is that the modeling of `comb.mul` is `(i8, i8) -> i8`, i.e., it needs to have all the types match. This is a reasonable way for this op to work in the IR. However, it creates unnecessary sign extensions in the output Verilog which open source tooling (notably Yosys) has problems with whereas proprietary tooling doesn't care.

That HW dialect for the above is:

```mlir
hw.module @Foo(%a: i4, %b: i4) -> (d: i8) {
  %c0_i4 = hw.constant 0 : i4
  %0 = comb.concat %c0_i4, %a : i4, i4
  %1 = comb.concat %c0_i4, %b : i4, i4
  %2 = comb.mul bin %0, %1 {sv.namehint = "c"} : i8
  hw.output %2 : i8
}
```

I do think that when using the explicit bit cast or by making that the default we could probably avoid this, as you say. Trying to output `d = a * b` is fraught with problems due to Verilog context sensitive widths. However, `d = 8'(a * b)` should make the width context insensitive and avoid complicated checking at the output.

I like the idea. 👍

SingularityKChen, 2023-07-27:
> There's no option to get the output that you're looking for. I agree that what you want is the desired output.
>
> This is doable to fix, but a bit tricky. The problem is that the modeling of [...]

Thanks Seldridge. 

I'm running quick PDCA cycle from Chisel to GDSII. From the synthesis results, I quite care the bit cast issue and the signedness, whose coding style (or we say, the generated RTL) does affect the PPA.

From the Genus Datapath Coding Styles, we are suggested  **Avoiding Manual Signal Extension Techniques**: Guideline: Declare appropriate signal types to automatically handle signal extension.

```verilog
// Example 4-20 Manual Zero-Extension for Unsigned Signals
wire [15:0] a, b;  // to be used as unsigned
wire [16:0] s;     // to be used as unsigned 

assign s = {1’b0, a} + {1’b0, b};

// Example 4-21 Recommended Coding Style for Unsigned Signals

wire [15:0] a, b;  // unsigned
wire [16:0] s;     // unsigned 

assign s = a + b;

// Example 4-22 Manual Sign-Extension for Signed Signals

wire [15:0] a, b;  // to be used as signed
wire [16:0] s;     // to be used as signed

assign s = {a[15], a} + {b[15], b};

// Example 4-23 Recommended Coding Style for Signed Signals

wire signed [15:0] a, b;  // signed
wire signed [16:0] s;     // signed

assign s = a + b;
```

In example 4-10 and 4-11, without the manual sign bit extend, the Genus could infer a smaller multiplier.

```verilog
 // example 4-10
wire [16:0] a;          // to be used as signed
wire [14:0] b;          // to be used as signed
wire [31:0] y;         // to be used as signed
wire [31:0] ax, bx;

assign ax = {{15{a[16]}}, a};
assign bx = {{17{b[14]}}, b};
assign y = ax * bx;     // a 32x32 unsigned multiplier

// example 4-11
wire signed [16:0] a;
wire signed [14:0] b; 
wire signed [31:0] y;

assign y = a * b;         // infers a 17x15 signed multiplier
```

Hence right now I manually modify the output hdl from firtool to match the above suggested coding styles.

It could be great if circt could also refer (or add some options) the coding styles that some synthesis tools suggested for better PPA.

CSharperMantle, 2025-03-24:
Similar to SingularityKChen, I encountered the same problem with my design, in which I had two `UInt<32>`s to multiply.

> However, it creates unnecessary sign extensions in the output Verilog which [...] proprietary tooling doesn't care.

It is worth noting that Vivado 2024.1 struggled with this pattern. The ordinary `a * b` synthesized to 4 DSPs with 6 optimal cycles on my target, and `{32'h0, a} * {32'h0, b}` synthesized to 8 DSPs with 16 optimal cycles. But considering Vivado is not *the* tool for synthesis, this should not be that surprising...?