llvm/circt #5721: [FIRRTL] FIRRTL Exporter needs to unique names

seldridge, 2023-07-28:
Currently, the FIRRTL exporter will not properly unique names during emission (like what `ExportVerilog` does). Consider the following:

```mlir
firrtl.circuit "Foo" {
  firrtl.module @Foo() {
    %0 = firrtl.wire {name = "a"} : !firrtl.uint<1>
    %1 = firrtl.wire {name = "a"} : !firrtl.uint<1>
  }
}
```

This produces the following illegal FIRRTL (`circt-translate -export-firrtl Foo.mlir`):
```
FIRRTL version 3.1.0
circuit Foo :
  module Foo : 
    wire a : UInt<1> 
    wire a : UInt<1> 
```

`wire a` is defined twice.

dtzSiFive, 2023-07-31:
Ability to name (uniquely) entities would definitely be useful!
This would help with https://github.com/llvm/circt/issues/5205 , for example.

Previously @fabianschuiki 's multibit mux emitter support has code for managing + generating names, FWIW: https://github.com/llvm/circt/issues/2966#issuecomment-1577135676 .

dtzSiFive, 2023-08-11:
Also, emitter uses a `CircuitNamespace` presently for generating unique-ish names for materialized "invalid" values, but that only guarantees uniqueness at the top level (symbols -- mostly just modules), not within the names in the current module.

When addressing this issue, we should generate unique names for these as well.