llvm/circt #5731: Passes

MuhammadTalha-10, 2023-07-31:
I am just starting out with CIRCT I wanted to know the procedure on lowering dialects to Verilog. I went through the CIRCT documentation and came upon passes. I want to know if these passes and optimizations are supposed to be done through C++ (importing classes and functions) or command line flags of the compiled executables after compiling CIRCT. If it is the former, do I just import the classes from the repository? Is there a specific directory containing all the passes or do I have to look around?

One more thing. Can the workflow of transformation of a high level language to Verilog using CIRCT can be done with the Python bindings and C++ programming (collectively and/or individually).

MuhammadTalha-10, 2024-03-22:
I want to know that are there any optimization passes available in calyx?

cgyurgyik, 2025-01-18:
Hi Muhammad, there are optimization passes in Calyx available in the native compiler (see: [how to run passes](https://docs.calyxir.org/compiler.html). In CIRCT, the Calyx dialect typically focuses on lowering from other dialects and simpler peephole optimizations.