llvm/circt #5815: [PipelineToHW]  `Conversion/PipelineToHW/test_outlined.mlir` failure

uenoku, 2023-08-09:
Fail randomly (probability is around 10%).
```
FAIL: CIRCT :: Conversion/PipelineToHW/test_outlined.mlir (582 of 641)
******************** TEST 'CIRCT :: Conversion/PipelineToHW/test_outlined.mlir' FAILED ********************
Script:
--
: 'RUN: at line 1';   /scratch/hidetou/circt/build/bin/circt-opt --lower-pipeline-to-hw="outline-stages" /scratch/hidetou/circt/test/Conversion/PipelineToHW/test_outlined.mlir | /scratch/hidetou/circt/build/bin/FileCheck /scratch/hidetou/circt/test/Conversion/PipelineToHW/test_outlined.mlir
--
Exit Code: 2

Command Output (stderr):
--
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
 #0 0x0000000000447a67 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /scratch/hidetou/circt/llvm/llvm/lib/Support/Unix/Signals.inc:602:13
 #1 0x0000000000445bf0 llvm::sys::RunSignalHandlers() /scratch/hidetou/circt/llvm/llvm/lib/Support/Signals.cpp:105:18
 #2 0x000000000044837f SignalHandler(int) /scratch/hidetou/circt/llvm/llvm/lib/Support/Unix/Signals.inc:413:1
 #3 0x00007f15c6e39cf0 __restore_rt (/lib64/libpthread.so.0+0x12cf0)
 #4 0x0000000001049a30 llvm::ilist_node_base<true>::isSentinel() const /scratch/hidetou/circt/llvm/llvm/include/llvm/ADT/ilist_node_base.h:45:36
 #5 0x0000000001049a30 llvm::ilist_node_base<true>::isKnownSentinel() const /scratch/hidetou/circt/llvm/llvm/include/llvm/ADT/ilist_node_base.h:46:41
 #6 0x0000000001049a30 llvm::ilist_iterator<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>, false, false>::operator*() const /scratch/hidetou/circt/llvm/llvm/include/llvm/ADT/ilist_iterator.h:138:5
 #7 0x0000000001049a30 mlir::SymbolTable::lookupSymbolIn(mlir::Operation*, mlir::StringAttr) /scratch/hidetou/circt/llvm/mlir/lib/IR/SymbolTable.cpp:325:17
 #8 0x0000000000ce4823 circt::hw::instance_like_impl::getReferencedModule(circt::hw::HWSymbolCache const*, mlir::Operation*, mlir::FlatSymbolRefAttr) /scratch/hidetou/circt/lib/Dialect/HW/InstanceImplementation.cpp:34:1
 #9 0x0000000000e950c9 inlineAndEraseIfEmpty(circt::hw::InstanceOp) /scratch/hidetou/circt/lib/Conversion/PipelineToHW/PipelineToHW.cpp:39:42
#10 0x0000000000e950c9 PipelineOutlineLowering::run() /scratch/hidetou/circt/lib/Conversion/PipelineToHW/PipelineToHW.cpp:565:5
#11 0x0000000000e93b2a (anonymous namespace)::PipelineToHWPass::runOnOperation() /scratch/hidetou/circt/lib/Conversion/PipelineToHW/PipelineToHW.cpp:872:23
#12 0x0000000001540b26 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int)::$_7::operator()() const /scratch/hidetou/circt/llvm/mlir/lib/Pass/Pass.cpp:0:17
#13 0x0000000001540b26 void llvm::function_ref<void ()>::callback_fn<mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int)::$_7>(long) /scratch/hidetou/circt/llvm/llvm/include/llvm/ADT/STLFunctionalExtras.h:45:12
#14 0x0000000001540b26 llvm::function_ref<void ()>::operator()() const /scratch/hidetou/circt/llvm/llvm/include/llvm/ADT/STLFunctionalExtras.h:68:12
#15 0x0000000001540b26 void mlir::MLIRContext::executeAction<mlir::PassExecutionAction, mlir::Pass&>(llvm::function_ref<void ()>, llvm::ArrayRef<mlir::IRUnit>, mlir::Pass&) /scratch/hidetou/circt/llvm/mlir/include/mlir/IR/MLIRContext.h:275:7
#16 0x0000000001540b26 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) /scratch/hidetou/circt/llvm/mlir/lib/Pass/Pass.cpp:479:21
#17 0x00000000015412f0 mlir::LogicalResult::failed() const /scratch/hidetou/circt/llvm/mlir/include/mlir/Support/LogicalResult.h:44:33
#18 0x00000000015412f0 mlir::failed(mlir::LogicalResult) /scratch/hidetou/circt/llvm/mlir/include/mlir/Support/LogicalResult.h:72:58
#19 0x00000000015412f0 mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) /scratch/hidetou/circt/llvm/mlir/lib/Pass/Pass.cpp:551:9
#20 0x0000000001546ca0 mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_15::operator()(mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo&) const /scratch/hidetou/circt/llvm/mlir/lib/Pass/Pass.cpp:776:5
#21 0x0000000001546d8f mlir::LogicalResult::failed() const /scratch/hidetou/circt/llvm/mlir/include/mlir/Support/LogicalResult.h:44:33
#22 0x0000000001546d8f mlir::failed(mlir::LogicalResult) /scratch/hidetou/circt/llvm/mlir/include/mlir/Support/LogicalResult.h:72:58
#23 0x0000000001546d8f mlir::LogicalResult mlir::failableParallelForEach<__gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_15&>(mlir::MLIRContext*, __gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>>, __gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_15&)::'lambda'()::operator()() const /scratch/hidetou/circt/llvm/mlir/include/mlir/IR/Threading.h:62:11
#24 0x0000000001546d8f __gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>> std::__invoke_impl<void, mlir::LogicalResult mlir::failableParallelForEach<__gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_15&>(mlir::MLIRContext*, __gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>>, __gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_15&)::'lambda'()&>(std::__invoke_other, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_15&) /sifive/tools/gcc/gcc/11.2.0/lib/gcc/x86_64-pc-linux-gnu/11.2.0/../../../../include/c++/11.2.0/bits/invoke.h:61:14
#25 0x0000000001546d8f std::enable_if<is_invocable_r_v<__gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_15&>, __gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>>>::type std::__invoke_r<void, mlir::LogicalResult mlir::failableParallelForEach<__gnu_cxx::__normal_iterator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo*, std::vector<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo, std::allocator<mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo>>>, mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_15&>(mlir::MLIRContext*, __gnu_cxx::_ [... cut ...]
#37 0x0000000001682b15 void llvm::thread::GenericThreadProxy<std::tuple<llvm::ThreadPool::grow(int)::$_0>>(void*) /scratch/hidetou/circt/llvm/llvm/include/llvm/Support/thread.h:46:3
#38 0x0000000001682b15 void* llvm::thread::ThreadProxy<std::tuple<llvm::ThreadPool::grow(int)::$_0>>(void*) /scratch/hidetou/circt/llvm/llvm/include/llvm/Support/thread.h:55:5
#39 0x00007f15c6e2f1ca start_thread (/lib64/libpthread.so.0+0x81ca)
#40 0x00007f15c5caee73 clone (/lib64/libc.so.6+0x39e73)
FileCheck error: '<stdin>' is empty.
FileCheck command line:  /scratch/hidetou/circt/build/bin/FileCheck /scratch/hidetou/circt/test/Conversion/PipelineToHW/test_outlined.mlir
````

dtzSiFive, 2023-08-09:
Looks like we're adding new modules (and removing them) causing concurrent SymbolTable lookups to run into problems.  TSAN report coming shortly.

dtzSiFive, 2023-08-09:
<details>
<summary> Example TSAN report </summary>

```
AIL: CIRCT :: Conversion/PipelineToHW/test_outlined.mlir (36 of 641)
******************** TEST 'CIRCT :: Conversion/PipelineToHW/test_outlined.mlir' FAILED ********************
Script:
--
: 'RUN: at line 1';   /build/sifive/tsan/bin/circt-opt --lower-pipeline-to-hw="outline-stages" /home/will/src/sifive/circt/test/Conversion/PipelineToHW/test_outlined.mlir | /build/sifive/tsan/bin/FileCheck /home/will/src/sifive/circt/test/Conversion/PipelineToHW/test_outlined.mlir
--
Exit Code: 66

Command Output (stderr):
--
==================
WARNING: ThreadSanitizer: data race (pid=3539272)
  Write of size 8 at 0x7b1800006308 by thread T1:
    #0 llvm::ilist_node_base<true>::setNext(llvm::ilist_node_base<true>*) /home/will/src/sifive/circt/llvm/llvm/include/llvm/ADT/ilist_node_base.h:41:52 (circt-opt+0x24fe371)
    #1 llvm::ilist_base<true>::insertBeforeImpl(llvm::ilist_node_base<true>&, llvm::ilist_node_base<true>&) /home/will/src/sifive/circt/llvm/llvm/include/llvm/ADT/ilist_base.h:26:10 (circt-opt+0x24fe371)
    #2 void llvm::ilist_base<true>::insertBefore<llvm::ilist_node_impl<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>>>(llvm::ilist_node_impl<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>>&, llvm::ilist_node_impl<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>>&) /home/will/src/sifive/circt/llvm/llvm/include/llvm/ADT/ilist_base.h:77:5 (circt-opt+0x24fe371)
    #3 llvm::simple_ilist<mlir::Operation>::insert(llvm::ilist_iterator<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>, false, false>, mlir::Operation&) /home/will/src/sifive/circt/llvm/llvm/include/llvm/ADT/simple_ilist.h:158:5 (circt-opt+0x24fe371)
    #4 llvm::iplist_impl<llvm::simple_ilist<mlir::Operation>, llvm::ilist_traits<mlir::Operation>>::insert(llvm::ilist_iterator<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>, false, false>, mlir::Operation*) /home/will/src/sifive/circt/llvm/llvm/include/llvm/ADT/ilist.h:167:28 (circt-opt+0x24fe371)
    #5 mlir::OpBuilder::insert(mlir::Operation*) /home/will/src/sifive/circt/llvm/mlir/lib/IR/Builders.cpp:412:28 (circt-opt+0x24fe371)
    #6 mlir::OpBuilder::create(mlir::OperationState const&) /home/will/src/sifive/circt/llvm/mlir/lib/IR/Builders.cpp:447:10 (circt-opt+0x24fe6cb)
    #7 circt::hw::HWModuleOp mlir::OpBuilder::create<circt::hw::HWModuleOp, mlir::StringAttr, llvm::SmallVector<circt::hw::PortInfo, 1u>&>(mlir::Location, mlir::StringAttr&&, llvm::SmallVector<circt::hw::PortInfo, 1u>&) /home/will/src/sifive/circt/llvm/llvm/../mlir/include/mlir/IR/Builders.h:495:16 (circt-opt+0x2291c94)
    #8 PipelineOutlineLowering::buildPipelineLike(llvm::Twine, mlir::TypeRange, mlir::ValueRange, mlir::TypeRange, bool, llvm::ArrayRef<mlir::Attribute>, llvm::ArrayRef<mlir::Attribute>) /home/will/src/sifive/circt/lib/Conversion/PipelineToHW/PipelineToHW.cpp:754:20 (circt-opt+0x2290d08)
    #9 PipelineOutlineLowering::buildStage(mlir::Block*, PipelineLowering::StageArgs, unsigned long, llvm::ArrayRef<mlir::Attribute>, PipelineLowering::StageEgressNames&) /home/will/src/sifive/circt/lib/Conversion/PipelineToHW/PipelineToHW.cpp:782:26 (circt-opt+0x228a5f0)
    #10 PipelineOutlineLowering::lowerStage(mlir::Block*, PipelineLowering::StageArgs, unsigned long, llvm::ArrayRef<mlir::Attribute>) /home/will/src/sifive/circt/lib/Conversion/PipelineToHW/PipelineToHW.cpp:587:9 (circt-opt+0x2288e6b)
    #11 PipelineOutlineLowering::lowerStage(mlir::Block*, PipelineLowering::StageArgs, unsigned long, llvm::ArrayRef<mlir::Attribute>) /home/will/src/sifive/circt/lib/Conversion/PipelineToHW/PipelineToHW.cpp:645:14 (circt-opt+0x2289867)
    #12 PipelineOutlineLowering::run() /home/will/src/sifive/circt/lib/Conversion/PipelineToHW/PipelineToHW.cpp:548:40 (circt-opt+0x2287481)
    #13 (anonymous namespace)::PipelineToHWPass::runOnOperation() /home/will/src/sifive/circt/lib/Conversion/PipelineToHW/PipelineToHW.cpp:872:23 (circt-opt+0x2285338)
    #14 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int)::$_1::operator()() const /home/will/src/sifive/circt/llvm/mlir/lib/Pass/Pass.cpp:485:17 (circt-opt+0x369eb41)
    #15 void llvm::function_ref<void ()>::callback_fn<mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int)::$_1>(long) /home/will/src/sifive/circt/llvm/llvm/include/llvm/ADT/STLFunctionalExtras.h:45:12 (circt-opt+0x369eb41)
    #16 llvm::function_ref<void ()>::operator()() const /home/will/src/sifive/circt/llvm/llvm/include/llvm/ADT/STLFunctionalExtras.h:68:12 (circt-opt+0x369eb41)
    #17 void mlir::MLIRContext::executeAction<mlir::PassExecutionAction, mlir::Pass&>(llvm::function_ref<void ()>, llvm::ArrayRef<mlir::IRUnit>, mlir::Pass&) /home/will/src/sifive/circt/llvm/mlir/include/mlir/IR/MLIRContext.h:275:7 (circt-opt+0x369eb41)
    #18 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) /home/will/src/sifive/circt/llvm/mlir/lib/Pass/Pass.cpp:479:21 (circt-opt+0x369eb41)
    #19 mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) /home/will/src/sifive/circt/llvm/mlir/lib/Pass/Pass.cpp:551:16 (circt-opt+0x369f8a2)
[... cut ...]
    #31 mlir::MlirOptMain(int, char**, llvm::StringRef, mlir::DialectRegistry&) /home/will/src/sifive/circt/llvm/mlir/lib/Tools/mlir-opt/MlirOptMain.cpp:531:14 (circt-opt+0x2b6719c)
    #32 main /home/will/src/sifive/circt/tools/circt-opt/circt-opt.cpp:74:23 (circt-opt+0xef42dc)

SUMMARY: ThreadSanitizer: data race /home/will/src/sifive/circt/llvm/llvm/include/llvm/ADT/ilist_node_base.h:41:52 in llvm::ilist_node_base<true>::setNext(llvm::ilist_node_base<true>*)
==================
```
</details>

mortbopet, 2023-08-09:
Thank you for the report, will look into this ASAP.

PR #5817 (2023-08-09): [PipelineToHW] Fix `PipelineToHW` TSAN issue
Fixing a TSAN issue due to concurrent additions (and removals) to the top-level mlir::ModuleOp symbol table, whilst the pass is running at a hw::HWModuleOp granularity.

fixes #5815
closed 2023-08-09 by mortbopet