llvm/circt #5827: [CI] Z3 version too old for use w/circt-lec

dtzSiFive, 2023-08-10:
CI builds have this warning:

```
  Cannot build circt-lec with outdated Z3 version 4.8.7, requires 4.8.11.
```

(example: https://github.com/llvm/circt/actions/runs/5825770350/job/15798151008 )

That's awfully close, can we bump/provide a suitable Z3 so this can be built + tested as part of those flows?

Not sure if this is a GH runner environment or docker container thing, but flagging in case it's something we can get fixed without too much trouble :+1: .

teqdruid, 2023-08-19:
It's a container thing. Do we have any normal tests which test the lec functionality? I only see tests in the integration_tests directory. The integration test image does have a newer version of z3 so circt-lec is compiled (and thus is checked by the short integration tests in each PR).

dtzSiFive, 2023-09-09:
Oh okay! Thanks, that sounds like all is well!

closed 2023-09-09 by dtzSiFive