llvm/circt #5855: Add more integration tests for Calyx `hlstool` flow

rachitnigam, 2023-08-16:
#5847 added the Calyx-based compilation flow to `hlstool`. We should start adding more integration tests for Calyx so that we have long-term confidence in what the various (scf/affine/loopschedule)->calyx flows can and cannot do. A reasonable place to start could be the [`handshake` integration tests](https://github.com/llvm/circt/tree/main/integration_test/Dialect/Handshake). To start us off, we probably want tests for:

- [ ] `memref` argument lowering
- [ ] Nested loops in `scf`
- [ ] Function calls and multi-component designs

rachitnigam, 2023-08-16:
@linuxlonelyeagle might be interested in the third one here: we probably want to add support for the call operation when lowering multi-function `scf` programs and use the newly added `invoke` statement.

linuxlonelyeagle, 2023-08-17:
Thanks, I'll try to do `Function calls and multi-component designs`, but I need to familiarize myself with the code.