llvm/circt #5861: Building Circt

Asghar-Raza, 2023-08-16:
Hi, I am new to opensource. I wanted to know what happens when we build LLVM and Circt ? And can Circt can be used without compiling and building?

rachitnigam, 2023-08-19:
Depends on what tools you want to use. If you just want to call the binary on some MLIR code, you can use the prebuilt binaries here: https://github.com/llvm/circt/actions/runs/5904717038

rachitnigam, 2023-08-19:
@Asghar-Raza if this answers your question, feel free to close this issue

Asghar-Raza, 2023-08-19:
I want to be able to use dialect transformations and passes by inheriting classes. From what I understand, building CIRCT will only produce binaries and I can use said binaries from the commandline. I can't import classes or functions for the transformations and passes. Is this correct or am I misunderstanding it?

rachitnigam, 2023-08-20:
Ah, in that case you need to build it from source as far as I know. For example, you can take a look at [`hlstool`][hlstool] which uses MLIR and CIRCT dialects and conversion passes.

[hlstool]: https://github.com/llvm/circt/tree/main/tools/hlstool

closed 2023-08-23 by Asghar-Raza