llvm/circt #5864: [ExportVerilog] Inconsistent behavior with disallowPackedStructAssignments

leonardt, 2023-08-16:
This MLIR
```
module attributes {circt.loweringOptions = "locationInfoStyle=none,disallowPackedStructAssignments"} {
    hw.module @Foo(%I: !hw.struct<x: i8, y: i1>, %S: i1) -> (O: !hw.struct<x: i8, y: i1>) {
        %0 = hw.struct_extract %I["x"] : !hw.struct<x: i8, y: i1>
        %2 = hw.constant -1 : i8
        %1 = comb.xor %2, %0 : i8
        %3 = hw.struct_extract %I["y"] : !hw.struct<x: i8, y: i1>
        %5 = hw.constant -1 : i1
        %4 = comb.xor %5, %3 : i1
        %6 = hw.struct_create (%1, %4) : !hw.struct<x: i8, y: i1>
        hw.output %6 : !hw.struct<x: i8, y: i1>
    }
}
```
correctly produces:
```
// Generated by CIRCT firtool-1.48.0-34-g7018fb13b
module Foo(
  input  struct packed {logic [7:0] x; logic y; } I,
  input                                           S,
  output struct packed {logic [7:0] x; logic y; } O
);

  wire struct packed {logic [7:0] x; logic y; } _GEN;
  assign _GEN.x = ~I.x;
  assign _GEN.y = ~I.y;
  assign O = _GEN;
endmodule

```

However this MLIR
```
module attributes {circt.loweringOptions = "locationInfoStyle=none,disallowPackedStructAssignments"} {
    hw.module @Foo(%I: !hw.struct<x: i8, y: i1>, %S: i1) -> (O: !hw.struct<x: i8, y: i1>) {
        %0 = hw.constant 190 : i8
        %1 = hw.constant 1 : i1
        %2 = hw.struct_create (%0, %1) : !hw.struct<x: i8, y: i1>
        hw.output %2 : !hw.struct<x: i8, y: i1>
    }
}
```
creates a packed assignment
```
// Generated by CIRCT firtool-1.48.0-34-g7018fb13b
module Foo(
  input  struct packed {logic [7:0] x; logic y; } I,
  input                                           S,
  output struct packed {logic [7:0] x; logic y; } O
);

  assign O = '{x: 8'hBE, y: 1'h1};
endmodule

```

leonardt, 2023-08-16:
It seems that the code may be correctly generating the unpacked assignments, but a subsequent pass is removing the temporary variables.  I inserted a named wire to try to prevent this and it seems to work okay:
```
module attributes {circt.loweringOptions = "locationInfoStyle=none,disallowPackedStructAssignments"} {
    hw.module @Foo(%I: !hw.struct<x: i8, y: i1>, %S: i1) -> (O: !hw.struct<x: i8, y: i1>) {
        %0 = hw.constant 190 : i8
        %2 = sv.wire sym @Foo.b name "b" : !hw.inout<i8>
        sv.assign %2, %0 : i8
        %1 = sv.read_inout %2 : !hw.inout<i8>
        %3 = hw.constant 1 : i1
        %5 = sv.wire sym @Foo.c name "c" : !hw.inout<i1>
        sv.assign %5, %3 : i1
        %4 = sv.read_inout %5 : !hw.inout<i1>
        %6 = hw.struct_create (%1, %4) : !hw.struct<x: i8, y: i1>
        hw.output %6 : !hw.struct<x: i8, y: i1>
    }
}
```
produces
```
// Generated by CIRCT firtool-1.48.0-34-g7018fb13b
module Foo(
  input  struct packed {logic [7:0] x; logic y; } I,
  input                                           S,
  output struct packed {logic [7:0] x; logic y; } O
);

  wire [7:0]                                    b = 8'hBE;
  wire                                          c = 1'h1;
  wire struct packed {logic [7:0] x; logic y; } _GEN;
  assign _GEN.x = b;
  assign _GEN.y = c;
  assign O = _GEN;
endmodule

```

leonardt, 2023-08-16:
CC @mortbopet who originally work on https://github.com/llvm/circt/pull/5114

mortbopet, 2023-08-17:
Would a simple fix to this be to add names to the generated wires/regs [here](https://github.com/llvm/circt/pull/5114/files#diff-dad08144b49b782e0f156ff2d6360d071aa71e11ef4023987213a69ddd452e0aR902-R904) similarly to what you describe above?

leonardt, 2023-08-17:
Adding a name seems like a decent quick fix, but I'd imagine the better longterm solution might be to figure out which pass is optimizing the assignments out and make it aware of this config flag.

leonardt, 2023-08-17:
(FWIW I'd be okay with the quick fix if it corrects the above issue, since we're needing this flag for tool compatibility and it's blocking)