llvm/circt #5902: [FIRRTL] LowerClass pass assertion failure in tests

uenoku, 2023-08-22:
```
/scratch/hidetou/circt/build_debug/bin/circt-opt -firrtl-lower-classes /scratch/hidetou/circt/test/Dialect/FIRRTL/lower-classes.mlir
circt-opt: /scratch/hidetou/circt/llvm/llvm/include/llvm/ADT/ilist_iterator.h:138: llvm::ilist_iterator::reference llvm::ilist_iterator<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>, false, false>::operator*() const [OptionsT = llvm::ilist_detail::node_options<mlir::Operation, true, false, void>, IsReverse = false, IsConst = false]: Assertion `!NodePtr->isKnownSentinel()' failed.
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
Stack dump:
0.      Program arguments: /scratch/hidetou/circt/build_debug/bin/circt-opt -firrtl-lower-classes /scratch/hidetou/circt/test/Dialect/FIRRTL/lower-classes.mlir
 #0 0x000000000052dd4d llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) /scratch/hidetou/circt/llvm/llvm/lib/Support/Unix/Signals.inc:602:11
 #1 0x000000000052e1eb PrintStackTraceSignalHandler(void*) /scratch/hidetou/circt/llvm/llvm/lib/Support/Unix/Signals.inc:675:1
 #2 0x000000000052c466 llvm::sys::RunSignalHandlers() /scratch/hidetou/circt/llvm/llvm/lib/Support/Signals.cpp:104:5
 #3 0x000000000052ea05 SignalHandler(int) /scratch/hidetou/circt/llvm/llvm/lib/Support/Unix/Signals.inc:413:1
 #4 0x00007f87cbec0cf0 __restore_rt (/lib64/libpthread.so.0+0x12cf0)
 #5 0x00007f87cad4baff raise (/lib64/libc.so.6+0x4eaff)
 #6 0x00007f87cad1eea5 abort (/lib64/libc.so.6+0x21ea5)
 #7 0x00007f87cad1ed79 _nl_load_domain.cold.0 (/lib64/libc.so.6+0x21d79)
 #8 0x00007f87cad44456 (/lib64/libc.so.6+0x47456)
 #9 0x00000000005988cf llvm::ilist_iterator<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>, false, false>::operator*() const /scratch/hidetou/circt/llvm/llvm/include/llvm/ADT/ilist_iterator.h:0:5
#10 0x0000000000598819 mlir::Region::OpIterator::operator*() const /scratch/hidetou/circt/llvm/llvm/../mlir/include/mlir/IR/Region.h:144:36
#11 0x000000000059878a llvm::filter_iterator_base<mlir::Region::OpIterator, bool (*)(mlir::Operation&), std::forward_iterator_tag>::findNextValid() /scratch/hidetou/circt/llvm/llvm/include/llvm/ADT/STLExtras.h:468:36
#12 0x00000000005986e9 llvm::filter_iterator_base<mlir::Region::OpIterator, bool (*)(mlir::Operation&), std::forward_iterator_tag>::filter_iterator_base(mlir::Region::OpIterator, mlir::Region::OpIterator, bool (*)(mlir::Operation&)) /scratch/hidetou/circt/llvm/llvm/include/llvm/ADT/STLExtras.h:481:3
#13 0x00000000005983ee llvm::filter_iterator_impl<mlir::Region::OpIterator, bool (*)(mlir::Operation&), std::forward_iterator_tag>::filter_iterator_impl(mlir::Region::OpIterator, mlir::Region::OpIterator, bool (*)(mlir::Operation&)) /scratch/hidetou/circt/llvm/llvm/include/llvm/ADT/STLExtras.h:513:71
#14 0x00000000011fb4dd mlir::detail::op_filter_iterator<circt::firrtl::FModuleOp, mlir::Region::OpIterator>::op_filter_iterator(mlir::Region::OpIterator, mlir::Region::OpIterator) /scratch/hidetou/circt/llvm/llvm/../mlir/include/mlir/IR/BlockSupport.h:148:75
#15 0x00000000011fb2e2 llvm::iterator_range<mlir::detail::op_iterator<circt::firrtl::FModuleOp, mlir::Region::OpIterator>> mlir::Region::getOps<circt::firrtl::FModuleOp>() /scratch/hidetou/circt/llvm/llvm/../mlir/include/mlir/IR/Region.h:187:13
#16 0x00000000011dd689 auto mlir::OpTrait::OneRegion<circt::firrtl::CircuitOp>::getOps<circt::firrtl::FModuleOp>() /scratch/hidetou/circt/llvm/llvm/../mlir/include/mlir/IR/OpDefinition.h:549:33
#17 0x0000000001356e9e (anonymous namespace)::LowerClassesPass::runOnOperation() /scratch/hidetou/circt/lib/Dialect/FIRRTL/Transforms/LowerClasses.cpp:95:18
#18 0x0000000003720dfb mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int)::$_7::operator()() const /scratch/hidetou/circt/llvm/mlir/lib/Pass/Pass.cpp:0:17
#19 0x0000000003720d95 void llvm::function_ref<void ()>::callback_fn<mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int)::$_7>(long) /scratch/hidetou/circt/llvm/llvm/include/llvm/ADT/STLFunctionalExtras.h:45:5
#20 0x00000000004797e9 llvm::function_ref<void ()>::operator()() const /scratch/hidetou/circt/llvm/llvm/include/llvm/ADT/STLFunctionalExtras.h:68:5
#21 0x0000000003723e8d void mlir::MLIRContext::executeAction<mlir::PassExecutionAction, mlir::Pass&>(llvm::function_ref<void ()>, llvm::ArrayRef<mlir::IRUnit>, mlir::Pass&) /scratch/hidetou/circt/llvm/mlir/include/mlir/IR/MLIRContext.h:276:3
#22 0x000000000371c3e3 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) /scratch/hidetou/circt/llvm/mlir/lib/Pass/Pass.cpp:491:17
#23 0x000000000371c994 mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) /scratch/hidetou/circt/llvm/mlir/lib/Pass/Pass.cpp:551:16
#24 0x000000000371e40c mlir::PassManager::runPasses(mlir::Operation*, mlir::AnalysisManager) /scratch/hidetou/circt/llvm/mlir/lib/Pass/Pass.cpp:860:10
#25 0x000000000371e322 mlir::PassManager::run(mlir::Operation*) /scratch/hidetou/circt/llvm/mlir/lib/Pass/Pass.cpp:840:60
#26 0x00000000029b1752 performActions(llvm::raw_ostream&, std::shared_ptr<llvm::SourceMgr> const&, mlir::MLIRContext*, mlir::MlirOptMainConfig const&) /scratch/hidetou/circt/llvm/mlir/lib/Tools/mlir-opt/MlirOptMain.cpp:376:17
#27 0x00000000029b1389 processBuffer(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, mlir::MlirOptMainConfig const&, mlir::DialectRegistry&, llvm::ThreadPool*) /scratch/hidetou/circt/llvm/mlir/lib/Tools/mlir-opt/MlirOptMain.cpp:431:12
#28 0x00000000029b117b mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, mlir::DialectRegistry&, mlir::MlirOptMainConfig const&)::$_1::operator()(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&) const /scratch/hidetou/circt/llvm/mlir/lib/Tools/mlir-opt/MlirOptMain.cpp:472:12
#29 0x00000000029b10fd mlir::LogicalResult llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>::callback_fn<mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, mlir::DialectRegistry&, mlir::MlirOptMainConfig const&)::$_1>(long, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&) /scratch/hidetou/circt/llvm/llvm/include/llvm/ADT/STLFunctionalExtras.h:45:12
#30 0x0000000002a00309 llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>::operator()(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&) const /scratch/hidetou/circt/llvm/llvm/include/llvm/ADT/STLFunctionalExtras.h:68:12
#31 0x00000000029ff8e5 mlir::splitAndProcessBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>, llvm::raw_ostream&, bool, bool) /scratch/hidetou/circt/llvm/mlir/lib/Support/ToolUtilities.cpp:28:12
#32 0x00000000029ae119 mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, mlir::DialectRegistry&, mlir::MlirOptMainConfig const&) /scratch/hidetou/circt/llvm/mlir/lib/Tools/mlir-opt/MlirOptMain.cpp:475:10
#33 0x00000000029ae5ee mlir::MlirOptMain(int, char**, llvm::StringRef, mlir::DialectRegistry&) /scratch/hidetou/circt/llvm/mlir/lib/Tools/mlir-opt/MlirOptMain.cpp:531:14
#34 0x000000000040ccc6 main /scratch/hidetou/circt/tools/circt-opt/circt-opt.cpp:74:23
#35 0x00007f87cad37d85 __libc_start_main (/lib64/libc.so.6+0x3ad85)
#36 0x000000000040caee _start (/scratch/hidetou/circt/build_debug/bin/circt-opt+0x40caee)
```

trilorez, 2023-08-22:
Is it safe to erase the class op [here](https://github.com/llvm/circt/blob/main/lib/Dialect/FIRRTL/Transforms/LowerClasses.cpp#L226) in the middle of [this parallel for each](https://github.com/llvm/circt/blob/main/lib/Dialect/FIRRTL/Transforms/LowerClasses.cpp#L90)?

uenoku, 2023-08-23:
It seems to be fixed on my end, thanks!

closed 2023-08-23 by uenoku