llvm/circt #5908: [FIRRTL] Failing test in nightlies

teqdruid, 2023-08-22:
```
******************** TEST 'CIRCT :: Dialect/FIRRTL/lower-classes.mlir' FAILED ********************
Script:
--
: 'RUN: at line 1';   /__w/circt/circt/build/bin/circt-opt -firrtl-lower-classes /__w/circt/circt/test/Dialect/FIRRTL/lower-classes.mlir | /__w/circt/circt/build/bin/FileCheck /__w/circt/circt/test/Dialect/FIRRTL/lower-classes.mlir
--
Exit Code: 2

Command Output (stderr):
--
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
Stack dump:
0.	Program arguments: /__w/circt/circt/build/bin/circt-opt -firrtl-lower-classes /__w/circt/circt/test/Dialect/FIRRTL/lower-classes.mlir
Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
0  libLLVMSupport.so           0x00007f19d6c00747 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) + 39
1  libLLVMSupport.so           0x00007f19d6bfe57e llvm::sys::RunSignalHandlers() + 238
2  libLLVMSupport.so           0x00007f19d6c00e0f
3  libpthread.so.0             0x00007f19dbab7420
4  libCIRCTFIRRTLTransforms.so 0x00007f19db718bb0
5  libCIRCTFIRRTLTransforms.so 0x00007f19db7c71c2
6  libMLIRPass.so              0x00007f19d71aef58 mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) + 648
7  libMLIRPass.so              0x00007f19d71b2b6a mlir::PassManager::run(mlir::Operation*) + 1194
8  libMLIROptLib.so            0x00007f19db2a9f45
9  libMLIROptLib.so            0x00007f19db2a90e7
10 libMLIRSupport.so           0x00007f19d6ca0885 mlir::splitAndProcessBuffer(std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::function_ref<mlir::LogicalResult (std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, llvm::raw_ostream&)>, llvm::raw_ostream&, bool, bool) + 853
11 libMLIROptLib.so            0x00007f19db2a439a mlir::MlirOptMain(llvm::raw_ostream&, std::unique_ptr<llvm::MemoryBuffer, std::default_delete<llvm::MemoryBuffer>>, mlir::DialectRegistry&, mlir::MlirOptMainConfig const&) + 362
12 libMLIROptLib.so            0x00007f19db2a4817 mlir::MlirOptMain(int, char**, llvm::StringRef, mlir::DialectRegistry&) + 1095
13 circt-opt                   0x000000000020d142
14 libc.so.6                   0x00007f19d6529083 __libc_start_main + 243
15 circt-opt                   0x000000000020cb8e
FileCheck error: '<stdin>' is empty.
FileCheck command line:  /__w/circt/circt/build/bin/FileCheck /__w/circt/circt/test/Dialect/FIRRTL/lower-classes.mlir

--

********************
```

https://github.com/llvm/circt/actions/runs/5925927698/job/16066338429
https://github.com/llvm/circt/actions/runs/5938458724/job/16102953152

This test failed the last two nights in different configurations. Smells like a threading bug.

dtzSiFive, 2023-08-22:
cc #5902 , which is hopefully/probably the same?

I expect we are mutating the operation list while iterating over it in an unsafe way, but haven't confirmed as still rebuilding from the LLVM bump earlier (partly because MLIR broke it's linking, so had to patch around that..).  Anyway, hopefully same fix! Thanks for reporting!

dtzSiFive, 2023-08-23:
Issue is fixed, and last night's tests just passed! :tada: https://github.com/llvm/circt/actions/runs/5950993774

closed 2023-08-23 by dtzSiFive