llvm/circt #5951: ninja: error: 'tools/esi/EsiCosimDpiServer', needed by 'tools/esi/CMakeFiles/esi-cosim-runner', missing and no known rule to make itninja: error: 'tools/esi/EsiCosimDpiServer', needed by 'tools/esi/CMakeFiles/esi-cosim-runner', missing and no known rule to make it

linuxlonelyeagle, 2023-08-25:
I would like to add the new integration tests. but I seem to be having problems.
```
$ ninja check-circt                                                                                                       (base) 
Running the CIRCT regression tests
llvm-lit: /home/sen/circt/llvm/llvm/utils/lit/lit/llvm/subst.py:126: note: Did not find esi-tester in /home/sen/circt/build/bin::/home/sen/circt/llvm/build/./bin

$ninja check-circt-integration                                                                                           (base) 
ninja: error: 'tools/esi/EsiCosimDpiServer', needed by 'tools/esi/CMakeFiles/esi-cosim-runner', missing and no known rule to make it
```
l'm not sure what the problem is.

teqdruid, 2023-08-25:
Shit. That's my bad. Didn't realize that esi-cosim-tester was required by the integration tests. #5952 _should_ fix it, but I'm not in a position to test it right now. Will merge if the CI tests pass since I'm relatively certain it'll fix the issue.

closed 2023-08-25 by mortbopet

teqdruid, 2023-08-26:
@linuxlonelyeagle Did that fix the bug?

linuxlonelyeagle, 2023-08-26:
> @linuxlonelyeagle Did that fix the bug?

Yes.Thanks!

PR #5952 (2023-08-25): [ESI] Fix non-capnp builds
Should fix #5951.
commit 7dd309f6 (2023-08-25): [ESI] Fix non-capnp builds (#5952)

Should fix #5951.