llvm/circt #6029: [FIRRTL] LowerClasses Support ExtModules

seldridge, 2023-09-01:
Currently, FIRRTL external modules with either input or output property ports error during `LowerClasses`.

I haven't thought through exactly how this should work in terms of lowering. At a high level, this needs to work such that two OMIR dialect blobs are produced and then these are combined by the OMIR linker.

### Input Ports

```
FIRRTL version 4.0.0
circuit Foo:
  extmodule Bar:
    input a: Integer

  module Foo:
    inst bar of Bar
    propassign bar.a, Integer(42)
```

```
# firtool firrtl-snippets/ExtModuleInputProperty.fir
Assertion failed: (Index < this->size() && "Invalid index!"), function operator[], file ArrayRef.h, line 443.
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
Stack dump:
0.	Program arguments: firtool firrtl-snippets/ExtModuleInputProperty.fir
Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
0  firtool                  0x00000001031ee300 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) + 88
1  firtool                  0x00000001031ee91c PrintStackTraceSignalHandler(void*) + 28
2  firtool                  0x00000001031ec4b0 llvm::sys::RunSignalHandlers() + 152
3  firtool                  0x00000001031efbf4 SignalHandler(int) + 276
4  libsystem_platform.dylib 0x0000000190b42a24 _sigtramp + 56
5  libsystem_pthread.dylib  0x0000000190b13c28 pthread_kill + 288
6  libsystem_c.dylib        0x0000000190a21ae8 abort + 180
7  libsystem_c.dylib        0x0000000190a20e44 err + 0
8  firtool                  0x00000001038a6338 llvm::MutableArrayRef<mlir::BlockArgument>::operator[](unsigned long) const + 124
9  firtool                  0x00000001038a619c mlir::Region::getArgument(unsigned int) + 56
10 firtool                  0x0000000103530344 (anonymous namespace)::LowerClassesPass::lowerClass((anonymous namespace)::ClassLoweringState&) + 604
```

### Output Ports

```
FIRRTL version 4.0.0
circuit Foo:
  extmodule Bar:
    output a: Integer

  module Foo:
    output a: Integer
    inst bar of Bar
    propassign a, bar.a
```

```
# firtool firrtl-snippets/ExtModuleOutputProperty.fir                         
firrtl-snippets/ExtModuleOutputProperty.fir:8:5: error: failed to legalize operation 'firrtl.instance' that was explicitly marked illegal
    inst bar of Bar
    ^
firrtl-snippets/ExtModuleOutputProperty.fir:8:5: note: see current operation: %0 = "firrtl.instance"() {annotations = [], moduleName = @Bar, name = "bar", nameKind = #firrtl<name_kind droppable_name>, portAnnotations = [[]], portDirections = true, portNames = ["a"]} : () -> !firrtl.integer
```

trilorez, 2023-09-01:
Can this be handled by lowering to `om::ClassExternOp` similar to handling `firrtl::ExtClassOp` in #5954?

seldridge, 2023-09-01:
Yes, I think that makes sense.

dtzSiFive, 2023-09-01:
Test snippet from earlier today that seems related and produces different error:
```firrtl
FIRRTL version 3.2.0

circuit Ext:
  extmodule Test:
    output p : String

  module Ext:
    inst t of Test
```

Error:

```
extmodule-sym.fir:4:3: error: redefinition of symbol named 'Test'
  extmodule Test:
  ^
extmodule-sym.fir:4:3: note: see current operation: 
"om.class"() ({
^bb0:
}) {formalParamNames = [], sym_name = "Test"} : () -> ()
extmodule-sym.fir:4:3: note: see existing symbol definition here
```

For whoever works on this, probably also should be sure to consider `defname` although that depends on how this exactly ends up tying together.

mikeurbach, 2023-09-05:
> Can this be handled by lowering to om::ClassExternOp similar to handling 

+1 I think this should be added. This support was just never added, and the fact it's crapping out with a bad error is lack of defensive coding on my part.

closed 2023-09-08 by mikeurbach