llvm/circt #6039: [Seq] `seq.compreg` canonicalizations

mortbopet, 2023-09-04:
I'd like to see some canonicalizations for the generic `seq` register operations.

## `seq.compreg`

Should we inherit the canonicalizations currently defined for `FirRegOp`?
* Constant 0 reset: drop reset value and reset signal from the op
* Feedback register OR constant 1 clock: replace with the reset value if present, else replace with constant zero
* Constant 1 reset: replace with reset value
* Constant 0 clock: replace with reset value if present, else constant 0. 

## `seq.compreg.ce`
In case of constant clock-enable inputs, what would the following canoncalize to?
```mlir
%r = seq.compreg.ce %i, %clk, %ce, %rst, %rv : i32
```
* `%ce = hw.constant 1 : i1` trivially canonicalizes to `seq.compreg %i, %clk, %rst, %rv : i32`
* `%ce = hw.constant 0 : i1` should, if the last `seq.compreg` canonicalization is accepted, be identical to the constant 0 clock canonicalization.

mortbopet, 2023-09-04:
CC @teqdruid

teqdruid, 2023-09-05:
> I'd like to see some canonicalizations for the generic `seq` register operations.

I would too. I've been thinking about some canonicalizations also, but the problem with some of them is that on FPGAs, the power-up value is well-defined. Being conservative about this, only the first one for both compreg and ce are valid (I think). I think the last ones in each are good, but s/reset/initial/.