llvm/circt #6124: Cast struct to bits in an incorrect order in the emitted SV

poemonsense, 2023-09-14:
I'm using Chisel 6.0.0-M3 with firtool 1.54.0. The case is like this:

```scala
class MyTest extends Bundle {
  val a0 = UInt(8.W)
  val a1 = UInt(8.W)
  val a2 = UInt(8.W)
  val a3 = UInt(8.W)
  val a4 = UInt(8.W)
}

class Test extends Module {
  val io = IO(new Bundle {
    val in = Input(new MyTest)
    val out = Output(new MyTest)
  })

  val q = Module(new Queue(new MyTest, 2))
  q.io.enq.valid := true.B
  q.io.enq.bits := io.in
  q.io.deq.ready := true.B
  io.out := q.io.deq.bits
    when (q.io.deq.fire) {
    printf("deq: %x\n", io.out.asUInt)
  }
  when (q.io.enq.fire) {
    printf("enq: %x\n", io.in.asUInt)
  }
}
```

That is, it instantiates a queue and keep enq/deq the Bundle.

Without `--preserve-aggregate=all`, the fields in the bundle are separated into multiple variables. The order of fields maintain consistent before and after the queue.

However, with `--preserve-aggregate=all`, the elements are packed as a struct as shown before:
```SystemVerilog
module Queue2_MyTest(   // <stdin>:3:3
  input                                                                                                   clock,    // <stdin>:4:11
                                                                                                          reset,    // <stdin>:5:11
  input  struct packed {logic [7:0] a0; logic [7:0] a1; logic [7:0] a2; logic [7:0] a3; logic [7:0] a4; } io_enq_bits,  // src/main/scala/chisel3/util/Decoupled.scala:256:14
  output                                                                                                  io_enq_ready, // src/main/scala/chisel3/util/Decoupled.scala:256:14
                                                                                                          io_deq_valid, // src/main/scala/chisel3/util/Decoupled.scala:256:14
  output struct packed {logic [7:0] a0; logic [7:0] a1; logic [7:0] a2; logic [7:0] a3; logic [7:0] a4; } io_deq_bits   // src/main/scala/chisel3/util/Decoupled.scala:256:14

// some code here...

  ram_2x40 ram_ext (    // src/main/scala/chisel3/util/Decoupled.scala:257:91
    .R0_addr (wrap_1),  // src/main/scala/chisel3/util/Counter.scala:61:40
    .R0_en   (1'h1),    // <stdin>:3:3
    .R0_clk  (clock),
    .W0_addr (wrap),    // src/main/scala/chisel3/util/Counter.scala:61:40
    .W0_en   (~full),   // src/main/scala/chisel3/util/Decoupled.scala:263:24, :287:19
    .W0_clk  (clock),
    .W0_data ({_GEN, _GEN_0, _GEN_1, _GEN_2, _GEN_3}),  // src/main/scala/chisel3/util/Decoupled.scala:257:91, :270:16, :271:24
    .W0_mask ({_GEN_4, _GEN_5, _GEN_6, _GEN_7, _GEN_8}),    // src/main/scala/chisel3/util/Decoupled.scala:257:91, :270:16, :271:24
    .R0_data (_ram_ext_R0_data)
  );
  assign io_enq_ready = ~full;  // <stdin>:3:3, src/main/scala/chisel3/util/Decoupled.scala:263:24, :287:19
  assign io_deq_valid = ~empty; // <stdin>:3:3, src/main/scala/chisel3/util/Decoupled.scala:262:25, :286:19
  assign io_deq_bits = /*cast(bit)*/_ram_ext_R0_data;   // <stdin>:3:3, src/main/scala/chisel3/util/Decoupled.scala:257:91
```

Note that, for W0_data, `_GEN` corresponds to `a4` and `_GEN_3` corresponds to `a0`. That is, the first member of the bundle/struct is at the least bits. However, for `io_deq_bits`, it casts the output bits to a struct.

In constrast, Verilator simulates this type cast with the first member at the most significant position. See https://github.com/verilator/verilator/issues/1191

> Section 7.2.1 indicates:
> "A packed structure can be used as a whole with arithmetic and logical operators.
> The first member specified is the most significant and subsequent members follow in decreasing significance."
> This means that within SystemVerilog, your packed structure will have it's first elements in the high bits,
> and the last element in the lowest bits. Think of it like the '{...}' concatenation operator in Verilog.

Therefore, the intention of CIRCT emitted SV and the semantics interpreted by Verilator varies. The above code is simulated with wrong outcomes.
```
enq: 000000a3f4
deq: f4a3000000
enq: 00000051fa
deq: fa51000000
enq: 00000028fd
deq: fd28000000
enq: 000000147e
deq: 7e14000000
enq: 0000008a3f
deq: 3f8a000000
enq: 000000451f
deq: 1f45000000
```

Again, without `--preserve-aggregate=all`, the code simulates good. I think it should be an ordering issue when CIRCT emits the SV.

To reproduce the case, use https://github.com/poemonsense/rocket-chip/tree/chisel-6.0.0-M3 with command `git submodule update --init`, `make emu -j16 NO_DIFF=1`, and `./build/emu -i /path/to/any/file -C 10`. I'm sorry for not providing a minimal repo but the test case is quite simple at `src/main/scala/system/FuzzHarness.scala`. I'm using `master` branch Verilator `Verilator 5.015 devel rev v5.014-89-gec2e3ec0e`.

uenoku, 2023-09-15:
Thanks! Yes, this seems like a bug probably in LowerTypes.

poemonsense, 2023-09-17:
Thanks for the confirmation. Is there any roadmap or estimated time for fix? Sorry but I didn't know how the CIRCT community works. If there's anything I should/could do, please feel free to let me know.

SpriteOvO, 2023-09-17:
I'll investigate the issue tomorrow and see if I can fix it.

PR #6339 (2023-10-25): [FIRRTL][LowerTypes] Keep the order of bundle fields in lowered `cat`
Fixes #6124.

The lowering for `bitcast` is initially introduced in PR #1949, for aggregate type

```scala
class MyTest extends Bundle {
  val a0 = UInt(8.W)
  val a1 = UInt(8.W)
  val a2 = UInt(8.W)
  val a3 = UInt(8.W)
  val a4 = UInt(8.W)
}
```

it will be lowered to

fir
```mlir
%8 = firrtl.subfield %ram_MPORT_0[data] : !firrtl.bundle<addr: uint<1>, en: uint<1>, clk: clock, data: bundle<a0: uint<8>, a1: uint<8>, a2: uint<8>, a3: uint<8>, a4: uint<8>>, mask: bundle<a0: uint<1>, a1: uint<1>, a2: uint<1>, a3: uint<1>, a4: u [... cut]