llvm/circt #6143: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>jtag"}

oharboe, 2023-09-17:
firtool fails to generate Verilog from attached .fir file. It was generated using latest Chipyard and running `cd vlsi && make buildfile tutorial=sky130-openroad CONFIG=MegaBoomConfig`.

Failure:

```
firtool \
        --format=fir \
        --dedup \
        --export-module-hierarchy \
        --emit-metadata \
        --verify-each=true \
        --warn-on-unprocessed-annotations \
        --disable-annotation-classless \
        --disable-annotation-unknown \
        --mlir-timing \
        --lowering-options=emittedLineLength=2048,noAlwaysComb,disallowLocalVariables,verifLabels,locationInfoStyle=wrapInAtSquareBracket,disallowPackedArrays \
        --repl-seq-mem \
        --repl-seq-mem-file=/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.mems.conf \
        --repl-seq-mem-circuit=TestHarness \
        --annotation-file=/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.anno.json \
        --split-verilog \
        -o /home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/gen-collateral \
        /home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: cannot find name 'debug_brs' in BoomCore
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|BoomCore>debug_brs"}
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: cannot find name 'debug_jals' in BoomCore
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|BoomCore>debug_jals"}
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: cannot find name 'debug_jalrs' in BoomCore
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|BoomCore>debug_jalrs"}
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: cannot find name 'serial_tl' in ChipTop
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>serial_tl"}
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: cannot find name 'tl_slave' in ChipTop
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>tl_slave"}
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: cannot find name 'clock_uncore' in ChipTop
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>clock_uncore"}
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: cannot find name 'axi4_mem_0' in ChipTop
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>axi4_mem_0"}
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: cannot find name 'uart_0' in ChipTop
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>uart_0"}
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: cannot find name 'jtag' in ChipTop
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>jtag"}
circuit TestHarness :
^
```

To reproduce, run:

```
firtool \
        --format=fir \
        --dedup \
        --export-module-hierarchy \
        --emit-metadata \
        --verify-each=true \
        --warn-on-unprocessed-annotations \
        --disable-annotation-classless \
        --disable-annotation-unknown \
        --mlir-timing \
        --lowering-options=emittedLineLength=2048,noAlwaysComb,disallowLocalVariables,verifLabels,locationInfoStyle=wrapInAtSquareBracket,disallowPackedArrays \
        --repl-seq-mem \
        --repl-seq-mem-file=mems.conf \
        --repl-seq-mem-circuit=TestHarness \
        --annotation-file=anno.json \
        --split-verilog \
        -o foo \
        foo.fir
        ```

oharboe, 2023-09-17:
Examining `debug_jals`, it looks like `dontTouch(debug_jals)` didn't take, because it is absent in the .fir file.

https://github.com/riscv-boom/riscv-boom/blob/96da674bc97955e7fa068f0a9a1d0a7a479d1d0b/src/main/scala/exu/core.scala#L312

oharboe, 2023-09-17:
I tried making these changes:

```
diff --git a/src/main/scala/exu/core.scala b/src/main/scala/exu/core.scala
index 0f8d39e3..2e22433f 100644
--- a/src/main/scala/exu/core.scala
+++ b/src/main/scala/exu/core.scala
@@ -308,14 +308,14 @@ class BoomCore()(implicit p: Parameters) extends BoomModule
     }))
   }
 
-  dontTouch(debug_brs)
-  dontTouch(debug_jals)
-  dontTouch(debug_jalrs)
+  debug_brs
+  debug_jals
+  debug_jalrs
 
   debug_tsc_reg := debug_tsc_reg + 1.U
   debug_irt_reg := debug_irt_reg + PopCount(rob.io.commit.arch_valids.asUInt)
-  dontTouch(debug_tsc_reg)
-  dontTouch(debug_irt_reg)
+  debug_tsc_reg
+  debug_irt_reg
 
   //****************************************
   // Print-out information about the machine
```

And I got fewer errors:

```
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: cannot find name 'serial_tl' in ChipTop
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>serial_tl"}
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: cannot find name 'tl_slave' in ChipTop
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>tl_slave"}
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: cannot find name 'clock_uncore' in ChipTop
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>clock_uncore"}
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: cannot find name 'axi4_mem_0' in ChipTop
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>axi4_mem_0"}
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: cannot find name 'uart_0' in ChipTop
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>uart_0"}
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: cannot find name 'jtag' in ChipTop
circuit TestHarness :
^
/home/oyvind/chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir:2:1: error: Unable to resolve target of annotation: {class = "firrtl.transforms.DontTouchAnnotation", target = "~TestHarness|ChipTop>jtag"}
circuit TestHarness :
```

I did a bit of spelunking, I am way out of my depth here, and I did find uses of dontTouch, that seems related to the remaining error messages...

https://github.com/ucb-bar/chipyard/blob/174ebc50b701a03226fd71c32d185727078cceb3/generators/chipyard/src/main/scala/IOBinders.scala#L55C4-L55C4

https://github.com/ucb-bar/chipyard/blob/174ebc50b701a03226fd71c32d185727078cceb3/generators/chipyard/src/main/scala/IOBinders.scala#L84

seldridge, 2023-09-19:
I think I see what's going on... Are you running the design through SFC to lower types and then trying to use this with MFC? (Is this the split Chipyard SFC/MFC flow?)

In the input `.fir` I see lowered types, i.e., the original `debug_brs` is lowered to `[debug_brs_0, debug_brs_1, debug_brs_2, debug_brs_3]`. The annotation file only refers to `debug_brs` which does not exist in the circuit.

If you want to try to do this, the annotation file would need to be updated (whatever SFC outputs here).

Alternatively, I'm not sure about the status of Chipyard flows with MFC only. You may want to ping those guys. @abejgonzalez?

oharboe, 2023-09-19:
> I think I see what's going on... Are you running the design through SFC to lower types and then trying to use this with MFC? (Is this the split Chipyard SFC/MFC flow?)

I don't actually know what Chipyard is doing...

Based on https://chipyard.readthedocs.io/en/stable/VLSI/Sky130-OpenROAD-Tutorial.html and then running `cd vlsi && make buildfile tutorial=sky130-openroad CONFIG=MegaBoomConfig` on latest Chipyard master 

> In the input `.fir` I see lowered types, i.e., the original `debug_brs` is lowered to `[debug_brs_0, debug_brs_1, debug_brs_2, debug_brs_3]`. The annotation file only refers to `debug_brs` which does not exist in the circuit.

Ah, so manually edit the automatically generated annotation file so that it matches the mixed SFC/MFC flow...

Eagerly awaiting a MFC only flow where this works :-)

oharboe, 2023-09-25:
@seldridge I tried to wander through the `.fir` file and the `.anno.json` files, but I wasn't clear how to manually modify them to test out MegaBoomConfig on latest Chipyard. If I try to use the most recent release(1.10.0), then I run into problems with Ubuntu 23.04 that I'm on. It does seem like Ubuntu 23.04 is better supported on the most recent main branch of Chipyard.

seldridge, 2023-09-25:
> Ah, so manually edit the automatically generated annotation file so that it matches the mixed SFC/MFC flow...

😉 If this flow is using SFC to do partial lowering, that should be emitting an output `.anno.json` file which can then be passed to the MFC. There _shouldn't_ be any manual modification necessary unless SFC isn't updating the annotations correctly.

For these flows you may want to get one of the Chipyard people involved as I'm sure they will understand how the flow works better than me.

oharboe, 2023-09-26:
The solution to this problem will be in Chipyard.

closed 2023-09-26 by oharboe

jerryz123, 2023-09-29:
We need to remove the SFC part... I think we can now that there is a standalone fixedpoint library. Hopefully soon.

oharboe, 2023-09-29:
> We need to remove the SFC part... I think we can now that there is a standalone fixedpoint library. Hopefully soon.

Super! Thanks!

oharboe, 2023-10-16:
When I remove `--annotation-file=...../chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.anno.json`, the command below completes without error and generates files into the output folder:

```
firtool \
	--format=fir \
	--dedup \
	--export-module-hierarchy \
	--emit-metadata \
	--verify-each=true \
	--warn-on-unprocessed-annotations \
	--disable-annotation-classless \
	--disable-annotation-unknown \
	--mlir-timing \
	--lowering-options=emittedLineLength=2048,noAlwaysComb,disallowLocalVariables,verifLabels,locationInfoStyle=wrapInAtSquareBracket \
	--repl-seq-mem \
	--repl-seq-mem-file=.../chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.mems.conf \
	--repl-seq-mem-circuit=TestHarness \
	--annotation-file=.../chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.anno.json \
	--split-verilog \
	-o .../chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/gen-collateral \
	.../chipyard/vlsi/generated-src/chipyard.harness.TestHarness.MegaBoomConfig/chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir
```