llvm/circt #6144: Inline add in element-select context may be problematic

Tang-Haojin, 2023-09-18:
I am using chisel 6.0-M3 with firtool version 1.54.0.

To descride the issue, may I put an example here:

``` scala
class our extends Module {
  val in = IO(Input(UInt(5.W)))
  val testvar = VecInit.tabulate(32)(i => (114514+i).U(39.W))
  printf(p"in:$in, output:${testvar(in + 1.U)}\n");
  when(in === 31.U) {
    stop()
  }
}
```

The chisel code above would generate verilog output like:

``` verilog
module our(
  input       clock,
              reset,
  input [4:0] in	// src/main/scala/gcd/GCD.scala:13:14
);

  `ifndef SYNTHESIS	// src/main/scala/gcd/GCD.scala:15:9
    always @(posedge clock) begin	// src/main/scala/gcd/GCD.scala:15:9
      if ((`PRINTF_COND_) & ~reset) begin	// src/main/scala/gcd/GCD.scala:15:9
        automatic logic [31:0][38:0] _GEN =
          {39'h1BF71,
           // ------ many lines ------
           39'h1BF52};	// src/main/scala/gcd/GCD.scala:15:9
        $fwrite(32'h80000002, "in:%d, output:%d\n", in, _GEN[in + 5'h1]);	// src/main/scala/gcd/GCD.scala:15:{9,40}
      end
      if ((`STOP_COND_) & (&in) & ~reset)	// src/main/scala/gcd/GCD.scala:15:9, :16:11, :17:9
        $finish;	// src/main/scala/gcd/GCD.scala:17:9
    end // always @(posedge)
  `endif // not def SYNTHESIS
endmodule
```

Here, we may focus on the `_GEN[in + 5'h1]` in the `$fwrite` line. Accroding to what @wsnyder mentioned in https://github.com/verilator/verilator/issues/4498, when `in` equals to `5'd31`, `_GEN[in + 5'h1]` is not necessarily equals to `_GEN[5'h0]`, which is inconsistent with the semanteme of `+` in chisel.

I am not pretty sure whether what I mentioned above is correct, but if we wrap `in + 5'h1` in concat context, i.e. `_GEN[{in + 5'h1}]`, the result would always be correct (here "correct" refers to being consistent with chisel's semanteme), accroding to IEEE 1800-2017 _11.6.1 Rules for expression bit lengths_, where it states that all operands in a concat expression are self-determined. That means `Length({in + 5'h1}) = Length(in + 5'h1) = max{Length(in), Length(5'h1)} = 5`. I also tested this on verilator, and it did output as what I expected.

uenoku, 2023-09-19:
I think you can use `explicitBitcast` lowering options to force the bit width for this kind of problems.

Tang-Haojin, 2023-09-19:
Thank you. Let me have a try.

Tang-Haojin, 2023-09-25:
Excuse me for the late response. 

I have tried and this can actually work. Should we make it the default behaviour in this kind of element-select context?