llvm/circt #6153: [CI] Daily python wheel upload fails on mac os

uenoku, 2023-09-20:
It seems wheel has not been not uploaded for a week.
https://github.com/llvm/circt/actions/runs/6235293526

mikeurbach, 2023-09-20:
Hmm, maybe related to the change to use universal wheels in https://github.com/llvm/circt/pull/5822/files?

cc @leonardt

leonardt, 2023-09-20:
I opened a PR to revert to the standard wheel option for now: https://github.com/llvm/circt/pull/6158

I suspect the issue here is that the dependency library (Python bindings and/or circt itself) are not being compiled in a compatible manner.  In other packages, I think we built everything from source through a Python build script and the relevant x86/arm compiler options didn't present any issues, but now that I think about it, it would be reasonable to think that CIRCT requires some specific tweaks for each platform.

I will investigate whether there's a reasonable option for building the universal wheel, but I suspect the better longterm option would be to build/deploy separate x86/arm environments for macos.  The issue here is that, last time I checked, GitHub actions doesn't provide an arm runner (hence the need to use x86 to compile for arm).  Perhaps a temporary workaround would be to have two separate macOS environments (both are technically x86, but one will be used to cross-compile)

mikeurbach, 2023-09-20:
Thanks, that make sense. I have no experience with building universal wheels for Python projects that require native extensions, but it should be possible. It looks like torch-mlir releases include universal wheels, so maybe we can lift something from their CI setup. https://github.com/llvm/torch-mlir

leonardt, 2023-09-22:
Investigating the setup script, it seems like adding the CMAKE_OSX_ARCHITECTURES option could be done here: https://github.com/llvm/circt/blob/main/lib/Bindings/Python/setup.py#L67-L80 ?

From what I understand of the CI script: https://github.com/llvm/circt/blob/main/.github/workflows/uploadWheels.yml we build CIRCT and LLVM from scratch for each environment, so I think we could add another set of matrix entries that mirror these https://github.com/llvm/circt/blob/main/.github/workflows/uploadWheels.yml#L27-L34 except for arm architecture.

leonardt, 2023-09-22:
(We'd also want to make sure we add these options https://github.com/llvm/circt/pull/6169/files for the arm builds)

leonardt, 2023-09-22:
Or, rather the ARM specific option (since we wouldn't be doing the universal build)

makslevental, 2023-09-22:
> Investigating the setup script, it seems like adding the CMAKE_OSX_ARCHITECTURES option could be done here: https://github.com/llvm/circt/blob/main/lib/Bindings/Python/setup.py#L67-L80 ?

Yea this is correct - sorry I confused PyCDE for the bindings

> these https://github.com/llvm/circt/blob/main/.github/workflows/uploadWheels.yml#L27-L34 except for arm architecture.

Yea that seems like the way to go for you guys.

Just a small pre-emptive warning (even though it hasn't been mentioned here): the story is not at all so simple for  linux_aarch64 but :shrug: cross that bridge if you ever get to it.