llvm/circt #6159: [ExportVerilog] Local-only test failure in verilog-locations test

teqdruid, 2023-09-20:
```
$ ninja -C ~/internal_circt_build/ -j18 check-circt
ninja: Entering directory `/home/jodemme/internal_circt_build/'
[16/17] Running the CIRCT regression tests
FAIL: CIRCT :: Conversion/ExportVerilog/verilog-locations.mlir (674 of 675)
******************** TEST 'CIRCT :: Conversion/ExportVerilog/verilog-locations.mlir' FAILED ********************
Script:
--
: 'RUN: at line 1';   /home/jodemme/internal_circt_build/bin/circt-opt /home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/test/Conversion/ExportVerilog/verilog-locations.mlir -export-verilog -verify-diagnostics --mlir-print-debuginfo --split-input-file | /home/jodemme/internal_circt_build/bin/FileCheck /home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/test/Conversion/ExportVerilog/verilog-locations.mlir --strict-whitespace
--
Exit Code: 1

Command Output (stderr):
--
/home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/test/Conversion/ExportVerilog/verilog-locations.mlir:216:11: error: CHECK: expected string not found in input
// CHECK: #loc5 = loc("":4:0)
          ^
<stdin>:195:148: note: scanning from here
#loc4 = loc("within split at /home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/test/Conversion/ExportVerilog/verilog-locations.mlir:201 offset ":6:1)
                                                                                                                                                   ^
<stdin>:196:1: note: possible intended match here
#loc5 = loc("":5:0)
^

Input file: <stdin>
Check file: /home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/test/Conversion/ExportVerilog/verilog-locations.mlir

-dump-input=help explains the following input dump.

Input was:
<<<<<<
             .
             .
             .
           190: } loc(#loc)
           191: #loc = loc("within split at /home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/test/Conversion/ExportVerilog/verilog-locations.mlir:201 offset ":4:1)
           192: #loc1 = loc("within split at /home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/test/Conversion/ExportVerilog/verilog-locations.mlir:201 offset ":5:1)
           193: #loc2 = loc("":2:0)
           194: #loc3 = loc("":3:0)
           195: #loc4 = loc("within split at /home/jodemme/FPGAToolsAndCompilers/CIRCT/circt/test/Conversion/ExportVerilog/verilog-locations.mlir:201 offset ":6:1)
check:216'0                                                                                                                                                        X error: no match found
           196: #loc5 = loc("":5:0)
check:216'0     ~~~~~~~~~~~~~~~~~~~~
check:216'1     ?                    possible intended match
           197: #loc6 = loc("":6:0)
check:216'0     ~~~~~~~~~~~~~~~~~~~~
           198: #loc7 = loc(fused<"Range">[#loc2, #loc3])
check:216'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           199: #loc8 = loc(fused<"Range">[#loc5, #loc6])
check:216'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           200: #loc9 = loc(fused<"verilogLocations">[#loc7])
check:216'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
           201: #loc10 = loc(fused<"verilogLocations">[#loc8])
check:216'0     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
             .
             .
             .
>>>>>>

--

********************
********************
Failed Tests (1):
  CIRCT :: Conversion/ExportVerilog/verilog-locations.mlir
```

For some reason, my local build is including full paths in the locations whereas the CI builds do not. This failure is very likely related to #6092.

teqdruid, 2023-09-20:
@prithayan I suspect the fix is to add some regexes to the test to ignore the file locations, but I may be wrong.

prithayan, 2023-09-20:
Thanks, I was trying to check for verilog output file location in the test, and the output is supposed to be deterministic. 
Something is not right, I will take a look.

teqdruid, 2023-09-27:
How's this investigation going? It's annoying to get a test failure locally.

prithayan, 2023-09-27:
@teqdruid , sorry for the delay, I have been able to reproduce the issue. The output at your end seems to be consistent with `mlir-disable-threading`. I will fix this asap.

prithayan, 2023-09-27:
@teqdruid , can you please verify if the issue is fixed on your end ?

closed 2023-09-27 by prithayan

teqdruid, 2023-09-27:
Yes! The tests pass locally on my machine now. Thanks!

commit b3772231 (2023-09-28): [ExportVerilog] Fix line out in debug verilog locations (#6210)

Line offset count was incorrectly incremented outside else block, which
 resulted in incorrect line numbers when threading was disabled.
Fixes #6159
commit 69b5a325 (2023-10-02): [ExportVerilog] Fix line out in debug verilog locations (#6210)

Line offset count was incorrectly incremented outside else block, which
 resulted in incorrect line numbers when threading was disabled.
Fixes #6159