llvm/circt #6193: [FIRRTL][firtool] Unknown Windows Release Build Test Failures

seldridge, 2023-09-23:
There have been a number of odd test failures that have been observed on Windows release builds.

## Failures

### First Failure

A commit, https://github.com/llvm/circt/commit/3668f6760f9c7ca814035292435f930e30320fea, enabling properties as attributes in the Seq dialect resulted in similar failures in one test. See: https://github.com/llvm/circt/actions/runs/6043611927/job/16400911216 This corrected itself on the subsequent commit, https://github.com/llvm/circt/commit/c88186e18a74fe74e2d2bdb7de2610a0334eaa7d, passed.

The failing test was:
  - `Dialect/FIRRTL/SFCTests/width-spec-errors.fir`

The exit code was: 3221225477

### Second Failure

A commit adding two folders, https://github.com/llvm/circt/commit/68b79fc35fcc2d68abc8a0b830c0f5a303af3f04, failed intermittently. See: https://github.com/llvm/circt/actions/runs/6083208189/job/16502633046. This then passed on retry: https://github.com/llvm/circt/actions/runs/6083208189.

The failing test was:
  - `Dialect/FIRRTL/SFCTests/data-taps-flip.fir`

The exit code was: 3221225477

There was no stack trace.

### Third Failure

Commit https://github.com/llvm/circt/commit/3b6f737c360fa19e3f16f9ea4856128f36ce9296 made a change that disabled deduplication. This caused four tests to fail repeatedly. One such run is: https://github.com/llvm/circt/actions/runs/6278904029/job/17056030822

The failing tests were:
  - `Dialect/FIRRTL/SFCTests/GrandCentralInterfaces/Wire.fir`
  - `Dialect/FIRRTL/SFCTests/data-taps.fir`
  - `firtool/prefixMemory.fir`
  - `firtool/refs-in-aggs.fir`

Only `firtool/prefixMemory.fir` was a test that was sensitive to deduplication and required modifications, in the commit, to fix.

This was "fixed" by adding a `-no-dedup` option to `firtool` and using it for those four tests. It is unclear if this actually "fixed" anything, i.e., if the issue was that not running deduplication fixed it or if this just permuted the sources enough to get `cl` to produce something that didn't crash.

The exit code was: 0xc0000005

There was a stack trace. (This isn't particularly useful as it is a release build...) Example:

```
 #0 0x00007ff7b9dca204 (d:\a\circt\circt\build\bin\firtool.exe+0x1fa204)
 #1 0x00007ff7b9dd00a1 (d:\a\circt\circt\build\bin\firtool.exe+0x2000a1)
 #2 0x00007ff7ba5f0107 (d:\a\circt\circt\build\bin\firtool.exe+0xa20107)
 #3 0x00007ff7ba5f5a27 (d:\a\circt\circt\build\bin\firtool.exe+0xa25a27)
 #4 0x00007ff7ba5f7529 (d:\a\circt\circt\build\bin\firtool.exe+0xa27529)
 #5 0x00007ff7ba5f0616 (d:\a\circt\circt\build\bin\firtool.exe+0xa20616)
 #6 0x00007ff7ba5f6ee2 (d:\a\circt\circt\build\bin\firtool.exe+0xa26ee2)
 #7 0x00007ff7ba5f00f2 (d:\a\circt\circt\build\bin\firtool.exe+0xa200f2)
 #8 0x00007ff7ba5f5a27 (d:\a\circt\circt\build\bin\firtool.exe+0xa25a27)
 #9 0x00007ff7ba5f61fd (d:\a\circt\circt\build\bin\firtool.exe+0xa261fd)
#10 0x00007ff7b9be0544 (d:\a\circt\circt\build\bin\firtool.exe+0x10544)
#11 0x00007ff7b9be09ff (d:\a\circt\circt\build\bin\firtool.exe+0x109ff)
#12 0x00007ff7b9bdee3b (d:\a\circt\circt\build\bin\firtool.exe+0xee3b)
#13 0x00007ff7b9be26df (d:\a\circt\circt\build\bin\firtool.exe+0x126df)
#14 0x00007ff7ba60d560 (d:\a\circt\circt\build\bin\firtool.exe+0xa3d560)
#15 0x00007ffe8a4c7ac4 (C:\Windows\System32\KERNEL32.DLL+0x17ac4)
#16 0x00007ffe8bfda351 (C:\Windows\SYSTEM32\ntdll.dll+0x5a351)
```

Just before the third test, the windows build failed on the previous commit twice. This appeared to be due to issues running PowerShell's `Install-Package`. The two failures were:

```
WARNING: Unable to resolve package source 'https://www.powershellgallery.com/api/v2'.
Install-Package: No match was found for the specified search criteria and module name 'VSSetup'. Try Get-PSRepository to see all
available registered module repositories.

Error: Process completed with exit code 1.
```

```
Install-Package: No match was found for the specified search criteria and module name 'VSSetup'. Try Get-PSRepository to see all
available registered module repositories.

Error: Process completed with exit code 1.
```

This appears unrelated, though it has high temporal locality to the third failure.

For the third failure, I tried a clean rebuild by deleting all `sccache` caches. This again failed in the same way after a 3 hour complete rebuild.

## Analysis

1. All tests involved `firtool` end-to-end tests. 
2. All tests were some kind of hard crash with exit codes 3221225477 or 0xc0000005
3. The third test would repeatedly fail. The second test was intermittent. It is unknown if the first test was intermittent or repeatable.

## Ideas to try

1. The Unified Build/Test/Install GitHub Action does support running a Windows build in debug mode or relwithdebinfo. This won't have caching and will be about 3 hours to run.
2. More UBSan or other sanitizers. Is there a way to run these on Windows?

teqdruid, 2023-09-23:
@seldridge  Feel free to use my branch attempting to move the windows builds over to the new agents: https://github.com/llvm/circt/tree/dev/teqdruid/win-agent. A bunch of firrtl tests fail (which we spoke about) but I haven't had time to investigate further. Those agents are big enough that I think we can run debug builds and quickly.

teqdruid, 2023-09-23:
Last run of Windows build on that branch: https://github.com/llvm/circt/actions/runs/6193977023/job/16816284984

dtzSiFive, 2023-09-25:
Thank you! Just ran a debug build and all tests pass: https://github.com/llvm/circt/actions/runs/6299670167/job/17100859285 :thinking: .  Will try `relwithdebinfo` next.

dtzSiFive, 2023-09-25:
Dropping the `exit()`-leak approach for `firtool` maybe fixes this?

https://github.com/llvm/circt/actions/runs/6301781311/job/17107569408

Following run also passes, building on that by also reverting the `Seq` change as a data-point, but that's two passing runs.

Don't have details on why that would fix things, and only on release builds? It does skip the local destructors (that's the point), it does register some signal handlers that maybe are getting hit (SIGPIPE?).

Anyway that's a sketchy explanation at best but seems like a lead to follow or a possible solution.

dtzSiFive, 2023-09-25:
Note to self: Try this on master -- but drop the `-no-dedup` workaround? Doesn't prove it but would be good to check.

seldridge, 2023-11-11:
Possibly the same problem still existed and was reported here: https://github.com/llvm/circt/issues/6395

We ran ASAN and UBSAN on Linux/macOS builds on release and debug builds without being able to find anything.

The current "fix" is things are looking more stable with the Windows 2022 runners which have a newer version of MSVC. (I verified it was newer. I did not record the number.)

I'm going to close this. This can be re-opened or another issue can opened if we see this behavior again.

closed 2023-11-11 by seldridge