llvm/circt #6220: [Ibis] Nightly Tests Failed w/ Valgrind

seldridge, 2023-09-28:
Tracking issue for Valgrind failure on an Ibis test observed here: https://github.com/llvm/circt/actions/runs/6338557115/job/17215924247#step:6:4290

The error is reproduced below:
```
******************** TEST 'CIRCT :: Dialect/Ibis/Transforms/reblock.mlir' FAILED ********************
Exit Code: 123

Command Output (stderr):
--
RUN: at line 1: /__w/circt/circt/build/bin/circt-opt --pass-pipeline='builtin.module(ibis.class(ibis.method(ibis-reblock)))' /__w/circt/circt/test/Dialect/Ibis/Transforms/reblock.mlir | /__w/circt/circt/build/bin/FileCheck /__w/circt/circt/test/Dialect/Ibis/Transforms/reblock.mlir
+ /__w/circt/circt/build/bin/circt-opt '--pass-pipeline=builtin.module(ibis.class(ibis.method(ibis-reblock)))' /__w/circt/circt/test/Dialect/Ibis/Transforms/reblock.mlir
+ /__w/circt/circt/build/bin/FileCheck /__w/circt/circt/test/Dialect/Ibis/Transforms/reblock.mlir
==21379== Invalid read of size 8
==21379==    at 0x1635920: (anonymous namespace)::ReblockPass::runOnOperation() (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x210E17B: mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x210E941: mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x211454F: mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_15::operator()(mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo&) const (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x211004A: mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x210E2C0: mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x210E941: mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x211454F: mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_15::operator()(mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo&) const (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x211004A: mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x210E2C0: mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x210E941: mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x2110F48: mlir::PassManager::run(mlir::Operation*) (in /__w/circt/circt/build/bin/circt-opt)
==21379==  Address 0x6a63800 is 48 bytes inside a block of size 64 free'd
==21379==    at 0x483CA3F: free (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==21379==    by 0x1A3FBF9: llvm::iplist_impl<llvm::simple_ilist<mlir::Operation>, llvm::ilist_traits<mlir::Operation> >::erase(llvm::ilist_iterator<llvm::ilist_detail::node_options<mlir::Operation, true, false, void>, false, false>) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x1635856: (anonymous namespace)::ReblockPass::runOnOperation() (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x210E17B: mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x210E941: mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x211454F: mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_15::operator()(mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo&) const (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x211004A: mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x210E2C0: mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x210E941: mlir::detail::OpToOpPassAdaptor::runPipeline(mlir::OpPassManager&, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int, mlir::PassInstrumentor*, mlir::PassInstrumentation::PipelineParentInfo const*) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x211454F: mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::$_15::operator()(mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool)::OpPMInfo&) const (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x211004A: mlir::detail::OpToOpPassAdaptor::runOnOperationAsyncImpl(bool) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x210E2C0: mlir::detail::OpToOpPassAdaptor::run(mlir::Pass*, mlir::Operation*, mlir::AnalysisManager, bool, unsigned int) (in /__w/circt/circt/build/bin/circt-opt)
==21379==  Block was alloc'd at
==21379==    at 0x483B7F3: malloc (in /usr/lib/x86_64-linux-gnu/valgrind/vgpreload_memcheck-amd64-linux.so)
==21379==    by 0x1A37BDE: mlir::Operation::create(mlir::Location, mlir::OperationName, mlir::TypeRange, mlir::ValueRange, mlir::DictionaryAttr, mlir::OpaqueProperties, mlir::BlockRange, unsigned int) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x1A37796: mlir::Operation::create(mlir::Location, mlir::OperationName, mlir::TypeRange, mlir::ValueRange, mlir::NamedAttrList&&, mlir::OpaqueProperties, mlir::BlockRange, mlir::RegionRange) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x1A37653: mlir::Operation::create(mlir::OperationState const&) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x19AB4DF: mlir::OpBuilder::create(mlir::OperationState const&) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x20D46EB: (anonymous namespace)::OperationParser::parseOperation() (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x20DF8D0: (anonymous namespace)::OperationParser::parseBlock(mlir::Block*&) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x20DD239: (anonymous namespace)::OperationParser::parseRegion(mlir::Region&, llvm::ArrayRef<mlir::OpAsmParser::Argument>, bool) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x20DB3B8: (anonymous namespace)::CustomOpAsmParser::parseRegion(mlir::Region&, llvm::ArrayRef<mlir::OpAsmParser::Argument>, bool) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x15F86BD: mlir::ParseResult parseMethodLikeOp<circt::ibis::MethodOp>(mlir::OpAsmParser&, mlir::OperationState&) (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x20D453B: (anonymous namespace)::OperationParser::parseOperation() (in /__w/circt/circt/build/bin/circt-opt)
==21379==    by 0x20DF8D0: (anonymous namespace)::OperationParser::parseBlock(mlir::Block*&) (in /__w/circt/circt/build/bin/circt-opt)
==21379== 
```

seldridge, 2023-09-28:
FWIW, this is actually failing in my local build, too, not just in nightly. 🥲 

CC: @teqdruid, @mortbopet

dtzSiFive, 2023-09-28:
FWIW here's locally generated ASAN report for this: https://gist.github.com/dtzSiFive/e1f67f78a06a85a7ca91382e927d0a6a .

teqdruid, 2023-09-28:
@mortbopet can you take a look?

teqdruid, 2023-09-28:
Oops: miss Schuyler's fix. Thanks again Schuyler!

commit 98203ec8 (2023-09-28): [ibis] Fix reblock use of erased ops

Fix a bug in the IbisReblockPass where an erased op was being pushed into
a SmallVector.  Delay deletion of ops until runOnOperation finishes.

Fixes #6220.
PR #6222 (2023-09-28): [ibis] Fix reblock use of erased ops
Fix a bug in the IbisReblockPass where an erased op was being pushed into a SmallVector.  Delay deletion of ops until runOnOperation finishes.

Fixes #6220.
commit af2b3e13 (2023-09-28): [ibis] Fix reblock use of erased ops (#6222)

Fix a bug in the IbisReblockPass where an erased op was being pushed into
a SmallVector.  Delay deletion of ops until runOnOperation finishes.

Fixes #6220.
closed 2023-09-28 by seldridge