llvm/circt #6245: [Arc] Improve inliner performance

fabianschuiki, 2023-10-03:
The `InlineArcs` pass scales poorly to bigger designs. For example, on `boom-small` from the github.com/circt/arc-tests repo it accounts for 23s of the overall 32s of runtime. If tracing is enabled, the numbers get even worse. This is likely due to the fact that the inliner works from leaf arcs backwards. If arcs are nested, operations may potentially be inlined multiple times, which is pretty bad. Take a closer look at the inliner's performance and come up with a solution.

sequencer, 2023-10-06:
When I did the verilting performance benchmark years ago. from an uArch view, I noticed a giant footprint in I$, with intensive inlining strategy like this. Increasing inlining reduces BPU overhead, but increased IF pressure. I think adding more uArch metric(cache, BPU) when benchmark Arc performance can provide more information discussing how to improve the performance.

fabianschuiki, 2023-10-09:
Yeah that's a great point. We should definitely have more detailed trackign of these performance metrics.

The 23s and 32s of runtime mentioned in the issue are actually the compiler runtime, not the resulting runtime of the simulator. I'll add a comment.

closed 2023-11-06 by maerhart