llvm/circt #6259: [FIRRTL] MergeConnections Invalid IR for RWProbes

seldridge, 2023-10-05:
I hit the following issue involving a RWProbe for the following FIRRTL:

```
FIRRTL version 3.1.0
circuit Foo : %[[
  {
    "class":"firrtl.transforms.DontTouchAnnotation",
    "target":"~Foo|Foo>w"
  }
]]
  module Foo :
    ; output a : RWProbe<{a : UInt<1>, b : UInt<2> }>
    output a: {a : UInt<1>, b : UInt<2> }

    wire w : { a : UInt<1>, flip b : UInt<2> }
    invalidate w.a
    invalidate w.b
    ; define a = rwprobe(w)
    connect a, w
```

This errors out with invalid IR after `MergeConnections`:

```
xxx.fir:12:5: error: 'firrtl.aggregateconstant' op Cannot have constant bundle type with flip
    invalidate w.a
    ^
xxx.fir:12:5: note: see current operation: %1 = "firrtl.aggregateconstant"() {fields = [0 : ui1, 0 : ui2]} : () -> !firrtl.bundle<a: uint<1>, b flip: uint<2>>
xxx.fir:13:5: note: additional location here
    invalidate w.b
    ^
```

```mlir
"firrtl.module"() ({
^bb0(%arg0: !firrtl.rwprobe<bundle<a: uint<1>, b: uint<2>>>):
  %0:2 = "firrtl.wire"() {annotations = [{class = "firrtl.transforms.DontTouchAnnotation"}], forceable, name = "w", nameKind = #firrtl<name_kind droppable_name>} : () -> (!firrtl.bundle<a: uint<1>, b flip: uint<2>>, !firrtl.rwprobe<bundle<a: uint<1>, b: uint<2>>>)
  %1 = "firrtl.aggregateconstant"() {fields = [0 : ui1, 0 : ui2]} : () -> !firrtl.bundle<a: uint<1>, b flip: uint<2>>
  "firrtl.connect"(%0#0, %1) : (!firrtl.bundle<a: uint<1>, b flip: uint<2>>, !firrtl.bundle<a: uint<1>, b flip: uint<2>>) -> ()
  "firrtl.ref.define"(%arg0, %0#1) : (!firrtl.rwprobe<bundle<a: uint<1>, b: uint<2>>>, !firrtl.rwprobe<bundle<a: uint<1>, b: uint<2>>>) -> ()
}) {annotations = [], convention = #firrtl<convention scalarized>, portAnnotations = [[]], portDirections = true, portLocations = [loc("xxx.fir":9:12)], portNames = ["a"], portSyms = [], portTypes = [!firrtl.rwprobe<bundle<a: uint<1>, b: uint<2>>>], sym_name = "Foo"} : () -> ()
```

This is trying to build an aggregate constant with flips. 😕

dtzSiFive, 2023-10-05:
Yes.  This is because we don't reject it at LowerTypes, which we would with, e.g., https://github.com/llvm/circt/pull/5835 .

Was trying to be "best-effort" here instead of actively rejecting such things but that was probably a mistake, although IIRC it does sometimes work out (maybe if you preserve aggregates?).

We can of course reject it explicitly but I think maybe moving off Forceable would be good so we don't have so many things.

dtzSiFive, 2023-10-05:
Oh but perhaps MergeConnections can/should avoid this regardless, per issue title :+1:.

dtzSiFive, 2023-10-05:
FWIW your example has the `RWProbe`s commented-out, and also doesn't work as-is:

```
issue-6259.fir:13:5: error: 'firrtl.aggregateconstant' op Cannot have constant bundle type with flip
    invalidate w.a
    ^
issue-6259.fir:13:5: note: see current operation: %1 = "firrtl.aggregateconstant"() {fields = [0 : ui1, 0 : ui2]} : () -> !firrtl.bundle<a: uint<1>, b flip: uint<2>>
issue-6259.fir:14:5: note: additional location here
    invalidate w.b
    ^
```

Looks like half-way between two worlds re:testing, replacing the connect and the `a` port with the commented-out lines is probably what was meant / the issue is about.

dtzSiFive, 2023-10-05:
Yeah, ran that on #5835 rebased to master and while LowerTypes doesn't diagnose /why/ there's a symbol, it rejects it directly:

```
issue-6259.fir:12:5: error: unable to lower due to symbol "sym" with target not preserved by lowering
    wire w : { a : UInt<1>, flip b : UInt<2> }
    ^
```

dtzSiFive, 2023-10-06:
Looks like MergeConnections doesn't create aggregate constant ops directly, but these are materialized when doing a `createOrFold` on the vector/bundle create operations with constant inputs (and these operations don't themselves verify they are passive).

PR #6264 (2023-10-06): [FIRRTL][MergeConnections] Reject non-passive aggregates.
Previously: #5780.

Fixes #6259.

(along with #5835 for the specific FIRRTL input)
commit aeb6f0fe (2023-10-06): [FIRRTL][MergeConnections] Reject non-passive aggregates. (#6264)

Previously: #5780.

Fixes #6259.

(along with #5835 for the specific FIRRTL input)
closed 2023-10-06 by dtzSiFive