llvm/circt #6271: [HW] Unequal equal types on instancing a module outputting a parametrized array / ArrayType sizeAttr defaults to 64bit

DormBrand, 2023-10-09:
I’ve recently been exposed to Circt at my student assistant job and have been playing around with it, seeing what kind of stuff the IR supports. During testing the inclusion of already written Verilog modules, I've run into an error with parameterized array types.

To reproduce the error, I've made the following simplified code:

```
hw.module.extern @submodule<A: i32 = 3, B: i32 = 8>(out out : !hw.array<#hw.param.decl.ref<"A">x!hw.int<#hw.param.decl.ref<"B">>>)

hw.module @test<A: i32 = 3, B: i32 = 8>(out out: !hw.array<#hw.param.decl.ref<"A">x!hw.int<#hw.param.decl.ref<"B">>>) {

  %out = hw.instance "submodule" @submodule<
    A: i32 = #hw.param.decl.ref<"A">, 
    B: i32 = #hw.param.decl.ref<"B">
  >() -> (out : !hw.array<#hw.param.decl.ref<"A">x!hw.int<#hw.param.decl.ref<"B">>>)

  hw.output %out : !hw.array<#hw.param.decl.ref<"A">x!hw.int<#hw.param.decl.ref<"B">>>
}
```

Running circt-opt results in an error, it seems to consider the two array types as unequal, even though the dumps sound equal.
```
/local/projects/circt_cnn/src/mlir/example_notworking.mlir:7:10: error: 'hw.instance' op result type #0 must be '!hw.array<#hw.param.decl.ref<"A">xint<#hw.param.decl.ref<"B">>>', but got '!hw.array<#hw.param.decl.ref<"A">xint<#hw.param.decl.ref<"B">>>'
  %out = hw.instance "submodule" @submodule<
         ^
/local/projects/circt_cnn/src/mlir/example_notworking.mlir:7:10: note: see current operation: %0 = "hw.instance"() {argNames = [], instanceName = "submodule", moduleName = @submodule, parameters = [#hw.param.decl<"A": i32 = #hw.param.decl.ref<"A">> : i32, #hw.param.decl<"B": i32 = #hw.param.decl.ref<"B">> : i32], resultNames = ["out"]} : () -> !hw.array<#hw.param.decl.ref<"A">xint<#hw.param.decl.ref<"B">>>
/local/projects/circt_cnn/src/mlir/example_notworking.mlir:1:1: note: module declared here
hw.module.extern @submodule<A: i32 = 3, B: i32 = 8>(out out : !hw.array<#hw.param.decl.ref<"A">x!hw.int<#hw.param.decl.ref<"B">>>)
```

Attaching a debugger and looking at [InstanceImplementation.cpp#L140](https://github.com/llvm/circt/blob/2ec4cd09218b89b3ef35d9e3acfb0519e34da068/lib/Dialect/HW/InstanceImplementation.cpp#L140), I have found that the reason is the sizeAttr for the expected type and the result type being constructed differently. Both types have a `sizeAttr` with a StringAttr name "A", but the type of the `sizeAttr` is `i32` for the expected type and `i64` for the result typ. Switching the parameter A to be an `i64` instead did generate correct Verilog without any errors.

However I haven't been able to trace where exactly the attribute is constructed, due to the limitations with my debugger and probably not knowing enough about how CIRCT/MLIR work. Notably it is not in [evaluateParametricArrayType (HWAttributes.cpp#L1028](https://github.com/llvm/circt/blob/2ec4cd09218b89b3ef35d9e3acfb0519e34da068/lib/Dialect/HW/HWAttributes.cpp#L1028), changing that 64 to an 8 or 32 did not make a change in the error.

Maybe this error is intended behaviour, and the size of an array always has to be a i64 attribute. However, using i32 parameters as size works everywhere but in instancing a module. If it is intended, maybe an error should be implemented, like how the parametric !hw.int type only takes a i32 integer attribute and throws an error if you try to pass e.g. an i8 parameter ([HWTypes.cpp#L197](https://github.com/llvm/circt/blob/2ec4cd09218b89b3ef35d9e3acfb0519e34da068/lib/Dialect/HW/HWTypes.cpp#L197)).

If it's not intended behaviour, then wherever the size attribute for the result type of the instance is built needs to take into account the type of the parameter instead of making it a i64 by default.

Since I'm not that well versed in C++, if anyone has tips on how to debug things in CIRCT more cleany I'd also appreciate them, finding out what the issue was here took way too long thanks to fighting against the debugger and having to recompile the project often, see this [thread](https://discourse.llvm.org/t/help-debugging-circt-lldb-doesnt-find-half-the-names/73584). There are some other issues with parametric size arrays (e.g. `hw.array_get` crashing) I wanted to investigate a bit further before creating more issues.

seldridge, 2023-10-09:
There's clearly a problem here. Your analysis seems sound. The type of `!hw.array<#hw.param.decl.ref<"A"> x !hw.int<#hw.param.decl.ref<"B">>>` should be the same wherever it is used so long as `"A"` and `"B"` have the same types. Array types are supposed to be a fixed size of i64. However, something is injecting an i32 and it's causing problems.

seldridge, 2023-10-09:
The issue appears to be due to how `hw::evaluteParametricAttr` works. (This is called by `evaluateParametricArrayType`.) This is taking something which is originally a `#hw.param.decl.ref<"A"> : i64` and "evaluating it" to a `#hw.param.decl.ref<"A"> : i32`. Your observation that changing the `64` to an `8` or `32` not having an effect is because that code path wasn't occurring. It was the other path of the evaluation happens, but it is still a parameter.

A minimal diff that I haven't fully through through is:

```diff
diff --git a/lib/Dialect/HW/HWAttributes.cpp b/lib/Dialect/HW/HWAttributes.cpp
index 20ba77df8..4df67528a 100644
--- a/lib/Dialect/HW/HWAttributes.cpp
+++ b/lib/Dialect/HW/HWAttributes.cpp
@@ -1029,7 +1029,8 @@ FailureOr<Type> evaluateParametricArrayType(Location loc, ArrayAttr parameters,
                          intAttr.getValue().getSExtValue()));
 
   // Otherwise parameter references are still involved
-  return TArray::get(arrayType.getContext(), *elementType, *size);
+  return TArray::get(arrayType.getContext(), *elementType,
+                     arrayType.getSizeAttr());
 }
 
 FailureOr<Type> hw::evaluateParametricType(Location loc, ArrayAttr parameters,
```

See if that helps...

Side note: the parameter work in HW dialect is not well tested. Clearly you tried something which nobody else has. 😉  We desperately need more work here and we are planning to push on this by working towards parametric representations in FIRRTL dialect which will then make all this load bearing. Thanks for doing some pathfinding work. 😅

seldridge, 2023-10-09:
CC: @rwy7 who's starting to poke at this, too. 😉

DormBrand, 2023-10-12:
That change does fix the issue, the test code I've posted runs through fine now 👍 

With your fix right now it would be possible to use A: i8, A: none etc. and circt just overwrites that. However if it's not intended for the array size to be anything else than an i64, maybe there should be an error thrown? Probably doesn't matter too much right now, but maybe that could cause problems in the future.

Maybe something like the following diff would work? Although it only checks the actual type of the parameter on module instanciation. If you'd write a single flat top module you could still use an e.g. i8 parameter as an array size and wouldn't encounter an error.

```diff
diff --git a/lib/Dialect/HW/HWAttributes.cpp b/lib/Dialect/HW/HWAttributes.cpp
index 20ba77df8..960a7ef7c 100644
--- a/lib/Dialect/HW/HWAttributes.cpp
+++ b/lib/Dialect/HW/HWAttributes.cpp
@@ -1015,6 +1015,12 @@ FailureOr<Type> evaluateParametricArrayType(Location loc, ArrayAttr parameters,
   auto size = evaluateParametricAttr(loc, parameters, arrayType.getSizeAttr());
   if (failed(size))
     return failure();
+
+  Type sizeType = size->getType();
+  auto sizeTypeInt = sizeType.dyn_cast<IntegerType>();
+  if (!sizeTypeInt || sizeTypeInt.getWidth() != 64)
+    return emitError(loc) << "Array size must be a 64-bit integer";
+
   auto elementType =
       evaluateParametricType(loc, parameters, arrayType.getElementType());
   if (failed(elementType))
```