llvm/circt #6289: Release along with version MLIR?

sequencer, 2023-10-14:
it spent my another morning to rebuild firtool 1.57.0.
The current issue for downstream disto is CIRCT doesn't depend on a release version of MLIR, so distro must rebuild the entire llvm+mlir to build circt.
wondering if it is possible for circt to depend on the release version of MLIR from https://github.com/llvm/llvm-project/releases in the future?
I believe it can make downstream distro devs happy to not rebuild the whole world;p

mortbopet, 2023-10-16:
Is it possible? Yes. However, as far as i am aware (and of which i agree), we have deliberately chosen to track upstream MLIR/LLVM as close as possible (currently on an ~1 week submodule bump cadence, thanks to the people at SiFive).

There are many arguments for this, some being that we get
1) bug fixes and new features as soon as possible - but probably more importantly
2) MLIR isn't really "stable", and there's often significant changes in the core APIs which require major refactorings in CIRCT. These can be challenging to perform even on a weekly basis, so decreasing the frequency of bumps (which would be the case if we started pinning to LLVM releases) will equally increase the difficulty of performing these bumps. Hence, frequent submodule bumps makes the core devs happy 😉.  

Rebuilding the entirety of the LLVM project ~every week _does suck_ and takes a long time even on a performant machine. To reduce the pain, make sure to follow all of the developer tips in https://llvm.org/docs/CMake.html, only build the `LLVM_TARGETS_TO_BUILD` `host` target, only enable the `mlir` project, and use a fast linker e.g. `lld`.

sequencer, 2023-10-16:
Thanks for reply. 
> However, as far as i am aware (and of which i agree), we have deliberately chosen to track upstream MLIR/LLVM as close as possible (currently on an ~1 week submodule bump cadence, thanks to the people at SiFive).

I observe the current LLVM release flow hits to ~2 week, which is close to the bumping flow in CIRCT which is ~1 week.
And for recently bumping, see #6280, #6211, which didn't touch the circt codebase, the recent codebase change due to #6207 is 3 weeks ago.
Maybe take this suggestion into consideration, if further MLIR releasing is stable enough.

mikeurbach, 2023-10-16:
Thanks for the feedback, but I think Morten summed it up. Right now, MLIR-based projects in the community all deal with this, and the MLIR community doesn't have a good solution beyond tracking head of LLVM.

There are however, things we can do for downstream users of CIRCT. For example, the FIRRTL release include pre-built firtool binaries for different platforms. Is it possible to use those? This was set up with Chisel in mind, so Chisel devs can use a pre-built firtool. Would that work for you @sequencer? If not the firtool binary, we also distribute shared libraries for libCIRCTFirtool and libCIRCTCAPIFIRRTL.

If you can't use the pre-built binaries and libraries, we could also share some common setup tips that people dealing with MLIR use. The tips Morten pointed out are good ones, and I would also recommend setting up ccache and using LLVM_CCACHE_BUILD. A good chunk of the cache will be invalidated on each LLVM version bump, but it can help.