llvm/circt #6291: [Firtool][C-API] Options in Firtool-lib for C-API should not be registered as command line options

SpriteOvO, 2023-10-15:
We are trying to use the new Firtool-lib C-API introduced in #6036, and we found that if we called `firtoolOptionsCreateDefault` twice, an error will be thrown during the second call:

```
: CommandLine Error: Option 'o' registered more than once!
LLVM ERROR: inconsistency in registered CommandLine options
```

After investigation, in the constructor of `llvm::cl::opt`, it will register to the global variable `GlobalParser` in LLVM as a command line option, and the `GlobalParser` refused to register the same option multiple times and threw the above error.

An idea is to make the `firtoolOptions` be a template struct, so that we can have `firtoolOptions<llvm::cl::opt>` for Firtool-bin and `firtoolOptions<llvm::cl::opt_storage>` for C-API, but we also need all `populate-` functions to be templates as well, which means we have to move the definitions of these functions from `.cpp` to `.h` for template instantiations. For `firtoolOptions<llvm::cl::opt_storage>`, we also have to figure out how to discard properties such as `llvm::cl::desc`, `llvm::cl::value_desc`, `llvm::cl::cat`, etc. but keep `llvm::cl::init`, as `llvm::cl::opt_storage` doesn't support these properties. It all sounds a little dirty.

I'm happy to hear of any more elegant possible solutions.

sequencer, 2023-10-15:
The current issue is that `llvm::cl::opt` not being thread safe, so `libfirtool` is not thread safe. The straightforward solution is let each `populateXtoY` function not depending on `llvm::cl::opt` but only depends on a bare datatype.

sequencer, 2023-10-15:
I propose migrate all `llvm::cl::opt`(see https://github.com/SpriteOvO/circt/blob/1cd322d46dd40e36c2583b3f2d48a5548d9da527/include/circt/Firtool/Firtool.h#L26-L281C4) to firtool binary only, while duplicating corresponding variables in the `struct FirtoolOptions`, in the firtool binary, it converts `llvm::cl::opt` to corresponding bare variable via corresponding set and get options.

This can decouple the logic from libfirtool and firtool tool, and provide a thread safe way to invoke pass in the libfirtool.

nandor, 2023-10-16:
I agree with this approach. Individual populate methods could receive their own structs packing the relevant arguments, which should be derived from the `FirtoolOptions` struct. However, we should still try to template the options structure so we can declare the arguments in bulk in other tools.

sequencer, 2023-10-16:
After reviewing #6292, I think it is not a good approach for maintain a unified option, aka `FirtoolOption` for each `populateAtoB` functions.

There should be a unique Option for each `populateAtoB` function,  e.g. `populateAtoBOption`, to precisely store the options it does require.
While `FirtoolOption` can be a `llvm:cl:opt` only exist in the `firtool.cpp` binary, and convert to each `populateAtoBOption` in the user tools.

I know this may duplicate options, but on the other hand, it's more elegant for each `populate` function to be used as a lib by depending on the options are really required.

darthscsi, 2023-10-23:
Libraries should not be registering command line options.  command line options introduce static constructors.  populatePass stuff should never be referring to global values.  There should not be macro magic to switch between opt and storage.  Please have the pass functions be passed in a struct with all the config options.  

https://llvm.org/docs/CodingStandards.html#do-not-use-static-constructors
https://llvm.org/docs/CodingStandards.html#library-layering

darthscsi, 2023-10-23:
#6036 should NOT be doing lots of Macro stuff.  It should be using tablegen to generate stuff.  "One-source-of-truth". There should be ZERO need to keep two things synchronized.  This MUST be fixed and/or #6036 should be reverted.

sequencer, 2023-10-23:
Thanks for review. @darthscsi 
It is NOT going to be reverted, since Chisel already depends on these to experimentally generate the Verilog, and the RocketChip working group starts to use it.

>  It should be using tablegen to generate stuff.

I do agree tablegen is a much better solution! Can you point us a example to do this codegen? I think our next PR is refactoring the populate function definition with tablegen.

darthscsi, 2023-10-23:
You don't get to say it's not going to be reverted.  LLVM coding standards say otherwise.  Please come up with an acceptable solution.

"Code in this tier must not:

Negatively affect development of core tier code, with the sub-community involved responsible for making changes to address specific concerns.
" per https://llvm.org/docs/SupportPolicy.html

darthscsi, 2023-10-23:
I'm not saying I'm going to go revert it now, but be aware that that code CANNOT continue to exist in the repo.  Please take time to reengineer it.

SpriteOvO, 2023-10-26:
I have looked at LLVM/MLIR tablegen, but it doesn't seem to support just generating C code. The generator backends are all domain-specific. Maybe I'm missing something here? @darthscsi Would you mind briefly explaining how I should refactor the code with tablegen? Thanks in advance!

sequencer, 2023-11-04:
Is possible to get this PR into CIRCT firstly? Since there is no tablegen
backend for CAPI header declaration? This sv emitting from Chisel via C
binding is depending on this.

After chisel is fixed, we can start working on the LLVM table ten dialect
to make it support C-API and then refactor CIRCT to use it.

darthscsi, 2023-11-06:
First, why is firtoolOptionsCreateDefault being called more than once?
Second, shouldn't these be managed with an llvm::ManagedStatic<> like the MLIR options are in MLIRContext.cpp (or asmprinter options in AsmPrinter.cpp)?  Note also these separate the struct representing the options from the options themselves (see OpPrintingFlags::OpPrintingFlags()).

I've mentioned it on a few PRs, but it seems to be being ignored:  don't turn to macro magic unless there is no other choice.  Also, look to how other parts of llvm and mlir have solve the same problems and don't invent new ways.

sequencer, 2023-11-06:
> First, why is firtoolOptionsCreateDefault being called more than once?

We cannot assume only call verilog emitter only one time in a single JVM.

SpriteOvO, 2023-11-07:
> Second, shouldn't these be managed with an llvm::ManagedStatic<> like the MLIR options are in MLIRContext.cpp (or asmprinter options in AsmPrinter.cpp)? Note also these separate the struct representing the options from the options themselves (see OpPrintingFlags::OpPrintingFlags()).

I've checked `llvm::ManagedStatic<>` before. The point here is that if multiple instances of `MlirContext` can be built, why do we only allow Options as a static single instance? IMO, it makes sense to allow having multiple `MlirContext`s and configuring different Options to run Passes for them.

> I've mentioned it on a few PRs, but it seems to be being ignored: don't turn to macro magic unless there is no other choice. Also, look to how other parts of llvm and mlir have solve the same problems and don't invent new ways.

AFAIK, we are the first to design C-API for tool options, so there is no similar solution to look at.

mikeurbach, 2023-11-15:
> AFAIK, we are the first to design C-API for tool options, so there is no solution for a similar situation to look at.

If I understand the request, I think this is correct, I'm not sure of anything in LLVM/MLIR we can reuse. I was thinking about this problem a bit myself, and I think I can see where this would go. We could use a new, custom [tablegen](https://llvm.org/docs/TableGen/) format to describe simple structures (as in, C++ / C structs). Then we'd create a new [tablegen backend](https://llvm.org/docs/TableGen/BackGuide.html) for this format. The backend would emit both C++ and C header and implementation files that could be included. The tablegen file in the new format could be the single source of truth for the C++ and C structures for FirtoolOptions, and the functions generated with macros in the PR would instead be generated by tablegen from the single source of truth.

This is not getting into the whole discussion about ManagedStatic, just trying to brainstorm ways to have a single source of truth between C++ and C. This is something we could create within CIRCT, but it should be generic, and could be considered for upstreaming to LLVM/MLIR proper.

sequencer, 2023-11-15:
Yes, IMHO, this is a feature requested from CIRCT, or any other MLIR library exposing its C-API as library, to LLVM, we are happy to implement the tablegen backend declaration of C struct and function prototype. But I don't this that can be finished in a short time...

SpriteOvO, 2023-11-15:
@mikeurbach I agree with your idea of creating a new custom tablegen format for this and the similar cases, this will make the whole thing easier to maintain. Currently the Chisel Binder is blocked by it, we can't run more than one test that involves PassManager because of this issue. I'm wondering if it's possible to merge it first and we'll replace it with the codegen approach in the future once we've implemented the new custom format backend?

mikeurbach, 2023-11-16:
Let me try to reset the conversation, because now I am feeling like we are not discussing the core problem. The amount of macros is one thing, but the real problem here is you want to be able to use different options within one process right?

I would say let's focus on solving that problem in a way that doesn't use macro code generation. I think there is probably a way to solve the core problem with macros and without inventing a new tablegen backend.

I looked over https://github.com/llvm/circt/pull/6315 again, and I have two general points:

* Is it necessary to split up the options into separate structs for each phase of the pipline (e.g., `FirtoolCHIRRTLToLowFIRRTLOptions`)? Just trying to understand why that's required.
* I think most of the changes are reasonable, it's just all the macro stuff in circt-c/Firtool.h. What if you get rid of the codegen? It seems like you need to be able to pass in a C structure from a JVM, why do so many methods need to be generated?

I think a lot of this can be simplified without needing to resort to codegen. If you're able to simplify it, I think it's quite possible to get a solution that meets the policy guidelines Andrew mentioned previously.

SpriteOvO, 2023-11-17:
> Is it necessary to split up the options into separate structs for each phase of the pipline (e.g., `FirtoolCHIRRTLToLowFIRRTLOptions`)? Just trying to understand why that's required.

I have discussed this with @sequencer, the conclusion was that splitting up the options is based on reasons:

- Some options are used only once. If the user (e.g. Binder) chooses to run passes in different combinations, splitting options makes the user just need to fill in the options that will be really used.

- If we don't split them up, the user has to read the function definition to check whether a options is used or not.

> I think most of the changes are reasonable, it's just all the macro stuff in circt-c/Firtool.h. What if you get rid of the codegen? It seems like you need to be able to pass in a C structure from a JVM, why do so many methods need to be generated?

Do you mean removing all getters setters and just populating the struct directly from an external program? This still requires that we have two structs with the same name and different types, one for C-API and the other for command-line. If this is accepted, I will update the PR later with this approach.

darthscsi, 2023-11-20:
> We cannot assume only call verilog emitter only one time in a single JVM

That is a bug in use.  LLVM contexts assume single-definition.  The command line argument library are intended to support command line arguments.  If you find yourself needing to define the commandline arguments more than once, then BY DEFINITION they are not command line arguments.  LLVM doesn't abandon you in this instance, it uses a separation of config values from command line arguments so you are always free not to use the command line argument and instead just fill out the config struct and pass it in.

darthscsi, 2023-11-20:
> The point here is that if multiple instances of `MlirContext` can be built, why do we only allow Options as a static singleton

Command line options are intended for the command line, not for library interfaces.  They exist so people using the library can parse command line arguments for the library without knowing them.  If you are making multiple contexts and duplicating pipelines, you are outside this use case.  In this case, the standard pattern is to provide a config type which you can populate however you want and pass it to the pipeline.  The default commandline definitions should also simply be producing this config type and passing it to the pipeline.

SpriteOvO, 2023-11-20:
> My top level point is the command line option library is for command line options. These are BY DEFINITION singletons to the process. If your process is instantiating multiple contexts or multiple pipelines with different options, then you are outside the use-case for the command line options. You are now in the "construct the config option and pass it to each pipeline instantiation" realm.
> 
> I suspect this is coming from scala-test and trying to make pipelines for each test? One way would to just not do this, use a single context for all the tests. If you really need to modify the pipeline options per test (which I would discourage), then you should just make the config object and pass it in and never instantiate the command line options.

Both of these attempts #6292 #6315 are avoiding instantiating `llvm::cl::opt<T>`. #6315 is currently blocked because of the large number of macros that use and violate the "One-source-of-truth" principle. That does kind of bad, so @mikeurbach suggested a possible solution in https://github.com/llvm/circt/issues/6291#issuecomment-1814814621.

> I suspect this is coming from scala-test and trying to make pipelines for each test? One way would to just not do this, use a single context for all the tests.

Not just testing, there are practical cases. Maybe @sequencer will explain more about the practical cases later.

---

I would like to further explain my previous reply

> Do you mean removing all getters setters and just populating the struct directly from an external program? This still requires that we have two structs with the same name and different types, one [...]

If we removing the getters and setters functions for Options struct, two structs are still necessary, due to the fact that Firtool-bin needs to instantiate `llvm::cl::opt`, and C-API needs a POD type that can be constructed from an external program (e.g. `MlirStringRef`), and a `MlirStringRef` cannot be used as a external storage of `llvm::cl::opt<std::string>`.

Then we make `populateXXX` functions accept C-API struct, and in the case of Firtool-bin, it converts its CommandLine struct (members with `llvm::cl::opt`) into a C-API struct (members with POD types) to be passed into `populateXXX` functions (since C-API structs can be treated as a reference and don't own any data).

This is the simplest solution I can think of at the moment, and this may not require any macro use, just two structs. @darthscsi What do you think about this?

SpriteOvO, 2023-11-20:
Oh! Maybe expose the options needed by `populateXXX` to its parameter list? This way C-API should not need structs.

sequencer, 2023-11-20:
> I suspect this is coming from scala-test and trying to make pipelines for each test? One way would to just not do this, use a single context for all the tests.

So basically, that's my future plan in chisel to provide users, or at least developers, a convenient way to add new passes in the future. For example,
- Adding plugin like what @uenoku did [here](https://github.com/uenoku/firtool-standalone-plugin).
- Optionally extract OM in Scala at Chisel build time.
- Directly lower to ARC Dialect

That's the reason I don't like different `populate` functions coupling too much, since we want add more.

So I personally prefer exposing a "grand type"-based(w/o struct/class) parameter list for each `populate` function, for providing a concise C-API. No macros, no duplication, simple to read, simple to bind(that's what C-API really intend for).

darthscsi, 2023-11-21:
If your intention is to add passes to the pipeline, then you are starting to be outside the "give me a firrtl pipeline" functionality and should probably be adding bindings for pass construction so you can construct pipelines piecemeal.

sequencer, 2023-11-21:
> If your intention is to add passes to the pipeline, then you are starting to be outside the "give me a firrtl pipeline" functionality and should probably be adding bindings for pass construction so you can construct pipelines piecemeal.

yes, so if these firtool pipelines can be decoupled with their standalone parameters, it would be better to use.

SpriteOvO, 2023-11-30:
Fixed in #6438. Appreciate again everyone who participated in the discussion and review :)

closed 2023-11-30 by SpriteOvO