llvm/circt #6321: Integrating the CIRCT workflow into Google Colab

eda-ricercatore, 2023-10-19:
If possible, can you please kindly let me know how can I integrate the CIRCT workflow into Google Colab?

This would allow me to generate the domain-specific processor/architecture, or hardware accelerator, and its compiler using the Multi-Level Intermediate Representation (MLIR) Compiler Framework and LLVM, without having to integrate the Google Silicon framework/support for Jupyter notebook -based integrated circuit (I.C.) design with offline compiler design, and resultant embedded system simulation offline.

Thank you so much, and have a good day!

uenoku, 2023-10-20:
circt wheel is published in pypi so you can run `pip install circt` in Colab, run optimization passes and generate verilog ([sample](https://colab.research.google.com/drive/1eO7DG55HmEXvOlM7Av1Lh6lTeGFQkbo5?usp=sharing)).

uenoku, 2023-10-24:
I'll close the issue now as done, please feel free to reopen if you had an issue with Colab!

closed 2023-10-24 by uenoku