llvm/circt #6324: error: unsupported packed array expression, note: see current operation: %52 = "hw.array_concat"(%4, %51) : (!hw.array<3xi9>, !hw.array<5xi9>) -> !hw.array<8xi9>

oharboe, 2023-10-20:
Discovered here https://github.com/ucb-bar/chipyard/issues/1623#issuecomment-1771455376

Unzip [firfile.zip](https://github.com/ucb-bar/chipyard/files/13046023/firfile.zip)

Run:

```
firtool       --format=fir    --dedup         --export-module-hierarchy       --verify-each=true      --warn-on-unprocessed-annotations       --disable-annotation-classless  --disable-annotation-unknown    --mlir-timing   --lowering-options=emittedLineLength=2048,noAlwaysComb,disallowLocalVariables,verifLabels,locationInfoStyle=wrapInAtSquareBracket,disallowPackedArrays  --split-verilog         -o /tmp/blah chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir
```

Output:

```
$ firtool       --format=fir    --dedup         --export-module-hierarchy       --verify-each=true      --warn-on-unprocessed-annotations       --disable-annotation-classless  --disable-annotation-unknown    --mlir-timing   --lowering-options=emittedLineLength=2048,noAlwaysComb,disallowLocalVariables,verifLabels,locationInfoStyle=wrapInAtSquareBracket,disallowPackedArrays  --split-verilog         -o /tmp/blah chipyard.harness.TestHarness.MegaBoomConfig.sfc.fir
<unknown>:0: warning: option -dedup is deprecated since firtool 1.57.0, has no effect (deduplication is always enabled), and will be removed in firtool 1.58.0
generators/rocket-chip/src/main/scala/util/Arbiters.scala:43:16: error: unsupported packed array expression
generators/rocket-chip/src/main/scala/util/Arbiters.scala:43:16: note: see current operation: %52 = "hw.array_concat"(%4, %51) : (!hw.array<3xi9>, !hw.array<5xi9>) -> !hw.array<8xi9>
[deleted]
```

uenoku, 2023-10-20:
Thank you for the report. This is same issue to https://github.com/llvm/circt/issues/5355.

oharboe, 2023-10-20:
du

oharboe, 2023-10-20:
duplicate of #5355

closed 2023-10-20 by oharboe