llvm/circt #6360: [FIRRTL] Bitcasts reverse bundle field order in LowerTypes

fabianschuiki, 2023-10-31:
Since PR #6339, the `LowerTypes` pass in the FIRRTL dialect reverses the field order for bundles when bitcasting in one direction, but not the other. @SpriteOvO do you have any idea what might be going on here? Could it be that the bundle field order is only honored in one direction of bitcast, but not the other?

### Example 1
```
firrtl.circuit "Queue_4" {
  firrtl.module @Queue_4(
    in %a: !firrtl.bundle<data: uint<512>, strb: uint<64>, last: uint<1>>,
    out %b: !firrtl.bundle<data: uint<512>, strb: uint<64>, last: uint<1>>
  ) {
    %0 = firrtl.bitcast %a : (!firrtl.bundle<data: uint<512>, strb: uint<64>, last: uint<1>>) -> !firrtl.uint<577>
    %1 = firrtl.bitcast %0 : (!firrtl.uint<577>) -> !firrtl.bundle<data: uint<512>, strb: uint<64>, last: uint<1>>
    firrtl.strictconnect %b, %1 : !firrtl.bundle<data: uint<512>, strb: uint<64>, last: uint<1>>
  }
}
```
Running this through `firtool` produces before PR #6339 produces the following assignments:
```
  assign b_data = a_data;
  assign b_strb = a_strb;
  assign b_last = a_last;
```
After the PR the assignments look as follows:
```
  assign b_data = {a_data[446:0], a_strb, a_last};
  assign b_strb = a_data[510:447];
  assign b_last = a_data[511];
```

### Example 2
```
firrtl.circuit "Queue_9" {
  firrtl.module @Queue_9(
    in %a: !firrtl.bundle<addr: uint<25>, len: uint<8>, size: uint<3>, burst: uint<2>, lock: uint<1>, cache: uint<4>, prot: uint<3>, qos: uint<4>, region: uint<4>, id: uint<12>, user: uint<1>>,
    out %b: !firrtl.bundle<addr: uint<25>, len: uint<8>, size: uint<3>, burst: uint<2>, lock: uint<1>, cache: uint<4>, prot: uint<3>, qos: uint<4>, region: uint<4>, id: uint<12>, user: uint<1>>
  ) {
    %0 = firrtl.bitcast %a : (!firrtl.bundle<addr: uint<25>, len: uint<8>, size: uint<3>, burst: uint<2>, lock: uint<1>, cache: uint<4>, prot: uint<3>, qos: uint<4>, region: uint<4>, id: uint<12>, user: uint<1>>) -> !firrtl.uint<67>
    %1 = firrtl.bitcast %0 : (!firrtl.uint<67>) -> !firrtl.bundle<addr: uint<25>, len: uint<8>, size: uint<3>, burst: uint<2>, lock: uint<1>, cache: uint<4>, prot: uint<3>, qos: uint<4>, region: uint<4>, id: uint<12>, user: uint<1>>
    firrtl.strictconnect %b, %1 : !firrtl.bundle<addr: uint<25>, len: uint<8>, size: uint<3>, burst: uint<2>, lock: uint<1>, cache: uint<4>, prot: uint<3>, qos: uint<4>, region: uint<4>, id: uint<12>, user: uint<1>>
  }
}
```
Assignments before PR #6339:
```
  assign b_addr = a_addr;
  assign b_len = a_len;
  assign b_size = a_size;
  assign b_burst = a_burst;
  assign b_lock = a_lock;
  assign b_cache = a_cache;
  assign b_prot = a_prot;
  assign b_qos = a_qos;
  assign b_region = a_region;
  assign b_id = a_id;
  assign b_user = a_user;
```
After the PR the assignments look as follows:
```
  assign b_addr = {a_cache[0], a_prot, a_qos, a_region, a_id, a_user};
  assign b_len = {a_size[1:0], a_burst, a_lock, a_cache[3:1]};
  assign b_size = {a_len[1:0], a_size[2]};
  assign b_burst = a_len[3:2];
  assign b_lock = a_len[4];
  assign b_cache = {a_addr[0], a_len[7:5]};
  assign b_prot = a_addr[3:1];
  assign b_qos = a_addr[7:4];
  assign b_region = a_addr[11:8];
  assign b_id = a_addr[23:12];
  assign b_user = a_addr[24];
```

uenoku, 2023-11-01:
@SpriteOvO I think it was necessary to change this part as well https://github.com/SpriteOvO/circt/blob/9d17f30bdbb4f339fbad9cc04c9df1ae238baec3/lib/Dialect/FIRRTL/Transforms/LowerTypes.cpp#L1392-L1394

sequencer, 2023-11-03:
@SpriteOvO Push

SpriteOvO, 2023-11-03:
Sorry for the late response, I just got back from the other city yesterday. I've opened a new PR #6376 to fix it again without regression.

> @SpriteOvO I think it was necessary to change this part as well [SpriteOvO/circt@`9d17f30`/lib/Dialect/FIRRTL/Transforms/LowerTypes.cpp#L1392-L1394](https://github.com/SpriteOvO/circt/blob/9d17f30bdbb4f339fbad9cc04c9df1ae238baec3/lib/Dialect/FIRRTL/Transforms/LowerTypes.cpp#L1392-L1394)

Thanks for the hint, you are right.

PR #6376 (2024-10-30): [FIRRTL][LowerTypes] Keep the order of bundle fields in lowered `cat`
Fixes #6360.

---

The first fix #6339 (reverted) caused a regression reported in #6360.

In the first try, I forgot to handle the bit extract operation that follows it, the PR tries to fix it again without regression.

---

Credit: Thanks to @uenoku for the hint (https://github.com/llvm/circt/issues/6360#issuecomment-1789251252)
closed 2024-10-30 by prithayan