llvm/circt #6368: [LLHD] Typed pointers are deprecated (warning)

dtzSiFive, 2023-11-01:
Latest build produces warnings like:

```
In file included from /src/circt/lib/Conversion/LLHDToLLVM/LLHDToLLVM.cpp:19:
In file included from /src/circt/include/circt/Dialect/LLHD/IR/LLHDOps.h:24:
In file included from /src/circt/llvm/mlir/include/mlir/Interfaces/FunctionInterfaces.h:17:
/src/circt/llvm/mlir/include/mlir/IR/Builders.h:494:11: warning: 'build' is deprecated: the usage of typed pointers is deprecated [-Wdeprecated-declarations]
    OpTy::build(*this, state, std::forward<Args>(args)...);
          ^
/src/circt/lib/Conversion/LLHDToLLVM/LLHDToLLVM.cpp:108:16: note: in instantiation of function template specialization 'mlir::OpBuilder::create<mlir::LLVM::GEPOp, mlir::LLVM::LLVMPointerType, mlir::Value &, llvm::ArrayRef<mlir::Value>>' requested here
      rewriter.create<LLVM::GEPOp>(loc, LLVM::LLVMPointerType::get(voidPtrTy),
               ^
/src/circt/llvm/build/tools/mlir/include/mlir/Dialect/LLVMIR/LLVMOps.h.inc:5944:5: note: 'build' has been explicitly marked deprecated here
  [[deprecated("the usage of typed pointers is deprecated")]]
```

Something to keep an eye on, didn't look up deprecation timeline (when will this become an error?) for typed pointers but might be good to migrate off before it's removed entirely (and to quiet these warnings in the meantime :+1:).

Are there any challenges preventing moving off this now?

Looking at comments introduced here: #6322 , sounds like at least then the migration was "on-going", so maybe we are just waiting to be able to drop these code paths.

mikeurbach, 2023-11-01:
Yep this was brought up just yesterday in fact: https://discourse.llvm.org/t/psa-removal-of-typed-pointers-from-the-llvm-dialect/74502

maerhart, 2024-07-19:
Since `LLHDToLLVM.cpp` is removed now, this shouldn't be an issue anymore.

closed 2024-07-19 by maerhart