llvm/circt #6388: [HW] Error compiling with `loc(unknown)` for ports

leonardt, 2023-11-07:
Updating to the new module printing syntax from https://github.com/llvm/circt/pull/6205

I'm running into an issue with `loc(unknown)` for a port, here's an example:

```mlir
module {
    hw.module @test(in %I: i1 loc(unknown), out O: i1 loc(unknown)) {
        hw.output %I : i1
    }
}
```

```
circt._mlir_libs.MLIRError: Unable to parse module assembly:
error: "-":3:5: 'hw.module' op block argument locations should match signature locations
 note: "-":3:5: see current operation: 
  "hw.module"() ({
  ^bb0(%arg0: i1):
    "hw.output"(%arg0) : (i1) -> ()
  }) {module_type = !hw.modty<input I : i1, output O : i1>, parameters = [], sym_name = "test"} : () -> ()
```

Now if I replace it with a dummy location that isn't unknown, e.g.

```mlir
module {
    hw.module @test(in %I: i1 loc("foo":1:3), out O: i1 loc("foo":1:3)) {
        hw.output %I : i1
    }
}
```

this seems to compile fine

```verilog
// Generated by CIRCT firtool-1.58.0
module test(    // -:3:5
  input  I,     // foo:1:3
  output O      // foo:1:3
);

  assign O = I; // -:4:9
endmodule
```

seldridge, 2023-11-07:
Thanks for the report. For the MLIR you provided, this trips an assert if you have a build with asserts enabled:

```
Assertion failed: (idx < size()), function operator[], file SmallVector.h, line 294.
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
Stack dump:
0.	Program arguments: circt-opt circt/6388.mlir
Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
0  libLLVMSupport.dylib     0x000000010a828998 llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) + 56
1  libLLVMSupport.dylib     0x000000010a826f00 llvm::sys::RunSignalHandlers() + 112
2  libLLVMSupport.dylib     0x000000010a829000 SignalHandler(int) + 304
3  libsystem_platform.dylib 0x00000001a599aa24 _sigtramp + 56
4  libsystem_pthread.dylib  0x00000001a596bc28 pthread_kill + 288
5  libsystem_c.dylib        0x00000001a5879ae8 abort + 180
6  libsystem_c.dylib        0x00000001a5878e44 err + 0
7  libCIRCTHW.dylib         0x0000000109349958 circt::hw::HWModuleLike::getAllOutputAttrs() (.cold.1) + 0
8  libCIRCTHW.dylib         0x00000001092d8010 circt::hw::HWModuleLike::getInputLocs() + 296
9  libCIRCTHW.dylib         0x00000001092dd368 verifyModuleCommon(circt::hw::HWModuleLike) + 68
10 libCIRCTHW.dylib         0x00000001092dcfbc circt::hw::HWModuleOp::verify() + 88
...
```

I haven't dug into this beyond that.

fzi-hielscher, 2023-11-08:
I had a quick look at it, since it seemed related to #6257. It kind of is:
Arrays of UnknownLocationAttributes get stripped for HW module ports, just like arrays of empty DictionaryAttributes. This then causes a problem when we try to retrieve them.

 I have not done proper testing, but this appears to fix the issue:
```
diff --git a/lib/Dialect/HW/HWOps.cpp b/lib/Dialect/HW/HWOps.cpp
index ae251cd42..3b98148d2 100644
--- a/lib/Dialect/HW/HWOps.cpp
+++ b/lib/Dialect/HW/HWOps.cpp
@@ -1185,13 +1185,15 @@ void HWModuleExternOp::getAsmBlockArgumentNames(
 
 template <typename ModTy>
 static SmallVector<Location> getAllPortLocs(ModTy module) {
-  SmallVector<Location> retval;
   auto locs = module.getPortLocs();
   if (locs) {
+    SmallVector<Location> retval;
     for (auto l : *locs)
       retval.push_back(cast<Location>(l));
     // Either we have a length of 0 or the correct length
     assert(!locs->size() || locs->size() == module.getNumPorts());
+    return retval;
   }
-  return retval;
+  return SmallVector<Location>(module.getNumPorts(),
+                               UnknownLoc::get(module.getContext()));
 }
 
```

seldridge, 2023-11-08:
The shape of that fix sounds right to me and would align with what I saw where locations are properly handled if there is only one unknown location. If all locations are unknown, then it's not storing any location information. The other alternative would be to always store location information, even if unknown. FWIW, I think your fix is fine.

closed 2023-11-15 by fzi-hielscher