llvm/circt #6395: [FIRRTL] Windows builds failing randomly

teqdruid, 2023-11-08:
https://github.com/llvm/circt/actions/runs/6789634524/job/18457283622
https://github.com/llvm/circt/actions/runs/6800811293/job/18490222652
https://github.com/llvm/circt/actions/runs/6804514148/job/18502226729
https://github.com/llvm/circt/actions/runs/6805189828/job/18504213798

Crashes are non-deterministic and all seem to be in FIRRTL tests.

seldridge, 2023-11-08:
Thanks for the report! I had flagged this internally. For good or bad, this is clearly intermittently failing given that commits changing markdown files are causing it to no longer fail.

mikeurbach, 2023-11-09:
Since I run with asan/ubsan, I pulled main and ran check-circt, and we are clean... not sure what is up yet.

seldridge, 2023-11-09:
Poking at some ideas here. Unfortunately, ASAN/UBSAN do not work on Windows (see: https://github.com/llvm/circt/actions/runs/6815862602/job/18535898196#step:9:233):

```
CMake Error at cmake/modules/HandleLLVMOptions.cmake:1017 (message):
  This sanitizer is only supported by clang-cl: Undefined
```

Usually this type of thing has been immediately obvious from Linux/macOS ASAN/UBSAN. However, if we don't have that tool we may be in a lot of pain.

I'm trying bumping the Windows runner from 2019 to 2022 and seeing if that helps.

Other things to try would be to run with `-verbose-pass-executions` to get some idea of when in the pipeline this is failing. (That also may permute things enough that they pass. 😢)

seldridge, 2023-11-09:
I bumped to 2022 as that passed. Obviously, this is not indicative that it is fixed, only that that run worked. Regardless, it's probably good to be using latest. I'll keep an eye on this and see if it stabilizes.

seldridge, 2023-11-11:
This has been stable since the switch to the 2022 image. Will close this. We can reopen another issue if this crops up again.

closed 2023-11-11 by seldridge