llvm/circt #6418: [CMake] Local install doesn't produce CIRCT targets in lib/cmake/mlir/MLIRTargets.cmake

tymcauley, 2023-11-15:
# Context

Platform: macOS 14.1.1
Architecture: arm64
circt commit: Reproduced on both 4f1b8b26b and firtool-1.59.0

# Issue Description

I'm trying to test out the [firtool-standalone-plugin repo](https://github.com/uenoku/firtool-standalone-plugin) locally (platform is macOS arm64), trying to exercise the functionality added in https://github.com/llvm/circt/pull/6254. I can't use the released `circt-full-shared-macos-x64` package, since I get errors linking x86-64 libraries during an arm64 build. Since there's no macOS-arm64 release build, I'm doing a local build/install, but when I build `firtool-standalone-plugin` with that local build, I get this CMake error:

```
CMake Error at /path/to/local/circt/install/lib/cmake/circt/CIRCTConfig.cmake:10 (find_package):
  Found package configuration file:

    /path/to/local/firtool-standalone-plugin/../circt/install/lib/cmake/mlir/MLIRConfig.cmake

  but it set MLIR_FOUND to FALSE so package "MLIR" is considered to be NOT
  FOUND.  Reason given by package:

  The following imported targets are referenced, but are missing:
  CIRCTAffineToLoopSchedule CIRCTArcToLLVM CIRCTCalyxToFSM CIRCTCalyxToHW
  CIRCTCalyxNative CIRCTCombToArith CIRCTCombToLLVM CIRCTConvertToArcs
  CIRCTDCToHW CIRCTExportChiselInterface CIRCTExportVerilog CIRCTFIRRTLToHW
  CIRCTFSMToSV CIRCTHandshakeToDC CIRCTHandshakeToHW CIRCTHWArithToHW
  CIRCTHWToLLHD CIRCTHWToLLVM CIRCTHWToSV CIRCTHWToSystemC CIRCTLLHDToLLVM
  CIRCTLoopScheduleToCalyx CIRCTMooreToCore CIRCTPipelineToHW CIRCTSCFToCalyx
  CIRCTSeqToSV CIRCTCFToHandshake CIRCTVerifToSV CIRCTExportFIRRTL CIRCTComb
  CIRCTESI CIRCTFIRRTL CIRCTMSFT CIRCTMSFTTransforms CIRCTHW CIRCTLLHD
  CIRCTMoore CIRCTOM CIRCTOMEvaluator CIRCTSeq CIRCTSeqTransforms CIRCTSV
  CIRCTSVTransforms CIRCTFSM CIRCTFSMTransforms CIRCTHandshake
  CIRCTHandshakeTransforms CIRCTHWArith CIRCTVerif CIRCTLTL CIRCTFirtool

Call Stack (most recent call first):
  CMakeLists.txt:12 (find_package)
```

And indeed, if I diff my locally-built `lib/cmake/mlir/MLIRTargets.cmake` file with the one in the `firtool-1.59.0` `circt-full-shared-*` release, I see that many of the CIRCT targets are missing from my locally-built copy, such as this one:

```
# Create imported target CIRCTFirtool
add_library(CIRCTFirtool SHARED IMPORTED)

set_target_properties(CIRCTFirtool PROPERTIES
  INTERFACE_LINK_LIBRARIES "..."
)
```

It doesn't seem like there's a clearly-documented flow for reproducing the released artifacts (the two CMake commands in the README are quite different from the command run during the CI-release action), so I extracted the following CMake command from the `.github/workflows` area:

```
CIRCT_DIR="$(pwd)"
CIRCT_INSTALL_DIR="$CIRCT_DIR/install"
BUILD_TYPE=Release
ASSERT=ON
COMPILE_COMMANDS=ON
SHARED=ON
STATS=ON
CLANG_DIR=/opt/homebrew/opt/llvm/bin
C_COMPILER="$CLANG_DIR/clang"
CXX_COMPILER="$CLANG_DIR/clang++"

mkdir build
cd build

cmake \
    -G Ninja \
    -S "$CIRCT_DIR/llvm/llvm" \
    -DCMAKE_BUILD_TYPE=$BUILD_TYPE \
    -DCMAKE_C_COMPILER=$C_COMPILER \
    -DCMAKE_CXX_COMPILER=$CXX_COMPILER \
    -DCMAKE_INSTALL_PREFIX="$CIRCT_INSTALL_DIR" \
    -DCMAKE_EXPORT_COMPILE_COMMANDS=$COMPILE_COMMANDS \
    -DBUILD_SHARED_LIBS=$SHARED \
    -DLLVM_BUILD_TOOLS=ON \
    -DLLVM_BUILD_EXAMPLES=OFF \
    -DLLVM_ENABLE_ASSERTIONS=$ASSERT \
    -DLLVM_ENABLE_PROJECTS=mlir \
    -DLLVM_EXTERNAL_PROJECTS=circt \
    -DLLVM_EXTERNAL_CIRCT_SOURCE_DIR=".." \
    -DLLVM_STATIC_LINK_CXX_STDLIB=ON \
    -DLLVM_ENABLE_TERMINFO=OFF \
    -DLLVM_PARALLEL_LINK_JOBS=1 \
    -DLLVM_TARGETS_TO_BUILD="host" \
    -DLLVM_FORCE_ENABLE_STATS=$STATS \
    -DLLVM_ENABLE_ZSTD=OFF \
    -DVERILATOR_DISABLE=ON \
    -DLLVM_INSTALL_UTILS=ON

ninja check-circt check-circt-unit
ninja install
```

Any clues as to why this might be happening? Happy to provide any extra info that might help.

seldridge, 2023-11-15:
There were some changes to `llvm/circt` cmake which caused this (https://github.com/llvm/circt/commit/cde4642799450170a06f340f2332eecb60e7b34d and https://github.com/llvm/circt/commit/821ab8732065ef96d412e0d358704520096a1e7e). You can work around it by using:

```diff
-  find_package(CIRCT REQUIRED CONFIG)
+  find_package(MLIR CONFIG)
+  find_package(CIRCT CONFIG)
```

You will still get the above error, but it should be a warning.

I am not experienced enough with cmake, nor have I tried to investigate, to know if the original commits that caused this were problematic and where the issue is.

tymcauley, 2023-11-15:
> There were some changes to `llvm/circt` cmake which caused this ([cde4642](https://github.com/llvm/circt/commit/cde4642799450170a06f340f2332eecb60e7b34d) and [821ab87](https://github.com/llvm/circt/commit/821ab8732065ef96d412e0d358704520096a1e7e)). You can work around it by using:
> 
> ```diff
> -  find_package(CIRCT REQUIRED CONFIG)
> +  find_package(MLIR CONFIG)
> +  find_package(CIRCT CONFIG)
> ```
> 
> You will still get the above error, but it should be a warning.
> 
> I am not experienced enough with cmake, nor have I tried to investigate, to know if the original commits that caused this were problematic and where the issue is.

Thanks! Ya, I saw those two MRs, and figured they might have to do with this, but both https://github.com/llvm/circt/pull/6340 and https://github.com/llvm/circt/pull/6347 are included in the firtool-1.59.0 release, and the release artifacts for firtool-1.59.0 contain all of these targets in `lib/cmake/mlir/MLIRTargets.cmake`.

That being said, making your suggested change to the `CMakeLists.txt` in firtool-standalone-plugin did indeed demote this error to a warning, and I was able to complete the build, and reproduce the example from the README. Thanks so much!

I'm happy to close this issue, since I'm unblocked, but I'm still not sure why my local install build doesn't produce the same `lib/cmake/mlir/MLIRTargets.cmake` as the release build...

tymcauley, 2024-01-30:
Not sure what resolved the issue, but using `firtool-1.64.0` I'm able to go back to using `find_package(CIRCT REQUIRED CONFIG)` without error.

closed 2024-01-30 by tymcauley