llvm/circt #6457: [FIRRTL] port bundle flip fixups

darthscsi, 2023-11-28:
`module(input : bundle<all flips>` should be transformed to `module(output : bundle<non-flipped>`

seldridge, 2023-11-28:
I love this line of thinking. However, this needs to either: (1) be reconciled with flip/direction in the FIRRTL spec or (2) this can only be done as part of a whole-circuit pass that removes all (or some) flips.

The first example in https://circt.llvm.org/docs/Dialects/FIRRTL/RationaleFIRRTL/#not-canonicalizing-flip-types shows the problem with this. Copied below:

This circuit is illegal FIRRTL:
```
FIRRTL version 3.3.0
circuit Foo:
  module Foo:
    output a: { flip a: UInt<1> }
    output b: { a: UInt<1> }

    connect b, a
```

Applying the proposed rule would make it legal FIRRTL:
```
FIRRTL version 3.3.0
circuit Foo:
  module Foo:
    input a: { a: UInt<1> }
    output b: { a: UInt<1> }

    connect b, a
```

darthscsi, 2023-11-28:
Alternate take on that after a couple of years, the real problem is output is a duplex flow.

darthscsi, 2023-11-29:
Also, once the connect directions are resolved with strictconnect, it shouldn't matter anymore.