llvm/circt #6466: [FIRRTL] Compile time regression in `filterAnnotation` in LowerTypes

uenoku, 2023-11-29:
`filterAnnotation` filters annotations for a given field id but the current implementation travers every annotation. It takes O(NM) where N is a size of an annotation and M is maximum field id. This could be problematic when port is a large vector (e.g. 2^16 elements) and each leaf field has an annotation (e.g. dontTouch is scattered to each element).

uenoku, 2023-11-29:
This is reproducer:
```scala
//> using scala "2.13.11"
//> using lib "org.chipsalliance::chisel::6.0.0-M3"
//> using plugin "org.chipsalliance:::chisel-plugin::6.0.0-M3"
//> using options "-unchecked", "-deprecation", "-language:reflectiveCalls", "-feature", "-Xcheckinit", "-Xfatal-warnings", "-Ywarn-dead-code", "-Ywarn-unused", "-Ymacro-annotations"

import chisel3._
import circt.stage.ChiselStage
import chisel3.dontTouch

class Foo extends Module {
  val a = IO(Input(Vec(65536, Bool())))
  a.map(dontTouch(_))
}

object Main extends App {
  /*println(
    ChiselStage.emitCHIRRTL(
      gen = new Foo
    )
  )*/
  println(
    ChiselStage.emitSystemVerilog(
      gen = new Foo,
      firtoolOpts = Array("-disable-all-randomization", "-strip-debug-info")
    )
  )
}
```

darthscsi, 2023-11-29:
Can you post the .fir file?

uenoku, 2023-12-07:
```scala
/> using scala "2.13.11"
//> using repository sonatype-s01:snapshots
//> using lib "org.chipsalliance::chisel::6.0.0-M3+117-2372b1c4-SNAPSHOT"
//> using plugin "org.chipsalliance:::chisel-plugin::6.0.0-M3+117-2372b1c4-SNAPSHOT"

//> using options "-unchecked", "-deprecation", "-language:reflectiveCalls", "-feature", "-Xcheckinit", "-Xfatal-warnings", "-Ywarn-dead-code", "-Ywarn-unused", "-Ymacro-annotations"

import chisel3._
import circt.stage.ChiselStage
import chisel3.dontTouch

class Foo extends Module {
  val a = IO(Input(Vec(65536, Bool())))
  val b = IO(Output(Vec(65536, Bool())))
  val r = Reg(Vec(65536, Bool()))
  r := a
  b := r
  r.map(dontTouch(_))
}

object Main extends App {
  /*println(
   ChiselStage.emitCHIRRTL(
    gen = new Foo
  ))*/
  println(
    ChiselStage.emitSystemVerilog(
     gen = new Foo,
      firtoolOpts = Array("-disable-all-randomization", "-strip-debug-info", "-verbose-pass-executions")
    )
   )
}
```

Note that we still need to fix the lower types even when lower-signature is enabled by default if we lower internal signals.