llvm/circt #6474: PyCDE breaks cmake configure

darthscsi, 2023-11-30:
This doesn't work: `CIRCT_BASE="-DLLVM_USE_LINKER=$MY_LINKER -Wno-dev -DCIRCT_RELEASE_TAG=sifive -DCIRCT_RELEASE_TAG_ENABLED=On -DVERILATOR_PATH=/home/andrew/local/verilator/bin/verilator -DCMAKE_CXX_FLAGS_DEBUG=-gdwarf-4 -Wno-dev -DCIRCT_BINDINGS_PYTHON_ENABLED=ON -DCIRCT_BINDINGS_TCL_ENABLED=ON -DCIRCT_ENABLE_FRONTENDS=PyCDE"`

This does: `CIRCT_BASE="-DLLVM_USE_LINKER=$MY_LINKER -Wno-dev -DCIRCT_RELEASE_TAG=sifive -DCIRCT_RELEASE_TAG_ENABLED=On -DVERILATOR_PATH=/home/andrew/local/verilator/bin/verilator -DCMAKE_CXX_FLAGS_DEBUG=-gdwarf-4 -Wno-dev -DCIRCT_BINDINGS_PYTHON_ENABLED=ON -DCIRCT_BINDINGS_TCL_ENABLED=ON"`

I see:
```
CMake Error at cmake/modules/AddLLVM.cmake:1954 (add_dependencies):
  The dependency target "ESIPythonRuntime" of target "check-all" does not
  exist.
Call Stack (most recent call first):
  cmake/modules/AddLLVM.cmake:1994 (add_lit_target)
  CMakeLists.txt:1251 (umbrella_lit_testsuite_end)

CMake Error at cmake/modules/AddLLVM.cmake:1954 (add_dependencies):
  The dependency target "ESIRuntime" of target "check-all" does not exist.
Call Stack (most recent call first):
  cmake/modules/AddLLVM.cmake:1994 (add_lit_target)
  CMakeLists.txt:1251 (umbrella_lit_testsuite_end)

CMake Error at cmake/modules/AddLLVM.cmake:1954 (add_dependencies):
  The dependency target "ESIPythonRuntime" of target
  "check-pycde-integration" does not exist.
Call Stack (most recent call first):
  cmake/modules/AddLLVM.cmake:2020 (add_lit_target)
  /home/andrew/sifive/circt/frontends/PyCDE/integration_test/CMakeLists.txt:46 (add_lit_testsuite)

CMake Error at cmake/modules/AddLLVM.cmake:1954 (add_dependencies):
  The dependency target "ESIRuntime" of target "check-pycde-integration" does
  not exist.
Call Stack (most recent call first):
  cmake/modules/AddLLVM.cmake:2020 (add_lit_target)
  /home/andrew/sifive/circt/frontends/PyCDE/integration_test/CMakeLists.txt:46 (add_lit_testsuite)
```

It looks like the targets are conditionally defined but unconditionally used.

If it matters:
```
-- CIRCT Python bindings are enabled.
-- Found Python3: /usr/bin/python3 (found suitable version "3.11.6", minimum required is "3.6") found components: Interpreter Development Development.Module Development.Embed 
-- Found Python3: /usr/bin/python3 (found suitable version "3.11.6", minimum required is "3.6") found components: Interpreter Development.Module NumPy 
-- Found python include dirs: /usr/include/python3.11
-- Found python libraries: /usr/lib/x86_64-linux-gnu/libpython3.11.so
-- Found numpy v1.24.2: /usr/lib/python3/dist-packages/numpy/core/include
-- Using explicit pybind11 cmake directory: /usr/lib/cmake/pybind11 (-Dpybind11_DIR to change)
-- Found pybind11: /usr/include (found version "2.11.1")
-- Found pybind11 v2.11.1: /usr/include
-- Python prefix = '', suffix = '', extension = '.cpython-311-x86_64-linux-gnu.so
```

teqdruid, 2023-11-30:
What commit do you see this error? I thought I fixed this exact issue yesterday: https://github.com/llvm/circt/commit/ef5727703de1e217e0d57985ea7792ed4d06f426

darthscsi, 2023-11-30:
> yesterday: ef57277

This was this morning (11/30), I'm not sure if your fix was it, I'll try
again.

teqdruid, 2023-12-03:
> yea, seems happy today. Thanks! Sorry for the noise.

The best turnaround times are negative!