llvm/circt #6486: [HandshakeToHW] Wrong lowering for arith.shli op

HahaLan97, 2023-12-04:
I added `arith.shli` op in my custom kernel and used the `hlstool` to generate the SystemVerilog files. There is a `arith_shli_in_ui32_ui32_out_ui32.sv` in the output folder, but inside the code the behavior is wrong as below:
```
// Generated by CIRCT unknown git version
module arith_shli_in_ui32_ui32_out_ui32(	// shl.mlir:3:8
  input  [31:0] in0,
  input         in0_valid,
  input  [31:0] in1,
  input         in1_valid,
                out0_ready,
  output        in0_ready,
                in1_ready,
  output [31:0] out0,
  output        out0_valid
);

  wire _GEN = in0_valid & in1_valid;	// shl.mlir:3:8
  wire _GEN_0 = out0_ready & _GEN;	// shl.mlir:3:8
  assign in0_ready = _GEN_0;	// shl.mlir:3:8
  assign in1_ready = _GEN_0;	// shl.mlir:3:8
  assign out0 = in0 | in1;	// shl.mlir:3:8
  assign out0_valid = _GEN;	// shl.mlir:3:8
endmodule
```
After checking the dialect conversion debug information I narrowed the problem down to the pass of `Handshake-to-HW` in `lib/Conversion/HandshakeToHW/HandshakeToHW.cpp` at line 1889:
```
UnitRateConversionPattern<arith::ShLIOp, comb::OrOp>
```
Changing the `OrOp` to `ShlOp` fixed the issue. Please take a look and fixed this in the upstream ;)

mortbopet, 2023-12-04:
Feel free to submit a PR for the fix 🚀.

RamirezLucas, 2023-12-04:
Apologies I forgot to reference the issue when merging the PR, this is fixed.

closed 2023-12-04 by RamirezLucas