llvm/circt #6496: [FIRRTL] Compiled failed using Clang on Apple Mac M2 platform

davine47, 2023-12-06:
Platform infos:
clang/Clang++ : Apple clang version 14.0.3 (clang-1403.0.22.14.1)
cmake: cmake version 3.27.9
ninja: 1.11.1
circt branch: main(latest)
llvm branch: main(latest)
When compile circt, will report these errors:
---
FAILED: include/circt/Dialect/FIRRTL/FIRRTLTypes.h.inc /Users/wenjunnan/projects/circt/build/include/circt/Dialect/FIRRTL/FIRRTLTypes.h.inc 
cd /Users/wenjunnan/projects/circt/build && /Users/wenjunnan/projects/circt/llvm/build/bin/mlir-tblgen -gen-typedef-decls -typedefs-dialect=firrtl -I /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL -I/Users/wenjunnan/projects/circt/llvm/llvm/include -I/Users/wenjunnan/projects/circt/llvm/build/include -I/Users/wenjunnan/projects/circt/llvm/mlir/include -I/Users/wenjunnan/projects/circt/llvm/build/tools/mlir/include -I/Users/wenjunnan/projects/circt/include -I/Users/wenjunnan/projects/circt/build/include /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTL.td --write-if-changed -o include/circt/Dialect/FIRRTL/FIRRTLTypes.h.inc -d include/circt/Dialect/FIRRTL/FIRRTLTypes.h.inc.d
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTL.td:20:
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLStructure.td:16:
/Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLAttributes.td:75:5: error: Initializer of 'attrName' in 'AugmentedBundleType' could not be fully resolved: !strconcat("firrtl", !strconcat(".", mnemonic))
def AugmentedBundleType : AugmentedType<"AugmentedBundleType"> {
    ^
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTL.td:20:
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLStructure.td:16:
/Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLAttributes.td:87:5: error: Initializer of 'attrName' in 'AugmentedVectorType' could not be fully resolved: !strconcat("firrtl", !strconcat(".", mnemonic))
def AugmentedVectorType : AugmentedType<"AugmentedVectorType"> {
    ^
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTL.td:20:
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLStructure.td:16:
/Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLAttributes.td:92:5: error: Initializer of 'attrName' in 'AugmentedGroundType' could not be fully resolved: !strconcat("firrtl", !strconcat(".", mnemonic))
def AugmentedGroundType : AugmentedType<"AugmentedGroundType"> {
    ^
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTL.td:20:
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLStructure.td:16:
/Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLAttributes.td:97:5: error: Initializer of 'attrName' in 'AugmentedStringType' could not be fully resolved: !strconcat("firrtl", !strconcat(".", mnemonic))
def AugmentedStringType : AugmentedType<"AugmentedStringType"> {
    ^
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTL.td:20:
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLStructure.td:16:
/Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLAttributes.td:101:5: error: Initializer of 'attrName' in 'AugmentedBooleanType' could not be fully resolved: !strconcat("firrtl", !strconcat(".", mnemonic))
def AugmentedBooleanType : AugmentedType<"AugmentedBooleanType"> {
    ^
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTL.td:20:
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLStructure.td:16:
/Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLAttributes.td:105:5: error: Initializer of 'attrName' in 'AugmentedIntegerType' could not be fully resolved: !strconcat("firrtl", !strconcat(".", mnemonic))
def AugmentedIntegerType : AugmentedType<"AugmentedIntegerType"> {
    ^
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTL.td:20:
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLStructure.td:16:
/Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLAttributes.td:109:5: error: Initializer of 'attrName' in 'AugmentedDoubleType' could not be fully resolved: !strconcat("firrtl", !strconcat(".", mnemonic))
def AugmentedDoubleType : AugmentedType<"AugmentedDoubleType"> {
    ^
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTL.td:20:
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLStructure.td:16:
/Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLAttributes.td:113:5: error: Initializer of 'attrName' in 'AugmentedLiteralType' could not be fully resolved: !strconcat("firrtl", !strconcat(".", mnemonic))
def AugmentedLiteralType : AugmentedType<"AugmentedLiteralType"> {
    ^
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTL.td:20:
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLStructure.td:16:
/Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLAttributes.td:117:5: error: Initializer of 'attrName' in 'AugmentedDeletedType' could not be fully resolved: !strconcat("firrtl", !strconcat(".", mnemonic))
def AugmentedDeletedType : AugmentedType<"AugmentedDeletedType"> {
    ^
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTL.td:27:
/Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLTypesImpl.td:39:5: error: Initializer of 'typeName' in 'SIntImpl' could not be fully resolved: !strconcat("firrtl", !strconcat(".", mnemonic))
def SIntImpl : FIRRTLImplType<"SInt",
    ^
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTL.td:27:
/Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLTypesImpl.td:58:5: error: Initializer of 'typeName' in 'UIntImpl' could not be fully resolved: !strconcat("firrtl", !strconcat(".", mnemonic))
def UIntImpl : FIRRTLImplType<"UInt",
    ^
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTL.td:27:
/Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLTypesImpl.td:77:5: error: Initializer of 'typeName' in 'ClockTypeImpl' could not be fully resolved: !strconcat("firrtl", !strconcat(".", mnemonic))
def ClockTypeImpl : FIRRTLImplType<"Clock"> {
    ^
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTL.td:27:
/Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLTypesImpl.td:91:5: error: Initializer of 'typeName' in 'ResetTypeImpl' could not be fully resolved: !strconcat("firrtl", !strconcat(".", mnemonic))
def ResetTypeImpl : FIRRTLImplType<"Reset"> {
[... 63 lines cut ...]
def PathImpl : PropImplType<"Path"> {
    ^
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTL.td:27:
/Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLTypesImpl.td:529:5: error: Initializer of 'typeName' in 'BoolImpl' could not be fully resolved: !strconcat("firrtl", !strconcat(".", mnemonic))
def BoolImpl : PropImplType<"Bool"> {
    ^
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTL.td:27:
/Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLTypesImpl.td:535:5: error: Initializer of 'typeName' in 'DoubleImpl' could not be fully resolved: !strconcat("firrtl", !strconcat(".", mnemonic))
def DoubleImpl : PropImplType<"Double"> {
    ^
Included from /Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTL.td:27:
/Users/wenjunnan/projects/circt/include/circt/Dialect/FIRRTL/FIRRTLTypesImpl.td:39:5: error: Record `SIntImpl', ` field `typeName' exists but does not have a string initializer!
def SIntImpl : FIRRTLImplType<"SInt",
    ^
[64/815] Building FIRRTL.h.inc...[68/815] Building CXX object CMakeFiles/llvm_gtest.dir/llvm/third-party/unittest/googletest/src/gtest-all.cc.o
In file included from /Users/wenjunnan/projects/circt/llvm/third-party/unittest/googletest/src/gtest-all.cc:42:
/Users/wenjunnan/projects/circt/llvm/third-party/unittest/googletest/src/gtest-death-test.cc:619:5: warning: default label in switch which covers all enumeration values [-Wcovered-switch-default]
    default:
    ^
In file included from /Users/wenjunnan/projects/circt/llvm/third-party/unittest/googletest/src/gtest-all.cc:49:
/Users/wenjunnan/projects/circt/llvm/third-party/unittest/googletest/src/gtest.cc:3128:5: warning: default label in switch which covers all enumeration values [-Wcovered-switch-default]
    default:
    ^
2 warnings generated.
ninja: build stopped: subcommand failed.

SpriteOvO, 2023-12-06:
macOS + clang builds are fine in CI ([link](https://github.com/llvm/circt/actions/runs/7111001863)), could you show what the exact build commands you ran?

davine47, 2023-12-06:
> macOS + clang builds are fine in CI ([link](https://github.com/llvm/circt/actions/runs/7111001863)), could you show what the exact build commands you ran?

Hi , here are the history commands i ran as in README
```text
 1312  git clone https://mirrors.tuna.tsinghua.edu.cn/git/llvm-project.git
 1313  ls
 1314  mv llvm-project llvm
 1315  ls
 1316  cd llvm
 1317  ls
 1318  git branch
 1319  mkdir build
 1320  cd build
 1321  cat ../../README.md
 1322  cmake -G Ninja ../llvm \\n    -DLLVM_ENABLE_PROJECTS="mlir" \\n    -DLLVM_TARGETS_TO_BUILD="host" \\n    -DLLVM_ENABLE_ASSERTIONS=ON \\n    -DCMAKE_BUILD_TYPE=DEBUG \\n    -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
 1323  ninja
 1324  ls
 1325  cd ..
 1326  ls
 1327  cd ..
 1328  ls
# here llvm builds success, I delete the old circt build dir
 1329  rm -rf build
 1330  ls
 1331  mkdir build
 1332  cd build
 1333  cat ../README.md
 1334  cmake -G Ninja .. \\n    -DMLIR_DIR=$PWD/../llvm/build/lib/cmake/mlir \\n    -DLLVM_DIR=$PWD/../llvm/build/lib/cmake/llvm \\n    -DLLVM_ENABLE_ASSERTIONS=ON \\n    -DCMAKE_BUILD_TYPE=DEBUG \\n    -DCMAKE_EXPORT_COMPILE_COMMANDS=ON
 1335  ninja
# report the above ERRORs

```

SpriteOvO, 2023-12-06:
CIRCT bound LLVM as a dependency with a strict git commit ID, not a release, which means that it's very likely to cause the build to fail if you are not using the matched LLVM git commit ID.

Try to use `git submodule` commands to fetch LLVM repository, if you have to use a mirror, you can modify the LLVM repository URL in `.gitmodules`.

```bash
git clone https://github.com/llvm/circt.git
cd ./circt
git checkout <commit-id-you-want-to-build-with>
sed -i'' -e "s|https://github.com/llvm/llvm-project.git|https://mirrors.tuna.tsinghua.edu.cn/git/llvm-project.git|g" ./.gitmodules
git submodule update --init

# Explanation of sed -i''
# https://stackoverflow.com/a/4247319/13433323
```

davine47, 2023-12-06:
> CIRCT bound LLVM as a dependency with a strict git commit ID, not a release, which means that it's very likely to cause the build to fail if you are not using the matched LLVM git commit ID.
> 
> Try to use `git submodule` commands to fetch LLVM repository, if you have to use a mirror, you can modify the LLVM git URL in `.gitmodules` file.
> 
> ```shell
> git clone https://github.com/llvm/circt.git
> cd ./circt
> git checkout <commit-id-you-want-to-build-with>
> sed -i'' -e "s|https://github.com/llvm/llvm-project.git|https://mirrors.tuna.tsinghua.edu.cn/git/llvm-project.git|g" ./.gitmodules
> git submodule update --init
> 
> # Explanation of sed -i''
> # https://stackoverflow.com/a/4247319/13433323
> ```

Cool! It works! ^ ^
I pass the whole compilation process with the correct commit ID.
Thanks!

closed 2023-12-06 by davine47