llvm/circt #6524: [Question] A way to getOrAddInnerSym from external programs?

SpriteOvO, 2023-12-14:
Last week, I implemented *probe* related stuff in Chisel binder (chipsalliance/chisel#3668). In CIRCT, `firrtl.ref.rwprobe` needs a `target` attribute with `InnerRefAttr` type. So I filled `inner_sym` attribute for all ops that have it. It worked fine then.

In recent days, I've noticed that CIRCT complains an error if an op's `inner_sym` is filled and not referenced later.

```
error: unable to lower due to symbol "_WIRE" with target not preserved by lowering
```

firtool doesn't have this problem, because firtool fills the `inner_sym` only when it needs to refer to an op result, via function [`getOrAddInnerSym`](https://github.com/llvm/circt/blob/96ddd2fed1a344979b8ffc3a1ca9343f2b5f7cb6/include/circt/Dialect/FIRRTL/FIRRTLUtils.h#L127-L146) (called from [`FIRStmtParser::parseRWProbe`](https://github.com/llvm/circt/blob/96ddd2fed1a344979b8ffc3a1ca9343f2b5f7cb6/lib/Dialect/FIRRTL/Import/FIRParser.cpp#L3397-L3400)).

But in the external program, there is no such information as `InnerSymbolNamespace` and `uint64_t fieldID` (because they are private). What would be the best way to handle a situation like the `target` attribute of `rwprobe`?

SpriteOvO, 2024-01-13:
Solved by set the `inner_sym` only when needed, by using `mlirOperationSetInherentAttributeByName` function.

Thanks @seldridge's help!

closed 2024-01-13 by SpriteOvO

seldridge, 2024-01-13:
Nice! I'm glad you got this working.