llvm/circt #6543: [FIRRTL] Chisel assertions are translated into concurrent SVA rather than immediate SVA

techninja1008, 2023-12-29:
When using `firtool` through Chisel 5.1.0 with the `--emit-chisel-asserts-as-sva` flag, assertions are always outputted as concurrent SVA properties. This is problematic for using with the free SymbiYosys verification toolchain, which only supports immediate assertions.

As a workaround I am currently using [this patch](https://gist.github.com/techninja1008/4a3f51aec47588673c02d3e41393ec5e) which adds an `--emit-asserts-as-sby-sva` flag to firtool, forcing all chisel verification statements to be emitted as immediate SVA.

uenoku, 2023-12-29:
Is it necessary to use `--emit-chisel-asserts-as-sva` in the first place? Even though I added `--emit-chisel-asserts-as-sva` flag for SFC migration a few years ago but probably we would want to work toward removing the flag to get more predictable results.

techninja1008, 2023-12-30:
Without the flag, FIRRTL automatically transforms any asserts it recognizes as produced in the same pattern chisel writes them into when-guarded prints and fails.

Bearing in mind I'm fairly new to CIRCT - ideally, I think neither flag would exist, and instead we'd have a few extra FIRRTL passes:

* A "simplify chisel asserts" pass that turns the assertion constructs that chisel outputs into single assert statements
* Passes for conversion between immediate and concurrent verification statements
* A pass to convert verification statements into the same when-guarded print/fail construct that FIRRTL outputs currently

This would still leave SymbiYosys to be figured out though, as there would need to be some way to remove else-clauses from verification statements, which (from my understanding) aren't supported, even with immediate SVA. Perhaps another pass for this?

The idea would be these passes could then be requested using flags on `firtool` etc.

uenoku, 2024-04-01:
Sorry for delay but https://github.com/llvm/circt/pull/6885 should fix address this issue.

closed 2024-04-02 by uenoku