llvm/circt #6557: Segfault when running firtool 1.60.0, standalone example included

oharboe, 2024-01-08:
```
$ firtool --version
LLVM (http://llvm.org/):
  LLVM version 18.0.0git
  Optimized build.
CIRCT firtool-1.60.0
```

untar [firtool-crash.tar.gz](https://github.com/llvm/circt/files/13859206/firtool-crash.tar.gz)

Run:

```
firtool       --format=fir    --export-module-hierarchy       --verify-each=true    --warn-on-unprocessed-annotations       --disable-annotation-classless  --disable-annotation-unknown    --mlir-timing         --lowering-options=emittedLineLength=2048,noAlwaysComb,disallowLocalVariables,verifLabels,locationInfoStyle=wrapInAtSquareBracket,disallowPackedArrays,omitVersionComment     --repl-seq-mem-file=megaboom/mems.conf  --annotation-file=anno.json   --strip-debug-info   --strip-fir-debug-info   --extract-test-code       --split-verilog     --disable-all-randomization         -o megaboom/    ChipLikeMegaBoomConfig.sfc.fir
```

Output:

```
ChipLikeMegaBoomConfig.sfc.fir:2:1: warning: unprocessed annotation:'firrtl.transforms.NoCircuitDedupAnnotation$' still remaining after LowerToHW
circuit TestHarness :
^
ChipLikeMegaBoomConfig.sfc.fir:2:1: warning: unprocessed annotation:'logger.LogLevelAnnotation' still remaining after LowerToHW
circuit TestHarness :
^
ChipLikeMegaBoomConfig.sfc.fir:43120:3: warning: unprocessed annotation:'freechips.rocketchip.util.RegFieldDescMappingAnnotation' still remaining after LowerToHW
  module PeripheryBus :
  ^
ChipLikeMegaBoomConfig.sfc.fir:282740:3: warning: unprocessed annotation:'freechips.rocketchip.util.RetimeModuleAnnotation' still remaining after LowerToHW
  module IntToFP :
  ^
ChipLikeMegaBoomConfig.sfc.fir:557900:3: warning: unprocessed annotation:'freechips.rocketchip.util.ParamsAnnotation' still remaining after LowerToHW
  module BoomTile :
  ^
ChipLikeMegaBoomConfig.sfc.fir:696542:3: warning: unprocessed annotation:'freechips.rocketchip.util.AddressMapAnnotation' still remaining after LowerToHW
  module DigitalTop :
  ^
generators/rocket-chip/src/main/scala/util/DescribedSRAM.scala:17:26: warning: unprocessed annotation:'freechips.rocketchip.util.SRAMAnnotation' still remaining after LowerToHW
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
Segmentation fault (core dumped)
```

oharboe, 2024-01-08:
Same crash with:

```
$ firtool --version
LLVM (http://llvm.org/):
  LLVM version 18.0.0git
  Optimized build with assertions.
CIRCT firtool-1.61.0
```

oharboe, 2024-01-08:
Without `--extract-test-code`, it doesn't crash.

yupferris, 2024-01-08:
I was able to reproduce in the debugger on current master (3ce5c8c5b90e089c3e526053f20b58bfc20055ed). Looks like a null reference in `mlir::Operation::getLoc` from [this call](https://github.com/llvm/circt/blob/main/lib/Dialect/SV/Transforms/HWLegalizeModules.cpp#L259). It's likely that it's lowering some packed array stuff which has a module input (or similar) as an input (the indexer, probably) rather than a local op. I'll dig a bit deeper.

uenoku, 2024-01-10:
> Looks like a null reference in mlir::Operation::getLoc from [this call](https://github.com/llvm/circt/blob/main/lib/Dialect/SV/Transforms/HWLegalizeModules.cpp#L259)

We can avoid the failure by just calling `input.getLoc()`. 

> The fix should be to lower packed array module inputs into individual inputs for each array element, and fix up module instantiations accordingly (similar treatment for outputs would also fix [this test case](https://github.com/llvm/circt/blob/main/test/Dialect/SV/hw-legalize-modules-packed-arrays.mlir#L3-L18))

I agree we can flatten IO in LegalizeModules but it would be very complicated to do that in that pass. I think extending `FlattenIO` is suitable for that purpose.