llvm/circt #6565: [Python] Nightly wheels filling up space on PyPI.

mikeurbach, 2024-01-09:
Our wheel build currently weighs in at 124.6MB, and by publishing this nightly, we have filled up our allocated 10GB on PyPI.

I have opened a request for more space via the process on pypi.org, to give us years of headroom at the current growth rate: https://github.com/pypi/support/issues/3423. But this has not been acknowledged, and there is no guarantee that it will be accepted, so I want to discuss other remedies. In the meantime, I have already had to delete older nightly releases from PyPI twice.

I think the options are to build less wheels, and to build less frequently. Anything else? I want to discuss with the community before making any changes.

To review, we currently build Python 3.8, 3.9, 3.10, and 3.11 for both MacOS and Linux.

@leonardt as far as I understand, you are the main user of these wheels. How often do you actually pull down new wheels? Can we reduce the cadence less than daily? What versions / OSes do you use?

@teqdruid do you use the nightly wheels at all?

For the work we're doing at SiFive, I think we'd be happy with weekly or semiweekly, and only really need a single Linux version.

leonardt, 2024-01-09:
We pin the wheel to the current version tested at the time of release, so nightly is not necessary for us unless we are trying out a new feature/bugfix, so it could be fine for us to dev builds manually when needed versus nightly, or to just wait for a release wheel.

Right now we test Python 3.8 and Python 3.10.  Based on https://devguide.python.org/versions/, I'd imagine it would be good to maintain at least one security version for stability, whether it's the oldest or newest I'm not sure what is best.  For the bugfix and development versions, I'd imagine using the latest  version could be okay because if a user is already on the bleeding edge, then they are probably not opposed to upgrading for the latest wheel.

leonardt, 2024-01-09:
We do use the macOS version of the wheel, weekly or less would be fine.

mikeurbach, 2024-01-09:
Ok, thanks @leonardt. I think we can prune 3.9 and 3.11 since it seems like no one uses those. So the matrix would be (MacOS, Linux) x (3.8, 3.10). And since it doesn't seem super important to have the latest wheels available nightly, I will update the dev release builds to happen weekly. Ideally there should be a full release weekly anyway, but in case it is delayed, it doesn't hurt to have relatively fresh wheels automatically. With that we should have much more headroom in the short term.

mikeurbach, 2024-01-10:
Another option I'm considering is to not push dev wheels to PyPI at all. They persist as artifacts on the GitHub action for 1 week. And if we run this weekly, there should always be a recent dev wheel available.

For now, I will reduce the matrix and frequency of dev wheels, but still upload to PyPI, because it's nice to be able to install from the main package index.

mikeurbach, 2024-01-10:
I made the changes here: https://github.com/llvm/circt/commit/392af2fe5e622f031a178cb2f46abc235e5f4c05. I'll close this issue, but we can revisit it if necessary.

closed 2024-01-10 by mikeurbach

mikeurbach, 2024-01-10:
@teqdruid sorry, I should have been more specific, do you use the `circt` nightly wheels? I was under the impression PyCDE wheels, nightly or otherwise, are still built from scratch and uploaded from a separate CI/CD system. I'm currently trying to make sure we don't fill up https://pypi.org/project/circt/.

teqdruid, 2024-01-10:
> You are receiving this email because you were mentioned.

No, I do not use the circt wheels at all.