llvm/circt #6581: Will the `AttributeAnnotation` support setting the target to port?

Discreater, 2024-01-16:
I noticed that “A reference target must be a wire, node, reg, or module” in [docs](https://circt.llvm.org/docs/Dialects/FIRRTL/FIRRTLAnnotations/#annotations).  And in the code it says "Currently ports are not supported". https://github.com/llvm/circt/blob/7c59a2ef4324576db7d2b8ff6d24823f7562776f/lib/Dialect/FIRRTL/Transforms/LowerAnnotations.cpp#L323-L325
 Is there any plan to support ports?

uenoku, 2024-01-16:
There is no plan but we can support that. I expect it wont't be that hard.

Discreater, 2024-01-18:
Thanks for the good news! I'm really looking forward to this feature, as it's been blocking me from upgrading my Chisel project to 5.0+.

lordspacehog, 2024-04-18:
+1 on this. It's required for adding attributes to ports for vivado to do port type inference. (https://docs.amd.com/r/2021.1-English/ug994-vivado-ip-subsystems/List-of-Supported-X_-Attributes). This is often much more convenient to do with existing bundle definitions then trying to use the chisel DataView API. (especially for diplomatic interfaces)