llvm/circt #6616: [FIRRTL] Crash with const fields in enum types

mbty, 2024-01-27:
The following circuit leads to a `firtool` crash:
```
circuit c:
  type enum = {|a: const UInt<8>, b: UInt<16>|}
  module c:
```

The backtrace is the following:
```
<unknown>:0: error: enum with 'const' elements must be 'const'
firtool:circt/llvm/mlir/include/mlir/IR/StorageUniquerSupport.h:180: static ConcreteT mlir::detail::StorageUserBase<ConcreteT, BaseT, StorageT, UniquerT, Traits>::get(mlir::MLIRContext*, Args&& ...) [with Args = {llvm::ArrayRef<circt::firrtl::FEnumType::EnumElement>&, bool&}; ConcreteT = circt::firrtl::FEnumType; BaseT = circt::firrtl::FIRRTLBaseType; StorageT = circt::firrtl::detail::FEnumTypeStorage; UniquerT = mlir::detail::TypeUniquer; Traits = {circt::hw::FieldIDTypeInterface::Trait}]: Assertion `succeeded(ConcreteT::verify(getDefaultDiagnosticEmitFn(ctx), args...))' failed.
PLEASE submit a bug report to https://github.com/llvm/circt and include the crash backtrace.
Stack dump:
0.	Program arguments: firtool const_in_enum.fir
Stack dump without symbol names (ensure you have llvm-symbolizer in your PATH or set the environment var `LLVM_SYMBOLIZER_PATH` to point to it):
0  firtool   0x00005c2b8efe82fa llvm::sys::PrintStackTrace(llvm::raw_ostream&, int) + 74
1  firtool   0x00005c2b8efe8708
2  firtool   0x00005c2b8efe5b3b llvm::sys::RunSignalHandlers() + 146
3  firtool   0x00005c2b8efe7be1
4  libc.so.6 0x00007a5561237710
5  libc.so.6 0x00007a556128783c
6  libc.so.6 0x00007a5561237668 raise + 24
7  libc.so.6 0x00007a556121f4b8 abort + 215
8  libc.so.6 0x00007a556121f3dc
9  libc.so.6 0x00007a556122fd26
10 firtool   0x00005c2b8fdbdd2a
11 firtool   0x00005c2b8fda9db1 circt::firrtl::FEnumType::get(mlir::MLIRContext*, llvm::ArrayRef<circt::firrtl::FEnumType::EnumElement>, bool) + 57
12 firtool   0x00005c2b8f13b1a0
13 firtool   0x00005c2b8f13c6c9
14 firtool   0x00005c2b8f157bcd
15 firtool   0x00005c2b8f1578fd
16 firtool   0x00005c2b8f159805
17 firtool   0x00005c2b8f15a196 circt::firrtl::importFIRFile(llvm::SourceMgr&, mlir::MLIRContext*, mlir::TimingScope&, circt::firrtl::FIRParserOptions) + 807
18 firtool   0x00005c2b8eec3499
19 firtool   0x00005c2b8eec446b
20 firtool   0x00005c2b8eec462d
21 firtool   0x00005c2b8eec4cd3
22 firtool   0x00005c2b8eec50fc main + 451
23 libc.so.6 0x00007a5561220cd0
24 libc.so.6 0x00007a5561220d8a __libc_start_main + 138
25 firtool   0x00005c2b8eec2d35 _start + 37
```

mbty, 2024-01-27:
Note that changing the definition to
```
circuit c:
  type enum = const {|a: const UInt<8>, b: UInt<16>|}
  module c:
```
doesn't seem to change anything (surprisingly, even the `<unknown>:0: error: enum with 'const' elements must be 'const'` error remains).

dtzSiFive, 2024-01-30:
Previously: #5884