llvm/circt #6618: [LTL] Add operators to enhance expressive power

liuyic00, 2024-01-29:
Current LTL dialect provides a limited set of operators and cannot achieve the expressive power of LTL or SVA. This issue will discuss different levels of expressive power progressively and possible plans to add operators in the LTL dialect.

To differentiate between concepts, `ltl` will represent the LTL dialect, while "LTL" will denote the academic Linear Temporal Logic.

## LTL and Regular Expression

`Seqeunce` and `Property` in SVA could be consider as regular expression and LTL. In my point of view, `Sequence` and `Property` serve different purposes in SVA, although some properties can be expressed in both `Sequence` and `Property`. `Sequence` uses its operators to easily express a fragment that will match on the trace. `Property` uses its propositional logical operators and temporal modal operators to express the temporal property that we want to hold on the trace.

This part adds operators that could achieve the expressive power of LTL and regular expressions. This means `ltl` can achieve the expressive power of SVA, although the lack of syntactic sugar may result in larger formulas.

### Add `NextOp` and `UntilOp` for LTL

Academic LTL has two parts of operators, propositional logical operators and temporal modal operators. `ltl` dialect already includes operators `NotOp`,  `AndOp` and `OrOp`, which satisfies the needs of propositional logical. But temporal modal operators $X$ (next) and $U$ (until) are missing, which are the most basic operators in the academic LTL, consider the operators like $F$ (future/eventually) and $G$ (globally/always) can be expressed using $X$ and $U$.

Thus, by adding `NextOp` and `UntilOp` to `Property` in `ltl`, we can achieve the expressive power of LTL. Although `DelayOp` has a similar effect to `NextOp`, it can only act on `Sequence`, not on `Property`.

### Add `RepeatOp` for Regular Expression

Basic operators of regular expression are $union$ (or), $concatenation$ (connect) and $closure$ (unbounded repetition).  Add `RepeatOp` to `Sequence` in `ltl` will complete the missing semantics.

`DelayOp` on `Sequence` can be considered as the combination of connect and wildcard if we use a regular expression to explain.

However, the expressive power of regular expression on infinite trace (ω-regular) is stronger than LTL. For example, "`b` must hold at odd positions" (in `Sequence` is `(##1 b)[*]`) cannot be described in LTL. In an intuitive sense, `RepeatOp` repeats `DelayOp`, but LTL does not have an operator to repeat `NextOp`. Due to the difference in expressive power, using `ltl` as the name for the dialect may not be quite appropriate.

It would be better to consider the `AndOp` on `Sequence` as a special case of the `intersect` discussion below.

## SVA Basic Operators

In addition to the mentioned operators above, SVA basic operators also include the following operators. These operators do not improve the expressive power.

- `Property`
  - `until_with`
- `Sequence`
  - `[+]` `[*...]`

`until_with` is an overlapping version of `UntilOp`. These can be implemented as a parameterized `UntilOp` with a boolean field `overlapping`, or expand `a until_with b` to `a until (a and b)`. There is also a *strong* version ` until ` in the SVA advanced operators. May be both `overlapping` and `strong/week` could be the parameter. Expanding an operator into some basic operators is also a good implementation approach, but I'm not sure the cost of patten match if we want to restore the original operator. Original `RepeatOp` and `[+]`, `[*...]` can be implemented as `RepeatOp(x,0,$)` `RepeatOp(x,1,$)` `RepeatOp(x,a,b)` like `DelayOp` in current `ltl`.

## SVA Advanced Operators

SVA Advanced operators imports many useful operators to reduce the size of formulas, especially `intersect`. These operators do not improve the expressive power.

- `Property`
  - `implies` `iff` `if else` `case`
    - These operators can be expanded to basic propositional logical operators like `NotOp` and `OrOp` in Chisel directly if we do not want add them into CIRCT.
  - `s_nexttime` `s_until` `s_until_with` `s_always` `eventyally` and bounded version
    - The `strong/week`, `overlapping/nonoverlapping`  and `bound` can be parameters of the operator.
    - These operators can also be expanded to basic operators. But the size of the expansion of `bound` will be related to the value of `bound`.
  - `#-#` `#=#`
    - `s #-# p` $\equiv$ `not (s |-> not p)`
    - `s #=# p` $\equiv$ `not (s |=> not p)`
- `Sequence`
  - `intersect`  `throughout` `within` `and`
    - `throughout` `within` `and` can be represented by `intersect`. However, expanding `intersect` requires listing all possible intersection traces of two `Sequence`, resulting in a significant expansion of the formula.
  - `[->…]` `[=…]`
    - `b[->m:n]` $\equiv$ `(!b[*] ##1 b)[*m:n]`
    - `b[=m:n]` $\equiv$ `b[->m:n] ##1 !b[*]`
    - `n` can be infinite `$`

These operators are also part of SVA advanced operators, but I don't have any ideas about implementation beyond simply adding the operators directly.

- `accept_on` `reject_on` `sync_accept_on` `sync_reject_on`
  - These operators are related to multi-clock scenarios, enhancing the ability to describe properties across multiple clocks, but not altering the expressive power we discussed earlier.
- `strong` `weak`
- `first_match`

## Other Features in SVA

- local variable

darthscsi, 2024-02-07:
Getting more LTL/SVA will be fantastic.  Implication and strong are things I've had users ask and haven't gotten to.

liuyic00, 2024-04-02:
Here is a full plan to add operators according to the following principles.

- Support all property/sequence operations in SVA.
  - Operations related to clocks, such as `accept_on` , are not considered.
- Merge similar operations to reduce operator and unnecessary grammar sugar.
- Only allow simple combinations of operators to express the operations in SVA.
  - e.g. `p iff q` is equivalent to `(not p and not q) or (p and q)`, but this operator combination is too complicated. Therefore, the `ltl.iff` operator is added.

The mapping from LTL dialect to SVA is list below, including existing ones to illustrate the overall situation. `b`, `s r`, `p q` and `bv` represent different variables with type of `!i1`, `!ltl.sequence`, `!ltl.property` and bool vector needed by `ltl.case`. `T` and `F` represent `True` and `False` as parameters.

```
!ltl.sequence:
--------------

                        0      | 1      | n         | n, 0   | n, m
ltl.delay  %s, n[, m] = ##[*]s | ##[+]s | ##[*n:$]s | ##n s  | ##[n:m]s
ltl.repeat %s, n[, m] = s[*]   | s[+]   | s[*n:$]   | s[*n]  | s[*n:n+m]
ltl.goto   %b, n[, m] =                   b[->n:$]  | b[->n] | b[->n:n+m]

ltl.concat %s0{, %si} = s0 {## si}

ltl.and %s0{, %si} = s0 {and si}
ltl.or %s0{, %si}  = s0 {or si}

ltl.intersect %r, %s = r intersect s

ltl.first %s = first_match(s)

!ltl.property:
--------------

ltl.implication %s, %p  = s |-> p
ltl.follow      %s, %p  = s #-# p

ltl.not %p         = not p
ltl.or  %p0{, %pi} = p0 {or pi}
ltl.and %p0{, %pi} = p0 {and pi}
ltl.implies %p, %q = p implies q
ltl.iff     %p, %q = p iff q
                                  
ltl.case %bv, %p0 {, %bvi, %pi} = case(bv) {bvi: pi;} defalut: p0; endcase
  ltl.case %bv, %true{, %bvi, %pi} = case(bv) {bvi: pi;} endcase
  ltl.case %b, %q,    %true, %p    = if(bv) p1 else p0
  ltl.case %b, %true, %true, %p    = if(bv) p1

                               False(weak)             |          True(strong)
                        1          | n                 | 1            | n
ltl.next   %p, n, T/F = nexttime p | nexttime[n] p     | s_nexttime p | s_nexttime[n] p
ltl.keep   %p, n, T/F =              always[0:n] p     |                s_always[0:n] p
ltl.happen %p, n, T/F =              eventually[0:n] p |                s_eventually[0:n] p

ltl.always     %p = always p
ltl.eventually %p = s_eventually p

                      False(weak) | True(strong)
ltl.until %p, %q, T/F = p until q | p s_until q

ltl.strong %s = strong(s)
ltl.weak   %s = weak(s)
```

Here is the mapping from SVA to LTL dialect:

<details>
<summary>Mapping from SVA to LTL dialect</summary>

```
Sequence:

s[*]   | s[+]                = ltl.repeat %s, 0    | ltl.repeat %s, 1
s[*n]  | s[*n:$]  | s[*n:m]  = ltl.repeat %s, n, 0 | ltl.repeat %s, n | ltl.repeat %s, n, m-n
##[*]s | ##[+]s              = ltl.delay  %s, 0    | ltl.delay  %s, 1
##n s  | ##[n:$]s | ##[n:m]s = ltl.delay  %s, n, 0 | ltl.delay  %s, n | ltl.delay  %s, n, m-n
b[->n] | b[->n:$] | b[->n:m] = ltl.goto   %b, n, 0 | ltl.goto   %b, n | ltl.goto   %b, n, m-n
b[=n]  | b[=n:$]  | b[=n:m]  = ltl.concat (ltl.goto %b, n, 0),   (ltl.delay (ltl.not %b), 1, 0) |
                               ltl.concat (ltl.goto %b, n),      (ltl.delay (ltl.not %b), 1, 0) |
                               ltl.concat (ltl.goto %b, n, m-n), (ltl.delay (ltl.not %b), 1, 0)

r intersect s  = ltl.intersect %r, %s
b throughout s = ltl.intersect (ltl.repeat %b, 0), %s
r within s     = (1[*] ##1 r ##1 1[*]) intersect s
               = ltl.intersect (
                   ltl.concat (ltl.repeat %true, 0),
                              (ltl.delay %r, 1, 0),
                              (ltl.delay (ltl.repeat %true, 0), 1, 0)
                 ), %s
and            = ((r ##1 1[*]) intersect s) or (r intersect (s ##1 1[*]))
               = ltl.and %s0{, %si}
or             = ltl.or %s0{, %si}

first_match(s) = ltl.first %s

Property:

s|->p | s|=>p = ltl.implication %s, %p | ltl.implication (ltl.delay %s, 1, 0), %p
s#-#p | s#=#p = ltl.follow      %s, %p | ltl.follow      (ltl.delay %s, 1, 0), %p

not p       = ltl.not %p 
or          = ltl.or %p0{, %pi}
and         = ltl.and %p0{, %pi} 
p implies q = ltl.implies %p, %q
p iff q     = ltl.iff %p, %q

nexttime   p | nexttime[n]   p = ltl.next %p, 1, F | ltl.next %p, n, F 
s_nexttime p | s_nexttime[n] p = ltl.next %p, 1, T | ltl.next %p, n, T

  always p      | always[n:$] p                  | always[n:m] p       
= ltl.always %p | ltl.next (ltl.always %p), n, F | ltl.next (ltl.keep %p, m-n), n, F

  s_eventually p    | s_eventually[n:$] p                | s_eventually[n:m] p 
= ltl.eventually %p | ltl.next (ltl.eventually %p), n, T | ltl.next (ltl.happen %p, m-n, T), n, T

s_always[n:m] p   = ltl.next (ltl.keep   %p, m-n, T), n, T
eventually[n:m] p = ltl.next (ltl.happen %p, m-n, F), n, F

p until q      | p s_until q      = ltl.until %p, %q, F               | ltl.until %p, %q, T
p until_with q | p s_until_with q = ltl.until %p, (ltl.and %p, %q), F | ltl.until %p, (ltl.and %p, %q), T

case(bv) {bvi: pi;} defalut: p0; endcase = ltl.case %bv, %p0  {, %bvi, %pi}
case(bv) {bvi: pi;} endcase              = ltl.case %bv, %true{, %bvi, %pi}
if(bv) p1 else p0                        = ltl.case %b, %q,    %true, %p
if(bv) p1                                = ltl.case %b, %true, %true, %p

strong(s) = ltl.strong %s
weak(s)   = ltl.weak   %s
```
</details>