llvm/circt #6628: [FIRRTL] FIRRTL Emitter Annotation Support

seldridge, 2024-01-30:
For some simple annotations, it would be useful (in a nebulous "seems like a good idea" sense) to have the FIRRTL emitter be able to serialize annotations. Currently all annotations are just dropped. This is pretty difficult to do for all annotations (and for user-defined annotations that are loaded from a library) because some annotations are scattered disparately out into the circuit. It would be good to at least support this for trivial single target annotations.

E.g., getting the following to round-trip would be great:

```
FIRRTL version 4.0.0
circuit Foo: %[[
  {"class": "firrtl.transforms.DontTouchAnnotation", "target": "~Foo|Foo>a"}
]]
  module Foo:

    wire a: UInt<1>
```

Currently this will strip the annotation:

```
# firtool -parse-only ~/repos/github.com/seldridge/firrtl-snippets/DontTouchAnnotation.fir | circt-translate -export-firrtl
FIRRTL version 4.0.0
circuit Foo :
  module Foo : @[<stdin> 3:5]
    wire a : UInt<1> @[<stdin> 4:12]
```

Moving the FIRRTL spec away from JSON for annotations and towards annotations on operations would help immensely here. Non-local annotations have no such simple representation today.

dtzSiFive, 2024-01-30:
cc https://github.com/llvm/circt/issues/5207

seldridge, 2024-01-31:
I'm going to edit #5207 with the example from here. Thanks for the link!

closed 2024-01-31 by seldridge