llvm/circt #6653: [ESI] Nightly Failure

seldridge, 2024-02-06:
There was a nightly failure seen in ESI here: https://github.com/llvm/circt/actions/runs/7799293590/job/21269799065#step:8:337

This may be related to this change: https://github.com/llvm/circt/commit/2d1df42ac326e3a568f67b29456524180cc1e64c#diff-cd3841e90b3c5aa54afa73e68ef831f8f5e2cf50390547b1578229871dd1c83bR339

teqdruid, 2024-02-06:
> This may be related to this change: 2d1df42
> #diff-cd3841e90b3c5aa54afa73e68ef831f8f5e2cf50390547b1578229871dd1c83bR339
> @teqdruid <https://github.com/teqdruid>: Could you poke at this?
> You are receiving this email because you were mentioned.

Saw that. Was gonna look into it today. I think I know the cause.

teqdruid, 2024-02-06:
I'm pretty sure this resolves the failure (which would have been non-deterministic).

commit 756f5adc (2024-02-06): [ESI][Runtime] Return a py::object so we can return None on failure

Fixes #6653 and adds functionality to test the fix.
closed 2024-02-06 by teqdruid