llvm/circt #6658: Ibis.class must implement symboltable if ibis.container is to be nested below

darthscsi, 2024-02-07:
LLVM bump fails as IBIS is violating the correctness constraint for symbols.

darthscsi, 2024-02-07:
https://github.com/llvm/circt/blob/main/docs/RationaleSymbols.md

mikeurbach, 2024-02-08:
@mortbopet I took a quick look at this, and I don't think it's a quick fix like the other cases. We can't just make ibis.class a SymbolTable, because that will break symbol lookups at the top-level module. The same reason hw.module isn't a SymbolTable.

But it looks like we do actually look up ibis.container by its Symbol name, unlike the other simple cases that didn't really need to be a Symbol. I don't really understand how this ibis.path stuff is working, but I think something needs to change here in order for ibis.container to be legal (i.e., not be a Symbol). Maybe it can be made an InnerSymbol, and whatever is looking it up as a Symbol can use the InnerSymbol infrastructure. Can you please take a look?

teqdruid, 2024-02-08:
I think this is the case we discussed at an ODM some months ago: Ibis needs InnerSymbolRefs to have more than two levels and nested InnerSymbolTables. Look at this note: https://github.com/llvm/circt/blob/a790af2991bfacb96c26cef0853b76901ba38ca3/include/circt/Dialect/Ibis/IbisOps.td#L573

mikeurbach, 2024-02-09:
Ah yes, this is the tracking issue: https://github.com/llvm/circt/issues/6330.

I am not really aware of what Ibis is trying to do by defining ibis.container as a Symbol, or if we need to make the nested InnerSymbolTables work to stop doing that. Hopefully there is something straightforward, because the current state is not legal, and MLIR is now enforcing that.

teqdruid, 2024-02-09:
This one isn't a quick fix. Let's xfail the tests and point at this issue.

PR #6739 (2024-02-26): [Ibis] Split ContainerOp in two
In order to not abuse symbol tables, create an OuterContainerOp for top-level containers, and InnerContainerOp for nested containers. The former has a symbol and is an InnerSymbolTable. The latter has an InnerSymbol.

Fixes #6658.
commit d17f2f1a (2024-02-26): [Ibis] Split ContainerOp in two (#6739)

In order to not abuse symbol tables, create an OuterContainerOp for top-level containers, and InnerContainerOp for nested containers. The former has a symbol and is an InnerSymbolTable. The latter has an InnerSymbol.

Fixes #6658.
closed 2024-02-26 by teqdruid