llvm/circt #6659: OM.class must implement symboltable if om.class.field is to be nested

darthscsi, 2024-02-07:
LLVM bump fails as OM is violating the correctness constraint for symbols.

mikeurbach, 2024-02-07:
Oops, thanks for bringing this up. These were intended to be SymbolTables. We've also discussed broader changes where `om.class.field` can be changed from a Symbol to a terminator, more like `hw.output`, but we should fix this in the meantime. I will take a look at this one.

darthscsi, 2024-02-07:
https://github.com/llvm/circt/blob/main/docs/RationaleSymbols.md

mikeurbach, 2024-02-07:
For the record... there is more nuance than my previous statement. These were originally SymbolTables, but for the reasons outlined above, that doesn't work. We made these no longer SymbolTables, but we didn't migrate them to InnerSymbolTables, and their fields to InnerSymbols, so now the fixed MLIR verifiers are complaining about the Symbols.

closed 2024-02-07 by mikeurbach