llvm/circt #6669: HW TypeScopeOp must be used in SymbolTable or stop being a Symbol.

mikeurbach, 2024-02-07:
Symbol must be contained in a SymbolTable, which MLIR is starting to enforce. HW TypeScope op is a Symbol, and is normally used in builtin ModuleOps. TypeAliasType references to type declarations are using the nested symbol reference mechanism like `@typescope::@typedecl`. This seems to actually work fine for the most part.

However, there is a test case where a HW TypeScope is defined within an ifdef: https://github.com/llvm/circt/blob/dab7c4b614cfa0cb1a9178ab265223576d9d02c5/test/Conversion/ExportVerilog/hw-typedecls.mlir#L42

This is illegal. I don't see how any TypeAliasTypes could even refer to such a TypeScope. We look up the TypeScope symbol here: https://github.com/llvm/circt/blob/dab7c4b614cfa0cb1a9178ab265223576d9d02c5/lib/Conversion/ExportVerilog/ExportVerilog.cpp#L1738

That uses both parts of the SymbolRefAttr to first look up the TypeScope, then the TypeDecl within the TypeScope: https://github.com/llvm/circt/blob/dab7c4b614cfa0cb1a9178ab265223576d9d02c5/lib/Dialect/HW/HWTypes.cpp#L800-L810

If any TypeAliasType tried to refer to `@pycde::@exTypedef` in the example above, it would return null and ExportVerilog would error out with "unresolvable type reference". Is this just an old broken example? If we actually want TypeScope underneath ifdef ops, we'll need to change something.

@teqdruid any thoughts on this, since it's PyCDE related?

teqdruid, 2024-02-08:
I think this pre-dates the automatic ifdef generation which I see now. PyCDE actually uses [VerbatimOps to generate the guards](https://github.com/llvm/circt/blob/28cdbbcccdcb9a7c402165fb573f8729663a917c/frontends/PyCDE/src/types.py#L230C1-L236C52)!

Either way, this functionality is no longer needed. Is it just the test which is failing? If so, just ditch it.

mikeurbach, 2024-02-08:
That makes sense, thanks for the pointer. I will remove this old test code and that should be it.

closed 2024-02-08 by mikeurbach