llvm/circt #6694: [FIRRTL] Failure with Nightly Chisel re: layers

seldridge, 2024-02-14:
The following circuit is currently failing to compile:

```
FIRRTL version 3.3.0
circuit Foo :
  declgroup A, bind :
    declgroup B, bind :
      declgroup C, bind :
    declgroup D, bind :
  module Foo :
    input port : UInt<1>

    group A :
      wire a : UInt<1>
      connect a, port
      group B :
        wire b : UInt<1>
        connect b, a
        group C :
          wire c : UInt<1>
          connect c, b

      group D :
        node _d_T = xor(port, a)
        wire d : UInt<1>
        connect d, _d_T
```

```
Website.scala:30:29: error: 'firrtl.strictconnect' op using value defined outside the region
Website.scala:30:29: note: see current operation: "firrtl.strictconnect"(%3#0, <<UNKNOWN SSA VALUE>>) : (!firrtl.uint<1>, !firrtl.uint<1>) -> ()
Website.scala:18:7: note: required by region isolation constraints
```

rwy7, 2024-02-14:
Thanks for reporting!

rwy7, 2024-02-14:
Likely related failure:
```
// RUN: circt-opt -firrtl-lower-layers -split-input-file %s | FileCheck %s

firrtl.circuit "CaptureHardware" {
  firrtl.layer @A bind {
    firrtl.layer @B bind {}
  }
  firrtl.module @CaptureHardware() {
    %c0_ui1 = firrtl.constant 0 : !firrtl.uint<1>
    firrtl.layerblock @A {
      %0 = firrtl.add %c0_ui1, %c0_ui1 : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<2>
    }
  }
}

```

generates:

```
// -----// IR Dump After LowerLayers Failed (firrtl-lower-layers) //----- //
"firrtl.circuit"() ({
  "sv.verbatim"() {format_string = "`include \22groups_CaptureHardware_A.sv\22\0A`ifndef groups_CaptureHardware_A_B\0A`define groups_CaptureHardware_A_B", output_file = #hw.output_file<"groups_CaptureHardware_A_B.sv", excludeFromFileList>, symbols = []} : () -> ()
  "sv.verbatim"() {format_string = "`ifndef groups_CaptureHardware_A\0A`define groups_CaptureHardware_A", output_file = #hw.output_file<"groups_CaptureHardware_A.sv", excludeFromFileList>, symbols = []} : () -> ()
  "firrtl.module"() ({
  ^bb0(%arg0: !firrtl.uint<1>, %arg1: !firrtl.uint<1>):
    %0 = "firrtl.add"(%arg1, %arg1) : (!firrtl.uint<1>, !firrtl.uint<1>) -> !firrtl.uint<2>
  }) {annotations = [], convention = #firrtl<convention internal>, layers = [], portAnnotations = [], portDirections = 0 : i2, portLocations = [loc("../scratch.mlir":10:12), loc("../scratch.mlir":10:12)], portNames = ["_", "_"], portSyms = [], portTypes = [!firrtl.uint<1>, !firrtl.uint<1>], sym_name = "CaptureHardware_A", sym_visibility = "private"} : () -> ()
  "firrtl.module"() ({
    %0 = "firrtl.constant"() {value = 0 : ui1} : () -> !firrtl.uint<1>
    %1:2 = "firrtl.instance"() {annotations = [], layers = [], lowerToBind, moduleName = @CaptureHardware_A, name = "captureHardware_A", nameKind = #firrtl<name_kind droppable_name>, output_file = #hw.output_file<"groups_CaptureHardware_A.sv", excludeFromFileList>, portAnnotations = [[], []], portDirections = 0 : i2, portNames = ["_", "_"]} : () -> (!firrtl.uint<1>, !firrtl.uint<1>)
    "firrtl.strictconnect"(%1#1, %arg0) : (!firrtl.uint<1>, !firrtl.uint<1>) -> ()
    "firrtl.strictconnect"(%1#0, %0) : (!firrtl.uint<1>, !firrtl.uint<1>) -> ()
  }) {annotations = [], convention = #firrtl<convention internal>, layers = [], parameters = [], portAnnotations = [], portDirections = 0 : i0, portLocations = [], portNames = [], portSyms = [], portTypes = [], sym_name = "CaptureHardware"} : () -> ()
  "sv.verbatim"() {format_string = "`endif // groups_CaptureHardware_A", output_file = #hw.output_file<"groups_CaptureHardware_A.sv", excludeFromFileList>, symbols = []} : () -> ()
  "sv.verbatim"() {format_string = "`endif // groups_CaptureHardware_A_B", output_file = #hw.output_file<"groups_CaptureHardware_A_B.sv", excludeFromFileList>, symbols = []} : () -> ()
}) {annotations = [], name = "CaptureHardware"} : () -> ()
```

lower-layers is busted when a value is captured by the layerblock twice.

This pass is extracting layerblocks into modules, which are then bound-in at the original site of the layerblock. When a layerblock refers to a value defined in the module, the value needs to be "passed in" to the layerblock-module via a port.

To decide if a value is defined outside the layerblock, we look at the value's defining op, and see if the layerblock is an ancestor of the op. If the value is a block-argument, then the layerblock is never considered an ancestor.

Then,  we have to go and capture the value. We create an argument on the layerblock's block, which will eventually become a port on the generated module. Then, for all uses of the original value under the layerblock, we replace it with a use of the new block argument.

The problem is, the layerblock is not considered an ancestor of the block argument. When a value is used more than once, the first capture works, but the second capture will attempt to re-capture the argument, resulting in a bogus second block argument, driven by the first block-argument, by a connect that is placed in the original module.

PR #6699 (2024-02-15): [FIRRTL][Lower-Layers] do not capture uses multiple times
This PR fixes two issues.

First, when a captured value has been replaced by a block-argument on the layerblock, if we encounter a use of the block-argument, we would attempt to "recapture" the argument, because our check for "is this value defined inside the layerblock" was broken for block arguments. This PR fixes that check. Now a value can be captured more than once.

Second, when capturing a value as a "source/input-port", we should only replaces uses where the value is used as a source. Similarly, when a value is captured as a "si [... cut]
commit 3859a19e (2024-02-15): [FIRRTL][Lower-Layers] do not capture uses multiple times (#6699)

Fixes: #6694
closed 2024-02-15 by rwy7